wmbusmeters/wmbusmeters #1123: no driver

ockproff, 2023-12-18:
This is a counter for which I can't find a driver.

It's actually just a "Cyble5" WMBus transmitter from Itron.

Received telegram from: 000xxxxx
          manufacturer: (ITU) ITRON United States Itron, United States of America (0x2695)
                  type: Radio converter (meter side) (0x37) encrypted
                   ver: 0x42
      Concerning meter: 0xxxxxxx
          manufacturer: (PIP) Hermann Pipersberg jr, Germany (0x4130)
                  type: Gas meter (0x03) encrypted
                   ver: 0x00
                device: cul
                  rssi: -67 dBm
                driver: unknown!

https://wmbusmeters.org/analyze/3B4495265341070042378c2049900f002c2555B80200B0c8d7048146B13172277927043041000349001007102f2f04943e7cBa01002f2f2f2f2f2f2f:auto:7A0EEA1A3114B2A067EA09D7E471065C

ockproff, 2023-12-25:
As soon as the decryption key is omitted, a driver (apator08) is suggested, but unfortunately the value that the software recognizes is not the count value of the gas meter. (04943e7c total consumption (694825.302667 m3) ) The value also jumps from 1,367.04 m3 to 1,427,430.4 m3 in one day.

[image]

MDS49FRE, 2023-12-28:
Same here. I'm looking for a suitable driver for ITRON Cyble 5 which is a radio converter for water & gaz. In my case it's mounted on a water meter.

I'm using a RTL2838UHIDIR with rtlwmbus.

Tried to create a new driver but I'm stuck with this log :

`(wmbusmeters) version: 1.14.0-71-g331c5a4`
`(config) store meter files in: "/var/lib/wmbusmeters/meter_readings"`
`(config) using device: rtlwmbus:t1` 
`(config) number of meters: 1`
`Started config rtlwmbus[00000001] listening on t1`
`(rtlwmbus) using command: /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s`
`(serialcmd) opened /bin/sh pid 2945 fd 5 (rtlwmbus)`
`(main) regular reset of rtlwmbus  will happen every 82800 seconds`
`(meter) WaterMeter: meter detection did not match the selected driver itron! correct driver is: unknown!`
`(meter) Not printing this warning again for id: 0021xxxx mfct: (ITU) ITRON United States Itron, United States of America (0x2695) type: Radio converter (meter side) (0x37) ver: 0x42`
`(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/`
`(meter) to add support for this unknown mfct,media,version combination`
`(meter) created WaterMeter itron 0021xxxx encrypted`
`(meter) used meter template WaterMeter 0021xxxx itron to match 0021xxxx`
`(meter) started meter 1 (WaterMeter 0021xxxx itron)`
`(meter) WaterMeter: meter detection did not match the selected driver itron! correct driver is: unknown!`
`(meter) Not printing this warning again for id: 0021xxxx mfct: (ITU) ITRON United States Itron, United States of America (0x2695) type: Radio converter (meter side) (0x37) ver: 0x42`
`(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/`
`(meter) to add support for this unknown mfct,media,version combination`
`(meter) WaterMeter(1) itron  handling telegram from 0021xxxx`
`(telegram) DLL L=0c C=47 (from meter ACC_NR) M=2695 (ITU) A=0021xxxx VER=42 TYPE=37 (Radio converter (meter side)) (driver unknown!) DEV=rtlwmbus[00000001] RSSI=116`
`(telegram) ELL CI=8c CC=c0 (bidir fast_resp) ACC=00`
`(meter) newly created meter (WaterMeter 0021xxxx itron) did not handle telegram!`
`(wmbus) telegram from 0021xxxx ignored by all configured meters!`
`(wmbus) skipping already handled telegram leng=13.`
`(wmbus) skipping already handled telegram leng=13.`
`(wmbus) skipping already handled telegram leng=13.`
`(wmbus) skipping already handled telegram leng=13.`
`(wmbus) skipping already handled telegram leng=13.`
`(wmbus) skipping already handled telegram leng=13.`
`(wmbus) skipping already handled telegram leng=13.`
`(wmbus) skipping already handled telegram leng=13.`
`(wmbus) skipping already handled telegram leng=13.`
`(serial) event loop stopped!`
`(serialcmd) closed /bin/sh pid=2945 fd=5 (rtlwmbus)`

ockproff, 2023-12-29:
My water meter with a Cyble 5 works with the driver: **"dme_07"** have you tried it yet?

For me, only the gas counter is the problem, although both have a Cyble 5 attachment.

MDS49FRE, 2023-12-30:
Yes I tried today and it's still the same problem. Stuck.

`(wmbusmeters) logging started 2023-12-30 09:47:01 1.14.0-71-g331c5a4`
`[2023-12-30_09:47:01] Started config rtlwmbus[00000001] listening on t1`
`(wmbusmeters) waiting for telegrams`
`[2023-12-30_09:47:03] (memory) rss 4489216 peak 4.28 MiB`
`[2023-12-30_09:47:15] (meter) MyTapWater: meter detection did not match the selected driver dme_07! correct driver is: unknown!`
`(meter) Not printing this warning again for id: 0021xxxx mfct: (ITU) ITRON United States Itron, United States of America (0x2695) type: Radio converter (meter side) (0x37) ver: 0x42`
`[2023-12-30_09:47:15] (meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/`
`[2023-12-30_09:47:15] (meter) to add support for this unknown mfct,media,version combination
(wmbusmeters) started meter 1 (MyTapWater 0021xxxx dme_07)`
`[2023-12-30_09:47:15] (meter) newly created meter (MyTapWater 0021xxxx dme_07) did not handle telegram!
(wmbusmeters) shutting down`

tjach, 2024-01-18:
Same problem here:

~~~
(serial) received ascii "C1;1;1;2024-01-18 08:32:51.000;136;102;01040486;0x0c4795268604040142378cc000<0A>C1;1;1;2024-01-18 08:32:51.000;111;113;01040486;0x0c4795268604040142378cc000<0A>"
(rtlwmbus) checkRTLWMBusFrame "C1;1;1;2024-01-18 08:32:51.000;136;102;01040486;0x0c4795268604040142378cc000<0A>C1;1;1;2024-01-18 08:32:51.000;111;113;01040486;0x0c4795268604040142378cc000<0A>"
(rtlwmbus) received full frame
(meter) no meter handled  checking 0 templates.
(wmbus) parseDLL @0 13
(wmbus) parseELL @10 3
(wmbus) parseDLL @0 13
(telegram) DLL L=0c C=47 (from meter ACC_NR) M=2695 (ITU) A=01040486 VER=42 TYPE=37 (Radio converter (meter side)) (driver unknown!) DEV=rtlwmbus[512913565] RSSI=136
(wmbus) parseELL @10 3
(telegram) ELL CI=8c CC=c0 (bidir fast_resp) ACC=00
Received telegram from: 01040486
          manufacturer: (ITU) ITRON United States Itron, United States of America (0x2695)
                  type: Radio converter (meter side) (0x37)
                   ver: 0x42
                device: rtlwmbus[512913565]
                  rssi: 136 dBm
                driver: unknown!
(wmbus) 000   : 0c length (12 bytes)
(wmbus) 001   : 47 dll-c (from meter ACC_NR)
(wmbus) 002   : 9526 dll-mfct (ITU)
(wmbus) 004   : 86040401 dll-id (01040486)
(wmbus) 008   : 42 dll-version
(wmbus) 009   : 37 dll-type (Radio converter (meter side))
(wmbus) 010   : 8c ell-ci-field (ELL: Extended Link Layer I (2 Byte))
(wmbus) 011   : c0 ell-cc (bidir fast_resp)
(wmbus) 012   : 00 ell-acc
telegram=|_0C4795268604040142378CC000|+15
~~~

TaxiPapiche, 2024-02-02:
I have a water meter with Cyble5
I receive the following frames:

wmbusmeters auto:c1 ITRON auto 00065044 NOKEY
result
Started auto rtlwmbus[00000001] listening on c1
(meter) ITRON: meter detection **could not find driver** for id: 00065044 mfct: (ITU) ITRON United States Itron, United States of America (0x2695) type: Radio converter (meter side) (0x37) ver: 0x42
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) newly created meter (ITRON 00065044 auto) did not handle telegram!
...................................
wmbusmeters auto:c1
result : 
Started auto rtlwmbus[00000001] listening on c1
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 00065044
          manufacturer: (ITU) ITRON United States Itron, United States of America (0x2695)
                  type: Radio converter (meter side) (0x37)
                   ver: 0x42
                device: rtlwmbus[00000001]
                  rssi: 135 dBm
                **driver: unknown!**
.....................................
sudo wmbusmeters --debug --t1 rtlwmbus
result
(rtlwmbus) checkRTLWMBusFrame "C1;1;1;2024-01-22 13:05:41.000;154;105;00065044;0x0c4795264450060042378cc000<0A>"
(rtlwmbus) received full frame
(meter) no meter handled  checking 0 templates.
......................................
sudo rtl_sdr -d 0 -f 868.620M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s
result
C1;1;1;2024-01-22 13:00:52.000;151;112;00065044;0x0c4795264450060042378cc000
C1;1;1;2024-01-22 13:00:52.000;124;127;00065044;0x0c4795264450060042378cc000
C1;1;1;2024-01-22 13:01:08.000;153;116;00065044;0x0c4795264450060042378cc000
??

Like you there is an unknown driver
Can you tell me a command to use to force the "dme_07" driver please
I would like to test it
Otherwise what do you think about?
Thank you

closed 2024-05-03 by github-actions[bot]