wmbusmeters/wmbusmeters #1127: Not all values decoded from Diehl

vhamburger, 2023-12-29:
### Type of request

Existing device improvement

### Meter description

Hi, 
I am using your component via https://github.com/SzczepanLeon/esphome-components integration. When I use the izar driver to decode the telegram I get some values (battery life, alarm, transmit interval) but not the actual water consumption. 

When using your online debugger I get the following output:
telegram=|1944A511780757550023A2_411000131F98C356B81C5BC160C063|

Auto driver  : izar
Best driver  : unknown 00/00
Using driver : izar 00/00
000   : 19 length (25 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : a511 dll-mfct (DME)
004   : 57550023 dll-id (23005557)
008   : 78 dll-version
009   : 07 dll-type (Water meter)
010   : a2 tpl-ci-field (Mfct specific)
011 C?: 411000131F98C356B81C5BC160C063 mfct specific

{
    "media":"water",
    "meter":"izar",
    "name":"",
    "id":"23005557",
    "last_month_total_m3":175.379,
    "remaining_battery_life_y":8,
    "total_m3":177.886,
    "transmit_period_s":8,
    "current_alarms":"no_alarm",
    "last_month_measure_date":"2023-12-19",
    "previous_alarms":"no_alarm",
    "timestamp":"2023-12-29T08:44:21Z"
}

Using: wmbusmeters: 1.14.0-48-gc4b9f0d
c4b9f0d10477cb6bb2472e1f2de5eb08ba42e7c4

It seems 011 C?: 411000131F98C356B81C5BC160C063 mfct specific this is a problem - maybe the IZAR driver is not fully recognizing my unit? Any help would be highly appreciated! 

I have an existing issue open here as well (more logs): https://github.com/SzczepanLeon/esphome-components/issues/64 

### Logged telegrams from meter in decrypted format

```shell
1944A511780757550023A2711000131F88B28C457E2FAB0DEB64
1944A511780757550023A2011000131FED7E70149B14A60CA469
1944A511780757550023A2111000131FFA5139BFBAC77FD7BD6B
1944A511780757550023A2211000131FC320E342D8B315BA966C
1944A511780757550023A2311000131FD40FAAE9F960CC618F6E
1944A511780757550023A2711000131F88B28C457E2FAB0DEB64
1944A511780757550023A2011000131FED7E70149B14A60CA469
1944A511780757550023A2111000131FFA5139BFBAC77FD7BD6B
1944A511780757550023A2211000131FC320E342D8B315BA966C
1944A511780757550023A2411000131FB1C356B81C5BC160C063
1944A511780757550023A2011000131FED7E70149B14A60CA469
```

Andi252, 2023-12-31:
Hi,

the decoding looks fine to me, the json-output contains all data you need (for example total_m3 contains the current meter value).

When you check the izar/diehl driver at: [izar driver cpp](https://github.com/wmbusmeters/wmbusmeters/blob/331c5a4018ac75bf4a0f0be4571b061836a6bf7c/src/driver_izar.cc#L187C18-L187C32)
you can see in the method ::processContent() the meter sends the data in a more compact way, probably to squeeze more data into a smaller frame, therefore the individual items are not shown in the standard output / log file, but are correctly decoded into the json output.

Other meters send the data in a more "standard" format with data-type, unit/scaling and value.

vhamburger, 2024-01-01:
Hey And252 - thanks for the quick reply. Indeed, when I configure the D1 mini to send the telegram to the HA-addon based on this wmbusmeter, it just works. So I guess it must be something with the way they implemented the driver on the D1 mini. 
Anyway - it works now - many thanks for the support!

gllmlbrt, 2024-01-04:
> when I configure the D1 mini to send the telegram to the HA-addon based

I am curious to hear how you would do that, essentially using the D1 mini with CC1101 to the HA add on ?
Any link I could use ?

vhamburger, 2024-01-08:
@gllmlbrt sure - there is this brilliant ESPHome project which allows you to do that:
https://github.com/SzczepanLeon/esphome-components

Actually it should be possible to also process the telegram directly on the D1 - however, im my case this did not work (and still does not work), but you can configure a client component to then forward the telegram via network to the HA add-on (TCP or UDP). This works like a charm since 2 weeks now without any disruption ;)

>clients:
    - name: "wmbusmeters-HA"
      ip_address: "10.0.0.1"
      port: 7227
      
This is exemplary for sending the telegram to the HA add-on (needs to be the IP of the machine where the HA-addon runs and an open port)

gllmlbrt, 2024-01-08:
I did it and it works great. Thank you thanks a lot for the help.

closed 2024-04-08 by github-actions[bot]