wmbusmeters/wmbusmeters #1130: Incorrect decoding of Qundis Q heat 5 US telegrams

kroimon, 2024-01-02:
Happy new year!

I have a Qundis Q heat 5 US (see #695) that was working fine with the following configuration until yesterday:
```
name=HeatMeter
driver=qheat
id=67329263
```

Since yesterday, I guess the meter has entered a mode where it sends more frequently over the day to allow easier remote reading for the first days of the year.
These additional messages have a different format from what it normally sends, which are decoded incorrectly:

[image]

### Example of broken telegram

[Analysis](https://wmbusmeters.org/analyze/4F4465B2639232672304780DFF5F350082508035F0D838D04B64D96A279329699260EB4298F792BACB4CEF3C6CF012793AF7050C80EB3087553C597EE1539E5782EFB78D046D1B0902310C7816640937)

```
[2024-01-02_10:25:14] (wmbusmeters) version: 1.14.0-71-g331c5a4
[2024-01-02_10:25:14] (config) store meter files in: "/config/wmbusmeters/logs/meter_readings"
[2024-01-02_10:25:14] (config) using device: /dev/serial/by-id/usb-Silicon_Labs_WiMOD_iM871A-usb_0275932F-if00-port0:im871a:s1 
[2024-01-02_10:25:14] (config) number of meters: 3
[2024-01-02_10:25:14] (serial) tty /dev/serial/by-id/usb-Silicon_Labs_WiMOD_iM871A-usb_0275932F-if00-port0 can be accessed
[2024-01-02_10:25:14] (serialtty) opened /dev/serial/by-id/usb-Silicon_Labs_WiMOD_iM871A-usb_0275932F-if00-port0 fd 3 (detect im871a)
[2024-01-02_10:25:14] (serialtty) closed /dev/serial/by-id/usb-Silicon_Labs_WiMOD_iM871A-usb_0275932F-if00-port0 (detect im871a)
[2024-01-02_10:25:14] (im871a/im170a) are you there? yes 00103773 im871a firmware: 16
[2024-01-02_10:25:14] Started config im871a[00103773] on /dev/serial/by-id/usb-Silicon_Labs_WiMOD_iM871A-usb_0275932F-if00-port0 listening on s1
[2024-01-02_10:25:14] (im871a) on /dev/serial/by-id/usb-Silicon_Labs_WiMOD_iM871A-usb_0275932F-if00-port0
[2024-01-02_10:25:14] (serialtty) opened /dev/serial/by-id/usb-Silicon_Labs_WiMOD_iM871A-usb_0275932F-if00-port0 fd 3 (im871a)
[2024-01-02_10:25:14] (im871a) get config
[2024-01-02_10:25:14] (im871a) get config completed
[2024-01-02_10:25:14] (im871a) got device id 00103773
[2024-01-02_10:25:14] (main) regular reset of im871a  on /dev/serial/by-id/usb-Silicon_Labs_WiMOD_iM871A-usb_0275932F-if00-port0 will happen every 82800 seconds
[2024-01-02_10:25:14] (im871a) get device info
[2024-01-02_10:25:14] (im871a) device info completed
[2024-01-02_10:25:14] (im871a) device info: type=im871a mode=collector firmware=16 hci=01 uid=00003a19
[2024-01-02_10:25:14] (im871a) set config to set link mode 00
[2024-01-02_10:25:14] (im871a) set config completed

...

[2024-01-02_10:27:03] (meter) HeatMeter: meter detection did not match the selected driver qheat! correct driver is: unknown!
(meter) Not printing this warning again for id: 67329263 mfct: (LSE) Unknown (0xb265) type: Heat meter (0x04) ver: 0x23
[2024-01-02_10:27:03] (meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
[2024-01-02_10:27:03] (meter) to add support for this unknown mfct,media,version combination
[2024-01-02_10:27:03] (meter) created HeatMeter qheat 67329263 not-encrypted
[2024-01-02_10:27:03] (meter) used meter template HeatMeter 67329263 qheat to match 67329263
[2024-01-02_10:27:03] (meter) started meter 3 (HeatMeter 67329263 qheat)
[2024-01-02_10:27:03] (meter) HeatMeter: meter detection did not match the selected driver qheat! correct driver is: unknown!
(meter) Not printing this warning again for id: 67329263 mfct: (LSE) Unknown (0xb265) type: Heat meter (0x04) ver: 0x23
[2024-01-02_10:27:03] (meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
[2024-01-02_10:27:03] (meter) to add support for this unknown mfct,media,version combination
[2024-01-02_10:27:03] (meter) HeatMeter(3) qheat  handling telegram from 67329263
[2024-01-02_10:27:03] (telegram) DLL L=4f C=44 (from meter SND_NR) M=b265 (LSE) A=67329263 VER=23 TYPE=04 (Heat meter) (driver unknown!) DEV=im871a[00103773] RSSI=-66
[2024-01-02_10:27:03] (telegram) TPL CI=78[2024-01-02_10:27:03] 
[2024-01-02_10:27:03] telegram=|4F4465B263923267230478_0DFF5F350082508035F0D838D04B64D96A279329699260EB4298F792BACB4CEF3C6CF012793AF7050C80EB3087553C597EE1539E5782EFB78D046D1B0902310C7816640937|+109
[2024-01-02_10:27:03] (telegram) DLL L=4f C=44 (from meter SND_NR) M=b265 (LSE) A=67329263 VER=23 TYPE=04 (Heat meter) (driver unknown!) DEV=im871a[00103773] RSSI=-66
[2024-01-02_10:27:03] (telegram) TPL CI=78[2024-01-02_10:27:03] 
[2024-01-02_10:27:03] Received telegram from: 67329263
[2024-01-02_10:27:03]           manufacturer: (LSE) Unknown (0xb265)
[2024-01-02_10:27:03]                   type: Heat meter (0x04)
[2024-01-02_10:27:03]                    ver: 0x23
[2024-01-02_10:27:03]                 device: im871a[00103773]
[2024-01-02_10:27:03]                   rssi: -66 dBm
[2024-01-02_10:27:03]                 driver: unknown!
[2024-01-02_10:27:03] telegram=|_4F4465B2639232672304780DFF5F350082508035F0D838D04B64D96A279329699260EB4298F792BACB4CEF3C6CF012793AF7050C80EB3087553C597EE1539E5782EFB78D046D1B0902310C7816640937|+109
```

### Example of correctly decoded telegram

[Analysis](https://wmbusmeters.org/analyze/2E4465B26392326723047A4D0000000C05064203004C0540410300426CFF2C326CFFFF046D071401310C7816640937)

```
[2024-01-01_21:06:01] (meter) HeatMeter: meter detection did not match the selected driver qheat! correct driver is: unknown!
(meter) Not printing this warning again for id: 67329263 mfct: (LSE) Unknown (0xb265) type: Heat meter (0x04) ver: 0x23
[2024-01-01_21:06:01] (meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
[2024-01-01_21:06:01] (meter) to add support for this unknown mfct,media,version combination
[2024-01-01_21:06:01] (meter) HeatMeter(3) qheat  handling telegram from 67329263
[2024-01-01_21:06:01] (telegram) DLL L=2e C=44 (from meter SND_NR) M=b265 (LSE) A=67329263 VER=23 TYPE=04 (Heat meter) (driver unknown!) DEV=im871a[00103773] RSSI=-66
[2024-01-01_21:06:01] (telegram) TPL CI=7a[2024-01-01_21:06:01]  ACC=4d STS=00 CFG=0000 ()[2024-01-01_21:06:01] 
[2024-01-01_21:06:01] telegram=|2E4465B26392326723047A4D000000_0C05064203004C0540410300426CFF2C326CFFFF046D071401310C7816640937|+9256
[2024-01-01_21:06:02] (telegram) DLL L=2e C=44 (from meter SND_NR) M=b265 (LSE) A=67329263 VER=23 TYPE=04 (Heat meter) (driver unknown!) DEV=im871a[00103773] RSSI=-66
[2024-01-01_21:06:02] (telegram) TPL CI=7a[2024-01-01_21:06:02]  ACC=4d STS=00 CFG=0000 ()[2024-01-01_21:06:02] 
[2024-01-01_21:06:02] Received telegram from: 67329263
[2024-01-01_21:06:02]           manufacturer: (LSE) Unknown (0xb265)
[2024-01-01_21:06:02]                   type: Heat meter (0x04)
[2024-01-01_21:06:02]                    ver: 0x23
[2024-01-01_21:06:02]                 device: im871a[00103773]
[2024-01-01_21:06:02]                   rssi: -66 dBm
[2024-01-01_21:06:02]                 driver: unknown!
[2024-01-01_21:06:02] telegram=|_2E4465B26392326723047A4D0000000C05064203004C0540410300426CFF2C326CFFFF046D071401310C7816640937|+9257
```

ene-rgy, 2024-01-26:
I had the same issue with the qundis heatmeter. The heatmeter sends two different messages, one massage was correct and in the other message the value was divied by 10. 

The stupid simple solution was adjusting the mqtt template:
I do a multiply by 10 if the value is less than 1000 ( under value_template )
/homeassistant/wmbusmeters/etc/mqtt_discovery/qheat.json

`    "total_energy_consumption_kwh": {
        "component": "sensor",
        "discovery_payload": {
            "device": {
                "identifiers": [
                    "wmbusmeters_{id}"
                ],
                "manufacturer": "Qundis",
                "model": "{driver}",
                "name": "{name}",
                "sw_version": "{id}"
            },
            "enabled_by_default": true,
            "json_attributes_topic": "wmbusmeters/{name}",
            "device_class": "energy",
            "state_class": "total",
            "name": "total energy consumption",
            "state_topic": "wmbusmeters/{name}",
            "unique_id": "wmbusmeters_{id}_{attribute}",
            "unit_of_measurement": "kWh",
            "value_template": "{% set tmx = value_json.{attribute} | replace('.', '') %}{% if float(tmx) > 1000 %}{{  tmx }}{% else %}{{ ( float(tmx) * 10 ) }}{% endif %}",
            "icon": "mdi:gauge"
        }
    }`

kroimon, 2024-01-26:
@ene-rgy If you look at my example telegrams and diagram, that's not what happens for me. The values are not deterministically off by a certain factor, rather they are decoded completely wrong and are off by millions.

ene-rgy, 2024-01-26:
are both message incorrect or only one? The incorrect mesages are sent are sent from 06:00 until 19:00. 
 I am not sure but it could be possible to apply a filter via home assistant, for too large values.

kroimon, 2024-01-26:
Only the additional messages sent during the day are decoded incorrectly.
This issue is not for finding a workaround, but to highlight the issue and maybe provide help finding a proper solution. Unfortunately, I don't know anything about the encoding so can not really provide more than logs and telegrams.

@ene-rgy can you provide example telegrams for your meter to see if it might be the same format?

ene-rgy, 2024-01-26:
Here are some messages. The auto driver ist not working for my messages. Qundis need to be manually selected.
[Link](https://wmbusmeters.org/analyze/58449344425263673c377807794252636793443c040dff5f3500829a0000f10007c006ffff24650000fe2600000000ff2c154200000000000000000000000000000000000000001c02250636082f02fd170000046d1f0e1a3158449344425263673c377807794252636793443c040dff5f3500829a0000110007c006ffff24650000fe2600000000ff2c154200000000000000000000000000000000000000001c02250636082f02fd170000046d230e1a3158449344425263673c377807794252636793443c040dff5f3500829B0000ef0007c006ffff24650000fe2600000000ff2c154200000000000000000000000000000000000000001c02250636082f02fd170000046d270e1a3158449344425263673c377807794252636793443c040dff5f3500828d0000110007c006ffff21650000fe2600000000ff2c154200000000000000000000000000000000000000001c02250636082f02fd170000046d000d1a31:qheat)

```
58449344425263673C3778_07794252636793443C040DFF5F3500829A0000F10007C006FFFF24650000FE2600000000FF2C154200000000000000000000000000000000000000001C02250636082F02FD170000046D1F0E1A31

58449344425263673C3778_07794252636793443C040DFF5F3500829A0000110007C006FFFF24650000FE2600000000FF2C154200000000000000000000000000000000000000001C02250636082F02FD170000046D230E1A31

58449344425263673C3778_07794252636793443C040DFF5F3500829B0000EF0007C006FFFF24650000FE2600000000FF2C154200000000000000000000000000000000000000001C02250636082F02FD170000046D270E1A31

58449344425263673C3778_07794252636793443C040DFF5F3500828D0000110007C006FFFF21650000FE2600000000FF2C154200000000000000000000000000000000000000001C02250636082F02FD170000046D000D1A31
```

kroimon, 2024-03-27:
Not stale.

weetmuts, 2024-03-27:
So, the broken telegrams contain manufacturer specific data, ie an unknown data format.

Wmbusmeters is all about decoding wmbus-compliant telegrams, like the ones that you have
that are working.

Unfortunately some meters send old style telegrams with their old proprietary format
and then new style telegrams that are wmbus compliant. Heh, this is probably so
that existing reading equipment that only understands the old format
will work with newer meters.

Someone helpfully developed a decoder for the manufacturer specific data, however
this decoder is clearly incomplete! No wonder given the lack of documentation etc.

Unfortunately the telegrams are sent using the same mfct.version.type combo
so there is no way to filter them out using the new adressing functionality. The only fix is to drop 
the code processContent in src/driver_qheat.cc to stop trying to decode the manufacturer specific 
data.

You could however pick the kamheat driver instead. Then it will not use the incomplete reverse
engineerd code in qheat.

weetmuts, 2024-03-27:
@stigger Ping Vyacheslav! Can you give any input on this?

stigger, 2024-03-27:
Not a lot, unfortunately. The reason why I added the decoding of the proprietary telegrams is that they are being sent more often: once every 112 seconds vs every 7.5 minutes for the compliant telegrams.

I was able to extract some values from the proprietary telegrams by comparing them with the known values from the regular telegrams, as well as comparing the proprietary telegrams against each other, but I don't actually know anything about the format.

@kroimon Could you post all telegrams you receive over a ~8 minute period, so it includes 2 properly decoded telegrams, as well as all the proprietary telegrams between them?

kroimon, 2024-04-14:
I am currently using the `qheat` driver because I thought it would match my meter (picture in #695) and worked fine all the time.

@stigger
My meter normally only sends six telegrams every day on S1 - roughly every four hours.
[JSON and raw telegrams from 2024-04-10](https://github.com/wmbusmeters/wmbusmeters/files/14972053/wmbusmeters.telegrams.2024-04-10.txt)

However, during the first eight weeks of the year (from Monday to Saturday, 06:00-18:00 local time), it sends additional telegrams in between that seem to have a different format and throws off the decoder.
[JSON from 2024-01-01 and 2024-01-02](https://github.com/wmbusmeters/wmbusmeters/files/14972081/wmbusmeters.json.2024-01-01.txt)
[Raw telegrams from 2024-01-01 and 2024-01-02](https://github.com/wmbusmeters/wmbusmeters/files/14972082/wmbusmeters.telegrams.2024-01-01.txt)
These logs contain both correctly decoded packets (where `total_energy_consumption_kwh` is between 3413 and 3433 kWh) and the garbage visualized in the original issue post.

Please note that I manually filtered the above logs for telegrams containing my serial number to reduce clutter from nearby flats. There are a few telegrams in the log that seem to contain my serial number but can not be decoded, but those might come from a repeater (e.g. at 2024-01-01 05:16:51).

eporsche, 2024-04-18:
I also get a warning using the  qheat driver.

Started config rtlwmbus listening on none using CMD(nc -lk 9022)
(meter) qundis: meter detection did not match the selected driver qheat! correct driver is: unknown!
(meter) Not printing this warning again for id: **** mfct: (QDS) Qundis, Germany (0x4493) type: Radio converter (meter side) (0x37) ver: 0x3c
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination

I am using it on a "Q heat 5.5 US R". Values seem to be populated correctly to mqtt though.
However I also have this issue:

"The heatmeter sends two different messages, one massage was correct and in the other message the value was divied by 10."

https://qundis.de/fileadmin/user_upload/01_Produkte/02_Qheat_W%C3%A4rmez%C3%A4hler/02_PDF/01_DE/QUNDIS_Datenblatt_Qheat_5.5_US_R_2023_dt.pdf

Happy to help if we can make this driver better: :)

closed 2024-07-18 by github-actions[bot]

mredd, 2024-09-29:
I'm seeing similar decoding issues on qheat 5.5 US devices. As this issues has been marked as stale, can one of the last commenters put the found solution in here? Thank you very much!

Maybe @eporsche @kroimon or @stigger ?

eporsche, 2024-09-29:
@mredd no solution.. only a workaround.. I set the last recieved value into a helper variable - the next value will then be compared with the helper.. if its smaller it will be discarded.

mredd, 2024-09-29:
Thank you for the swift response @eporsche even though it is not what I hoped for. :-/

eporsche, 2025-03-02:
@mredd Solution provided by @gotsila in https://github.com/wmbusmeters/wmbusmeters/issues/1347