wmbusmeters/wmbusmeters #1131: Diehl Gasmeter with unknown driver

Timmo3, 2024-01-03:
Hi there, 
I connected a Diehl Gasmeter but I got the "unknown diver" error. 
[image]

I assumed the "izar" driver but that did fail. Instead the https://wmbusmeters.org/ suggested to use the "apator08" driver, but that did not work either. 

https://wmbusmeters.org/analyze/3644a511640010253837722550100593158103d90020052f2f03741602000c137034220302fd74ee0f2f2f2f2f2f2f2f2f2f2f2f2f2f2f 

Here are the last telegrams with the izar driver:

telegram=|3644A511640010253837722550100593158103DD0020052F2F_03744D02000C137034220302FD74EE0F2F2F2F2F2F2F2F2F2F2F2F2F2F2F|+112
(izar) Decoding PRIOS data failed. Ignoring telegram.
Unable to connect (Invalid arguments provided.).
(shell) /bin/sh exited with non-zero return code: 1
telegram=|3644A511640010253837722550100593158103DE0020052F2F_03746502000C137034220302FD74EE0F2F2F2F2F2F2F2F2F2F2F2F2F2F2F|+135
(izar) Decoding PRIOS data failed. Ignoring telegram.
Unable to connect (Invalid arguments provided.).
(shell) /bin/sh exited with non-zero return code: 1
telegram=|3644A511640010253837722550100593158103DF0020052F2F_03746F02000C137034220302FD74EE0F2F2F2F2F2F2F2F2F2F2F2F2F2F2F|+146
(izar) Decoding PRIOS data failed. Ignoring telegram.
Unable to connect (Invalid arguments provided.).
(shell) /bin/sh exited with non-zero return code: 1
telegram=|3644A511640010253837722550100593158103E00020052F2F_03747902000C137034220302FD74EE0F2F2F2F2F2F2F2F2F2F2F2F2F2F2F|+156
(izar) Decoding PRIOS data failed. Ignoring telegram.
Unable to connect (Invalid arguments provided.).
(shell) /bin/sh exited with non-zero return code: 1
telegram=|3644A511640010253837722550100593158103E10020052F2F_03748F02000C137034220302FD74EE0F2F2F2F2F2F2F2F2F2F2F2F2F2F2F|+178
(izar) Decoding PRIOS data failed. Ignoring telegram.
Unable to connect (Invalid arguments provided.).
(shell) /bin/sh exited with non-zero return code: 1
telegram=|3644A511640010253837722550100593158103E20020052F2F_03749B02000C137034220302FD74EE0F2F2F2F2F2F2F2F2F2F2F2F2F2F2F|+190
(izar) Decoding PRIOS data failed. Ignoring telegram.
Unable to connect (Invalid arguments provided.).
(shell) /bin/sh exited with non-zero return code: 1
telegram=|3644A511640010253837722550100593158103E30020052F2F_0374A502000C137034220302FD74EE0F2F2F2F2F2F2F2F2F2F2F2F2F2F2F|+200
(izar) Decoding PRIOS data failed. Ignoring telegram.
Unable to connect (Invalid arguments provided.).
(shell) /bin/sh exited with non-zero return code: 1
telegram=|3644A511640010253837722550100593158103E40020052F2F_0374BA02000C137034220302FD74EE0F2F2F2F2F2F2F2F2F2F2F2F2F2F2F|+220
(izar) Decoding PRIOS data failed. Ignoring telegram.
Unable to connect (Invalid arguments provided.).
(shell) /bin/sh exited with non-zero return code: 1
telegram=|3644A511640010253837722550100593158103E50020052F2F_0374C502000C137034220302FD74EE0F2F2F2F2F2F2F2F2F2F2F2F2F2F2F|+232
(izar) Decoding PRIOS data failed. Ignoring telegram.
Unable to connect (Invalid arguments provided.).
(shell) /bin/sh exited with non-zero return code: 1
telegram=|3644A511640010253837722550100593158103E60020052F2F_0374D102000C137034220302FD74EE0F2F2F2F2F2F2F2F2F2F2F2F2F2F2F|+244
(izar) Decoding PRIOS data failed. Ignoring telegram.
Unable to connect (Invalid arguments provided.).
(shell) /bin/sh exited with non-zero return code: 1
telegram=|3644A511640010253837722550100593158103E70020052F2F_0374E602000C137034220302FD74EE0F2F2F2F2F2F2F2F2F2F2F2F2F2F2F|+265
(izar) Decoding PRIOS data failed. Ignoring telegram.
Unable to connect (Invalid arguments provided.).
(shell) /bin/sh exited with non-zero return code: 1

Could you check that problem?

Thanks
Timmo

weetmuts, 2024-02-12:
I have added a new elster driver for your gas meter. This device is a standard wmbus radio converter, the izar is irrelevant. Yay!

https://wmbusmeters.org/analyze/3644a511640010253837722550100593158103d90020052f2f03741602000c137034220302fd74ee0f2f2f2f2f2f2f2f2f2f2f2f2f2f2f

This is the first new text based driver, ie it is stored here:
https://github.com/wmbusmeters/wmbusmeters-drivers/blob/main/wmbusmeters.drivers.d/elster.xmq

The driver is built into wmbusmeters, but you can override it by copying elster.xmq into for example 
`/etc/wmbusmters.drivers.d`  or load it when testing:

```
wmbusmeters --driver=elster.xmq --analyze 3644a511640010253837722550100593158103d90020052f2f03741602000c137034220302fd74ee0f2f2f2f2f2f2f2f2f2f2f2f2f2f2f
```

Please test!

Timmo3, 2024-02-13:
Hi weetmuts, 

great, Thanks!

Here is my test with your elster.xmq:

c8a990ad-wmbusmeters-ha-addon-edge:/wmbusmeters# ./wmbusmeters --driver=elster.xmq --analyze 3644a511640010253837722550100593158103d90020052f2f03741602000c137034220302fd74ee0f2f2f2f2f2f2f2f2f2f2f2f2f2f2f

Auto driver  : elster
Best driver  : elster 07/09
Using driver : elster 00/00
000   : 36 length (54 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : a511 dll-mfct (DME)
004   : 64001025 dll-id (25100064)
008   : 38 dll-version
009   : 37 dll-type (Radio converter (meter side))
010   : 72 tpl-ci-field (EN 13757-3 Application Layer (long tplh))
011   : 25501005 tpl-id (05105025)
015   : 9315 tpl-mfct (ELS)
017   : 81 tpl-version
018   : 03 tpl-type (Gas meter)
019   : d9 tpl-acc-field
020   : 00 tpl-sts-field (OK)
021   : 2005 tpl-cfg 0520 (AES_CBC_IV nb=2 cntn=0 ra=0 hc=0 )
023   : 2f2f already decrypted check bytes
025   : 03 dif (24 Bit Integer/Binary Instantaneous value)
026   : 74 vif (Actuality duration seconds)
027 C!: 160200 ("actuality_duration_s":534)
030   : 0C dif (8 digit BCD Instantaneous value)
031   : 13 vif (Volume l)
032 C!: 70342203 ("total_m3":3223.47)
036   : 02 dif (16 Bit Integer/Binary Instantaneous value)
037   : FD vif (Second extension FD of VIF-codes)
038   : 74 vife (Reserved)
039 C?: EE0F
041   : 2F skip
042   : 2F skip
043   : 2F skip
044   : 2F skip
045   : 2F skip
046   : 2F skip
047   : 2F skip
048   : 2F skip
049   : 2F skip
050   : 2F skip
051   : 2F skip
052   : 2F skip
053   : 2F skip
054   : 2F skip

{
    "media":"gas",
    "meter":"elster",
    "name":"",
    "id":"05105025",
    "actuality_duration_s":534,
    "total_m3":3223.47,
    "timestamp":"2024-02-13T17:28:35Z"
}
c8a990ad-wmbusmeters-ha-addon-edge:/wmbusmeters# 
...

I also tried to use the file in the lastest HA addon 1.14.0-122, but there I got this output, where the elster driver cannot be found:

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/nginx.sh
cont-init: info: /etc/cont-init.d/nginx.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun nginx (no readiness notification)
s6-rc: info: service legacy-services successfully started
[18:31:45] INFO: CONFIG_CONF ...
[18:31:45] INFO: {"device":"auto:t1","donotprobe":"/dev/ttyAMA0","format":"json","logfile":"/dev/stdout","loglevel":"normal","logtelegrams":false,"shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[18:31:45] INFO: CONFIG_METERS ...
[18:31:45] INFO: [{"driver":"elster","id":"05105025","key":"FD0FB427D77737FE0C2DDFD17F6E3E42","name":"gaszaehler"}]
[18:31:45] INFO: Syncing wmbusmeters configuration ...
[18:31:45] INFO: Registering meters ...
[18:31:45] INFO: Adding meter-0001 ...
[18:31:45] INFO: Generating MQTT configuration ... 
[18:31:45] ERROR: Got unexpected response from the API:
Service not enabled
[18:31:45] INFO: Broker  will be used.
[18:31:45] INFO: 
MQTT Discovery ...
[18:31:46] INFO:  Adding meter: gaszaehler ...
[18:31:46] INFO:   File /config/wmbusmeters/etc/mqtt_discovery/elster.json not found.
[18:31:46] INFO: MQTT Discovery cleanup...
[18:31:47] INFO: Starting web configuration service.
[18:31:47] INFO: Running wmbusmeters ...
[18:31:47] INFO: Starting Nginx...
2024/02/13 18:31:47 [notice] 84#84: using the "epoll" event method
2024/02/13 18:31:47 [notice] 84#84: nginx/1.24.0
2024/02/13 18:31:47 [notice] 84#84: OS: Linux 6.1.74-haos
2024/02/13 18:31:47 [notice] 84#84: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/02/13 18:31:47 [notice] 84#84: start worker processes
2024/02/13 18:31:47 [notice] 84#84: start worker process 231
Started auto im871a[00103514] on /dev/ttyUSB3 listening on t1
Unable to connect (Invalid arguments provided.).
(shell) /bin/sh exited with non-zero return code: 1
Unable to connect (Invalid arguments provided.).
(shell) /bin/sh exited with non-zero return code: 1
...

Maybe I'm doing here something wrong...

Best regards
Timmo

closed 2024-05-15 by github-actions[bot]