wmbusmeters/wmbusmeters #1133: 169 Mhz N mode support

mauriziozito01, 2024-01-08:
HI
I live in Italy and here the meters only go at 169MHz
I tried but I only get the message:
telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.

I saw that there is the possibility of using amb3665-M there but it really costs a lot of money
I hope there will be support for 169MHz soon

I'm sorry for my english

thanks

Maurizio

MedaiP90, 2024-01-15:
How did you configure the app to use `N1` link mode required for 169MHz? When we try the app crashing whit the error: `Not a valid device "/dev/ttyUSB0:amb3665:n2"`

mauriziozito01, 2024-01-15:
Hi

 

I was in the testing phase and ran this command

 

wmbusmeters --debug --t1 rtlwmbus:169.44M

 

It worked and found telegrams but told me they were incorrect.

Then reading I discovered that 169Mhz does not work with y1 but in n mode it is not yet developed

 

Thanks

 

Maurizio

MedaiP90, 2024-01-15:
> Hi I was in the testing phase and ran this command wmbusmeters --debug --t1 rtlwmbus:169.44M It worked and found telegrams but told me they were incorrect. Then reading I discovered that 169Mhz does not work with y1 but in n mode it is not yet developed Thanks Maurizio 

I’m not a developer of this project, I was interested on how you started the app because I had a problem connecting to a 169Mhz dongle (the amber 3665) using the `T1` link mode, maybe you discovered some trick. I’ve deleted the previous response because I figured out that 169Mhz devices use the N1 mode that in this app corresponds to the code `n1a` or `n1f`, maybe this can help you too.

`wmbusmeters --debug --n1a rtlwmbus:169.44M`
(Don’t know if it is correct, because I use the docker version of the wmbusmeters app)

mauriziozito01, 2024-01-15:
I have made several attempts

 

consider that my dongle is rtl_sdr

in my case this:

rtlwmbus:169.44M

it is the device

otherwise you can try like this

device=rtlwmbus:rtl_sdr -f 434.47M -s 1.6e6 - 2>/dev/null | rtl_wmbus

 

I looked around a bit

weetmuts, 2024-02-12:
The N-modes (n1a-n1f modes) are supported by the Amber 3665-M (amb3665) and with a big antenna.
The encoding is very different from the other encodings, and so far no-one has managed to decode it properly.

I have rtl-sdr recordings that I can share if you want to have a go at it! :-)

weetmuts, 2024-02-12:
So wmbusmeters can both send and receive n1a-n1f telegrams if you have the amb3665.
(The other modes T1,S1,C1 do not work for 169MHz, but they do work for 433MHz.)

closed 2024-05-13 by github-actions[bot]

jetmcquack, 2024-05-29:
can you please reopen this?
i'm also working to read N-band telegrams using rtl-sdr hardware. I can see tickings in the 169mhz band using SDR++ with my antenna. i don't know if it's enough.