wmbusmeters/wmbusmeters #1136: [Documentation] Using RTL-SDR in unprivileged mode

kernkraftzwerg, 2024-01-10:
### Type of request

Enhancement

### OS version

Linux

### wmbusmeters version

all

### Your message goes here

It's possible to run the docker container in unprivileged mode with "--device-cgroup-rule":
`docker run \
    --name=wmbusmeters \
    --device-cgroup-rule='c 189:* rw' --device-cgroup-rule='c 212:* rw'
    --restart=always \
    -v /etc/localtime:/etc/localtime:ro \
    -v /opt/wmbusmeters:/wmbusmeters_data \
    -v /dev/:/dev/ \
    wmbusmeters/wmbusmeters`

BIBOLV, 2024-02-08:
Thanks for pointing that out and you are correct, but I`m not certain that all devices would be under 189 char device major number and I do not have many ones that I could test and verify.
What device you have?

kernkraftzwerg, 2024-04-10:
Sorry, I didn't see the question:
I'm using the RTL2832U (ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T)

closed 2024-07-11 by github-actions[bot]