wmbusmeters/wmbusmeters #1144: Adding support for Apator E-ITN 40

kuchar91, 2024-01-14:
### Type of request

New device support

### Meter description

Please add support for driver Apator E-ITN 40 - Heat Cost Allocator. 

Link do description: https://www.apator.com/nasze-rozwiazania/woda-i-cieplo/podzielniki-kosztow-ogrzewania/e-itn-40 

Room 1:  
Read value: 303
Numer: 00019913

40440106139901000908A00000A10081016B013F00A2004E00000000000000000000000000000018001E0007007800100002002F01C700000000002D3006000000

40440106139901000908A00000A10081016B013F00A2004E00000000000000000000000000000018001E0007007800100002002F01C700000000002D3006000000

Room 2: 
Read value: 192
Number: 00019914

40440106149901000908A00000A100B201AF01150064004700000000000000000000000000000021004E006700930015000000C0007E01000000002D3006000000

40440106149901000908A00000A100B201AF01150064004700000000000000000000000000000021004E006700930015000000C0007E01000000002D3006000000

40440106149901000908A00000A100B201AF01150064004700000000000000000000000000000021004E006700930015000000C0007E01000000002D3006000000

Room 3: 
Read value: 109
Number: 00019915

40440106159901000908A00000A1009C019501240023002600000000000000000000000000000006002600070049000E0000006D008A00000000002D3006000000

40440106159901000908A00000A1009C019501240023002600000000000000000000000000000006002600070049000E0000006D008A00000000002D3006000000

40440106159901000908A00000A1009C019501240023002600000000000000000000000000000006002600070049000E0000006D008A00000000002D3006000000

### Logged telegrams from meter in decrypted format

```shell
Room 1:  
Read value: 303
Numer: 00019913

40440106139901000908A00000A10081016B013F00A2004E00000000000000000000000000000018001E0007007800100002002F01C700000000002D3006000000

40440106139901000908A00000A10081016B013F00A2004E00000000000000000000000000000018001E0007007800100002002F01C700000000002D3006000000

Room 2: 
Read value: 192
Number: 00019914

40440106149901000908A00000A100B201AF01150064004700000000000000000000000000000021004E006700930015000000C0007E01000000002D3006000000

40440106149901000908A00000A100B201AF01150064004700000000000000000000000000000021004E006700930015000000C0007E01000000002D3006000000

40440106149901000908A00000A100B201AF01150064004700000000000000000000000000000021004E006700930015000000C0007E01000000002D3006000000

Room 3: 
Read value: 109
Number: 00019915

40440106159901000908A00000A1009C019501240023002600000000000000000000000000000006002600070049000E0000006D008A00000000002D3006000000

40440106159901000908A00000A1009C019501240023002600000000000000000000000000000006002600070049000E0000006D008A00000000002D3006000000

40440106159901000908A00000A1009C019501240023002600000000000000000000000000000006002600070049000E0000006D008A00000000002D3006000000
```

Alcazer, 2024-02-23:
There is a reading value where I marked in bold, is the last billing period correct?

> 
> Room 1 - 00019913 - 303 current billing period, 199 last billing period
> 4044 **0106** (APA) **13990100** (00019913) 09 (Version) 08 (Heat Cost Allocator) A00000A10081016B013F00A2004E00000000000000000000000000000018001E000700780010000200 **2F01** (303) **C700** (199) 000000002D3006000000
> 
> Room 2 - 00019914 - 192 current billing period, 382 last billing period
> 4044 **0106** (APA) **14990100** (00019914) 09 (Version) 08 (Heat Cost Allocator) A00000A100B201AF01150064004700000000000000000000000000000021004E006700930015000000 **C000** (192) **7E01** (382) 000000002D3006000000
> 
> Room 3 - 00019915 - 109 current billing period, 138 last billing period
> 4044 **0106** (APA) **15990100** (00019915) 09 (Version) 08 (Heat Cost Allocator) A00000A1009C019501240023002600000000000000000000000000000006002600070049000E000000 **6D00** (109) **8A00** (138) 000000002D3006000000
>

AlesVysocky, 2024-03-19:
@Alcazer Hello, i had this https://github.com/orgs/wmbusmeters/discussions/1141#discussion-6069578 comment in discussion, I have some E-ITN 40 devices, but I don't know how to read it in decrypted format. If I knew, I can share all available information (I can read it with NFC).

kuchar91, 2024-05-19:
Up

SteinerOk, 2024-07-21:
Up

SteinerOk, 2024-09-20:
Up

SteinerOk, 2024-11-30:
Up🥲

SteinerOk, 2025-01-31:
Up🥲

SteinerOk, 2025-04-02:
Up🥲

dawidcieszynski, 2025-05-08:
Hi, I just got mounted few of Apator E-ITN 40 in my apartment and I'm trying to prepare driver for esphome wmbus component, but I'm not so fluent in telegram analysis :) Maybe someone could help with finding these values in telegram data?

here is my telegram data:
40 44 0106 07323800 0D 08 A0000081000000F900**0700**1000000000000000000000000000000000000000000000000000000000001700000000000000A83204000000 

Here's what I read from the display:
u01.04 - Billing period
23 - Consumption in the current billing year
0 - Consumption in the previous billing year
7 - Consumption in the current billing month
1.6 - Consumption for the previous billing month
15.56° SM - some temperature maybe?
18.88° SM - some temperature maybe?
08.05 - current date (2025-05-08)

@AlesVysocky 
> [@Alcazer](https://github.com/Alcazer) Hello, i had this https://github.com/orgs/wmbusmeters/discussions/1141#discussion-6069578 comment in discussion, I have some E-ITN 40 devices, but I don't know [...]

current values can be read from the display by pressing the button :)

Alcazer, 2025-05-08:
> Hi, I just got mounted few of Apator E-ITN 40 in my apartment and I'm trying to prepare driver for esphome wmbus component, but I'm not so fluent in telegram analysis :) Maybe someone could help [...]

@dawidcieszynski We need more data to fully validate it.

```
40
44
0106 (APA)
07323800 (00383207)
0D (Version)
08 (Heat Cost Allocator)
A0000081000000 ??
F900 (249)
0700 (7) Consumption in the current billing month??
1000 (16/10=1.6)
0000000000000000000000000000000000000000000000000000 ??
0000 (0) Consumption in the previous billing year ??
1700 (23) Consumption in the current billing year ??
000000000000A83204000000 ??
```

dawidcieszynski, 2025-05-11:
@Alcazer I got data from other meters
Here's the summary of collected data so far (exact texts displayed on the screen)

telegram | current_hca | previous_hca | current_m_hca | previous_m_hca | temp_room_avg_c | temp_room_avg_c | date
-- | -- | -- | -- | -- | -- | -- | --
40440106073238000D08A0000081000000F90007001000000000000000000000000000000000000000000000000000000000001700000000000000A83204000000 | 23 | 0 SM | 7. | 1.6. SM | ‾15.56 SM | _18.88 SM | 08.05
40440106073238000D08A0000081000000F90007001000000000000000000000000000000000000000000000000000000000001700000000000000AA3206000000 | 23 | 0 SM | 7. | 1.6. SM | ‾15.56 SM | _18.88 SM | 10.05
40440106073238000D08A0000081000000F90007001000000000000000000000000000000000000000000000000000000000001700000000000000AB3200000000 | 23 | 0 SM | 7. | 1.6. SM | ‾15.56 SM | _18.88 SM | 11.05
40440106503238000D08A0000081000000F6006C001D00000000000000000000000000000000000000000000000000000000008900000000000000AB3200000000 | 137 | 0 SM | 1.0.8. | 2.9. SM | ‾15.38 SM | _18.69 SM | 11.05
40440106413238000D08A0000081000000F70001000100000000000000000000000000000000000000000000000000000000000200000000000000AB3200000000 | 2 | 0 SM | 1. | 1. SM | ‾15.44 SM | _18.69 SM | 11.05

dawidcieszynski, 2025-05-12:
With help of GPT found that date is stored as these AB32, AC32 values as number of days from Jan 1st 1989

`function decodeHexDate(hexString) {
  if (hexString.length % 2 !== 0) {
    throw new Error("Nieprawidłowy ciąg hex.");
  }

  const buffer = Buffer.from(hexString, 'hex');

  const daysSinceBase = buffer.readUInt16LE(0);
  const weekday = buffer[2]; // 3. bajt (0 = niedziela)

  const baseDate = new Date(Date.UTC(1989, 0, 1));
  baseDate.setUTCDate(baseDate.getUTCDate() + daysSinceBase);

  const actualWeekday = baseDate.getUTCDay(); // JS: 0 = niedziela

  return {
    date: baseDate.toISOString().split('T')[0],
    decodedWeekday: weekday,
    actualWeekday,
    weekdayMatches: weekday === actualWeekday
  };
}

console.log(decodeHexDate("AB3200000000")); // 11 05 2025, Sun
console.log(decodeHexDate("AC3201000000")); // 12 05 2025, Mon
`

AlesVysocky, 2025-05-13:
@dawidcieszynski Hello Dawid,
what HW setup do you use to read telegrams from E-ITN 40?

Thank you

dawidcieszynski, 2025-05-13:
> [@dawidcieszynski](https://github.com/dawidcieszynski) Hello Dawid, what HW setup do you use to read telegrams from E-ITN 40?
> 
> Thank you
Hi @AlesVysocky 

lolin32_lite + CC1101 868MHZ 

[image]

Alcazer, 2025-05-13:
> [@Alcazer](https://github.com/Alcazer) I got data from other meters Here's the summary of collected data so far (exact texts displayed on the screen) I see that the dot is not a decimal point but rather the way it displays the monthly value.
> 
>  	 	 	telegram	current_hca	previous_hca	current_m_hca	previous_m_hca	temp_room_avg_c	temp_room_avg_c	date	room
> 0x00383207	0101-00383207	u01.04	40440106073238000D08A0000081000000F90007001000000000000000000000000000000000000000000000000000000000001700000000000000A83204000000	23	0 SM	7.	1.6. SM	‾15.56° SM	_18.88° SM	08.05	office
> 0x00383207	0101-00383207	u01.04	40440106073238000D08A0000081000000F90007001000000000000000000000000000000000000000000000000000000000001700000000000000AA3206000000	23	0 SM	7.	1.6. SM	‾15.56° SM	_18.88° SM	10.05	office
> 0x00383207	0101-00383207	u01.04	40440106073238000D08A0000081000000F90007001000000000000000000000000000000000000000000000000000000000001700000000000000AB3200000000	23	0 SM	7.	1.6. SM	‾15.56° SM	_18.88° SM	11.05	office
> 0x00383207	0101-00383207	u01.04	40440106073238000D08A0000081000000F90008001000000000000000000000000000000000000000000000000000000000001800000000000000AC3201000000	24	0 SM	8	1.6. SM	‾15.56° SM	_18.88° SM	12.05	office
> 0x00383241	0101-00383241	u01.04	40440106413238000D08A0000081000000F70001000100000000000000000000000000000000000000000000000000000000000200000000000000AB3200000000	2	0 SM	1.	1. SM	‾15.44° SM	_18.69° SM	11.05	bedroom
> 0x00383241	0101-00383241	u01.04	40440106413238000D08A0000081000000F70001000100000000000000000000000000000000000000000000000000000000000200000000000000AC3201000000	2	0 SM	1.	1. SM	‾15.44° SM	_18.69° SM	12.05	bedroom
> 0x00383250	0101-00383250	u01.04	40440106503238000D08A0000081000000F6006C001D00000000000000000000000000000000000000000000000000000000008900000000000000AB3200000000	137	0 SM	1.0.8.	2.9. SM	‾15.38° SM	_18.69° SM	11.05	livingroom
> 0x00383250	0101-00383250	u01.04	40440106503238000D08A0000081000000F60070001D00000000000000000000000000000000000000000000000000000000008D00000000000000AC3201000000	141	0 SM	1.1.2.	2.9. SM	‾15.38° SM	_18.69° SM	12.05	livingroom
> 0x00383250	0101-00383250	u01.04	40440106503238000D08A0000081000000F60071001D00000000000000000000000000000000000000000000000000000000008E00000000000000AC3201000000	142	0 SM	1.1.3.	2.9. SM	‾15.38° SM	_18.69° SM	12.05	livingroom

@dawidcieszynski Those that I have marked in bold match the data in the table. Do you have the possibility to collect additional data?

40 44 0106 07323800 0D 08 A0 000081000000 F900 **0700 [  7 (current_m_hca)] 1000 [16 (previous_m_hca)]** 00000000000000000000000000000000000000000000000000000000 **1700 [ 23 (current_hca)] 0000 [0 (previous_hca)]** 00000000A832 **[12.05 (date)]** 04000000

40 44 0106 07323800 0D 08 A0 000081000000 F900 **0700 [  7 (current_m_hca)] 1000 [16 (previous_m_hca)]** 00000000000000000000000000000000000000000000000000000000 **1700 [ 23 (current_hca)] 0000 [0 (previous_hca)]** 00000000AA32 **[12.05 (date)]** 06000000

40 44 0106 07323800 0D 08 A0 000081000000 F900 **0700 [  7 (current_m_hca)] 1000 [16 (previous_m_hca)]** 00000000000000000000000000000000000000000000000000000000 **1700 [ 23 (current_hca)] 0000 [0 (previous_hca)]** 00000000AB32 **[11.05 (date)]** 00000000

40 44 0106 07323800 0D 08 A0 000081000000 F900 **0800 [  8 (current_m_hca)] 1000 [16 (previous_m_hca)]** 00000000000000000000000000000000000000000000000000000000 **1800 [ 24 (current_hca)] 0000 [0 (previous_hca)]** 00000000AC32 **[12.05 (date)]** 01000000

40 44 0106 41323800 0D 08 A0 000081000000 F700 **0100 [  1 (current_m_hca)] 0100 [ 1 (previous_m_hca)]** 00000000000000000000000000000000000000000000000000000000 **0200 [  2 (current_hca)] 0000 [0 (previous_hca)]** 00000000AB32 **[11.05 (date)]** 00000000

40 44 0106 41323800 0D 08 A0 000081000000 F700 **0100 [  1 (current_m_hca)] 0100 [ 1 (previous_m_hca)]** 00000000000000000000000000000000000000000000000000000000 **0200 [  2 (current_hca)] 0000 [0 (previous_hca)]** 00000000AC32 **[12.05 (date)]** 01000000

40 44 0106 50323800 0D 08 A0 000081000000 F600 **6C00 [108 (current_m_hca)] 1D00 [29 (previous_m_hca)]** 00000000000000000000000000000000000000000000000000000000 **8900 [137 (current_hca)] 0000 [0 (previous_hca)]** 00000000AB32 **[11.05 (date)]** 00000000

40 44 0106 50323800 0D 08 A0 000081000000 F600 **7000 [112 (current_m_hca)] 1D00 [29 (previous_m_hca)]** 00000000000000000000000000000000000000000000000000000000 **8D00 [141 (current_hca)] 0000 [0 (previous_hca)]** 00000000AC32 **[10.05 (date)]** 01000000

40 44 0106 50323800 0D 08 A0 000081000000 F600 **7100 [113 (current_m_hca)] 1D00 [29 (previous_m_hca)]** 00000000000000000000000000000000000000000000000000000000 **8E00 [142 (current_hca)] 0000 [0 (previous_hca)]** 00000000AC32 **[08.05 (date)]** 01000000

Alcazer, 2025-05-13:
> ### Type of request
> New device support
> 
> ### Meter description
> Please add support for driver Apator E-ITN 40 - Heat Cost Allocator.
> 
> Link do description: https://www.apator.com/nasze-rozwiazania/woda-i-cieplo/podzielniki-kosztow-ogrzewania/e-itn-40
> 
> Room 1: Read value: 303 Numer: 00019913
> 
> 40440106139901000908A00000A10081016B013F00A2004E00000000000000000000000000000018001E0007007800100002002F01C700000000002D3006000000
> 
> 40440106139901000908A00000A10081016B013F00A2004E00000000000000000000000000000018001E0007007800100002002F01C700000000002D3006000000
> 
> Room 2: Read value: 192 Number: 00019914
> 
> 40440106149901000908A00000A100B201AF01150064004700000000000000000000000000000021004E006700930015000000C0007E01000000002D3006000000
> 
> 40440106149901000908A00000A100B201AF01150064004700000000000000000000000000000021004E006700930015000000C0007E01000000002D3006000000
> 
> 40440106149901000908A00000A100B201AF01150064004700000000000000000000000000000021004E006700930015000000C0007E01000000002D3006000000
> 
> Room 3: Read value: 109 Number: 00019915
> 
> 40440106159901000908A00000A1009C019501240023002600000000000000000000000000000006002600070049000E0000006D008A00000000002D3006000000
> 
> 40440106159901000908A00000A1009C019501240023002600000000000000000000000000000006002600070049000E0000006D008A00000000002D3006000000
> 
> 40440106159901000908A00000A1009C019501240023002600000000000000000000000000000006002600070049000E0000006D008A00000000002D3006000000
> 
> ### Logged telegrams from meter in decrypted format
> Room 1:  
> Read value: 303
> Numer: 00019913
> 
> 40440106139901000908A00000A10081016B013F00A2004E00000000000000000000000000000018001E0007007800100002002F01C700000000002D3006000000
> 
> 40440106139901000908A00000A10081016B013F00A2004E00000000000000000000000000000018001E0007007800100002002F01C700000000002D3006000000
> 
> Room 2: 
> Read value: 192
> Number: 00019914
> 
> 40440106149901000908A00000A100B201AF01150064004700000000000000000000000000000021004E006700930015000000C0007E01000000002D3006000000
> 
> 40440106149901000908A00000A100B201AF01150064004700000000000000000000000000000021004E006700930015000000C0007E01000000002D3006000000
> 
> 40440106149901000908A00000A100B201AF01150064004700000000000000000000000000000021004E006700930015000000C0007E01000000002D3006000000
> 
> Room 3: 
> Read value: 109
> Number: 00019915
> 
> 40440106159901000908A00000A1009C019501240023002600000000000000000000000000000006002600070049000E0000006D008A00000000002D3006000000
> 
> 40440106159901000908A00000A1009C019501240023002600000000000000000000000000000006002600070049000E0000006D008A00000000002D3006000000
> 
> 40440106159901000908A00000A1009C019501240023002600000000000000000000000000000006002600070049000E0000006D008A00000000002D3006000000

According to the latest information, the details of this data are below. Can you collect the latest data for these meters? @kuchar91 

40 44 0106 13990100 09 08 A0 0000A1008101 6B01 **3F00 [63 (current_m_hca)] A200 [162 (previous_m_hca)]** 4E00000000000000000000000000000018001E000700780010000200 **2F01 [303 (current_hca)] C700 [199 (previous_hca)]** 00000000 **2D30 [13.01.24 (date)]** 06000000

40 44 0106 14990100 09 08 A0 0000A100B201 AF01 **1500 [21 (current_m_hca)] 6400 [100 (previous_m_hca)]** 4700000000000000000000000000000021004E006700930015000000 **C000 [192 (current_hca)] 7E01 [382 (previous_hca)]** 00000000 **2D30 [13.01.24 (date)]** 06000000

40 44 0106 15990100 09 08 A0 0000A1009C01 9501 **2400 [36 (current_m_hca)] 2300 [ 35 (previous_m_hca)]** 2600000000000000000000000000000006002600070049000E000000 **6D00 [109 (current_hca)] 8A00 [138 (previous_hca)]** 00000000 **2D30 [13.01.24 (date)]** 06000000

Date; 2D30 (13.01)
```
0011 0000 0010 1101
             0 1101 D -> 13
        0 001       M -> 01
  11 000            Y -> 24
```

dawidcieszynski, 2025-05-14:
It seems that one of the temperature can also be decoded:
F900 = 249 -> 249/16 = 15.5625 (displayed ‾15.56° SM)
F700 = 247 -> 247/16 = 15.4375 (displayed ‾15.44° SM)
F600 = 246 -> 246/16 = 15.375 (displayed ‾15.38° SM)

from @kuchar91 are hot but seems correct
6B01 = 363 ->22,6875
AF01 = 431 ->26,9375
9501 = 405 ->25,3125

kuchar91, 2025-11-03:
any news on that? @dawidcieszynski @Alcazer

kuchar91, 2025-11-03:
I have the same board - Lilygo T3s3 Lora.