wmbusmeters/wmbusmeters #1149: rtl_433 version in docker container quite old

pimlie, 2024-01-20:
### Type of request

Bug

### OS version

amd64

### wmbusmeters version

latest

### Your message goes here

The changes in PR https://github.com/wmbusmeters/wmbusmeters/pull/1121 by @BIBOLV to use rtl_433 from the alpine repo instead of building it from source are not really a step forward unfortunately. The alpine repo contains version `rtl_433 version v3.18.0_rc2-187-gad5d833beee branch master at 202305031537` which is, given the amount of changes to rtl_433, quite old.

So maybe it would be better to revert the Docker image to just build rtl_433 from source. F.e. version 202305031537 only supports 223 protocols while rtl_433 currently supports 251.

BIBOLV, 2024-02-08:
Rollback done.

BIBOLV, 2024-02-14:
@pimlie - all good and we can close this ticket?