wmbusmeters/wmbusmeters #1151: Not able to use new RTL-SDR v4 dongle in docker container

lukasz-burak, 2024-01-22:
### Type of request

Enhancement

### OS version

Raspberry Pi 5

### wmbusmeters version

Latest Image

### Your message goes here

I'm having issues running wmbusmeters either in standalone Docker container or Home Assistant add-on with new RTL-SDR v4. It gets detected but does not read any data.
I've done some troubleshooting and it seems that the underlaying Alpine os is to old to support that brand new stick.

I was able to make it work by building my own docker image from scratch using the most recent Alpine version and it seems to work just fine so if that's possible, can you bump that version in standalone Docker image as well as in Home Assistant Add-ons?

lukasz-burak, 2024-01-23:
@weetmuts I think new alpine image that you use has been released (https://hub.docker.com/r/asymworks/multiarch-alpine/tags) so this should be just matter of building fresh images.

Can you do that?

weetmuts, 2024-01-24:
Can you switch to the edge of the ha-addon?

lukasz-burak, 2024-01-24:
@weetmuts tried that, it seems though that it uses different base OS image than standalone Docker image and it does not work.
Base OS for ha-addon docker image would need to be updated to at newest alpine 3.19 (since this version contains proper rtl-sdr drivers) but not sure if that's possible.

weetmuts, 2024-01-24:
The latest edge docker should have been built with asymworks multiarch alpine latest, today appros 5 hours ago.
https://github.com/wmbusmeters/wmbusmeters-ha-addon/actions/runs/7639570647

The first change was this, supposedly to get the latest rtl-sdr....
https://github.com/wmbusmeters/wmbusmeters/commit/3d1319bd788a4cb537abc4cfbb9047438a789b1f

@BIBOLV any insights?

BIBOLV, 2024-01-24:
From latest build we can [see](https://github.com/wmbusmeters/wmbusmeters/actions/runs/7639567361/job/20812832617#step:7:221) that for docker latest alpine version is being used with latest rtlsdr version - @lukasz-burak, can you confirm that it latest docker container works for you?
For ha-addon for some reason previous versions is being [used](https://github.com/wmbusmeters/wmbusmeters-ha-addon/actions/runs/7639570647/job/20812848386#step:4:426), I will need to investigate reason for that.

lukasz-burak, 2024-01-24:
Yes, standalone docker container works fine, so it might be an issue with older RTL-sdr lib version used in ha-addon.

BIBOLV, 2024-01-24:
Did a bit of digging and remembered - base OS for addon is being controlled by addon itself 😄, meaning we are using latest version of HA addon base image and current one is 3.18. If we look when previous version was [synced](https://github.com/home-assistant/builder/commit/8e9a0ec8e3b0560ad89ff3e1ef8e1bd87443a68c) - it was one month after release, so it should have been done already. 
As this is out of our control, only thing that we can do is also build rtlsdr from source for both docker and HA.

BIBOLV, 2024-02-14:
Now both ha and docker in stable versions have rtl_sdr built from source - please test if everything works.

closed 2024-05-16 by github-actions[bot]