wmbusmeters/wmbusmeters #1159: Wired mbus

kzajac83, 2024-01-27:
### Type of request

Question

### HAOS version

Home Assistant OS 11.4

### wmbusmeters version

1.14.0

### Your message goes here

Hello

This addon work also for wired mbus connected via gateway mbus -> usb and visible under eg. /dev/ttyUSB0 ?

Or no, this addon is for wireless mbus (wmbs) only?

weetmuts, 2024-01-27:
Should work fine with mbus as well. 
The device must be specified as this:

device=MYBUS=/dev/ttyUSB1:mbus:2400

Then each meter must have a reference to the bus (ie MYBUS or whatever you want to call it.)
name=MyTapWater
id=12345678
driver=piigth:mbus
pollinterval=60s

Where piigth is a driver name for your meter.

//Fredrik

kzajac83, 2024-01-27:
Thank you Frederick, something start but only partially, my log is:

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/nginx.sh
cont-init: info: /etc/cont-init.d/nginx.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun nginx (no readiness notification)
s6-rc: info: service legacy-services successfully started
[00:04:52] INFO: CONFIG_CONF ...
[00:04:52] INFO: {"device":"MAIN=/dev/ttyUSB0:mbus:2400","donotprobe":"/dev/ttyAMA0","format":"json","logfile":"/dev/stdout","loglevel":"normal","logtelegrams":false,"shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[00:04:52] INFO: CONFIG_METERS ...
[00:04:52] INFO: [{"driver":"apator172:MAIN:mbus","id":"00051728","key":"NOKEY","name":"AT-MBUS-01"}]
[00:04:52] INFO: Syncing wmbusmeters configuration ...
[00:04:52] INFO: Registering meters ...
[00:04:52] INFO: Adding meter-0001 ...
[00:04:52] INFO: Generating MQTT configuration ... 
[00:04:53] INFO: Broker core-mosquitto will be used.
[00:04:53] INFO: 
MQTT Discovery ...
tar: wmbusmeters-master/ha-addon/mqtt_discovery: not found in archive
[00:04:54] INFO:  Adding meter: AT-MBUS-01 ...
[00:04:54] INFO:   File /config/wmbusmeters/etc/mqtt_discovery/apator172:MAIN:mbus.json not found.
[00:04:54] INFO: MQTT Discovery cleanup...
[00:04:55] INFO: Starting web configuration service.
[00:04:55] INFO: Running wmbusmeters ...
(meter) AT-MBUS-01 00051728 needs polling but has no pollinterval set!
Started config MAIN=mbus on /dev/ttyUSB0
[00:04:56] INFO: Starting Nginx...
2024/01/28 00:04:56 [notice] 83#83: using the "epoll" event method
2024/01/28 00:04:56 [notice] 83#83: nginx/1.24.0
2024/01/28 00:04:56 [notice] 83#83: OS: Linux 6.1.71-haos
2024/01/28 00:04:56 [notice] 83#83: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/01/28 00:04:56 [notice] 83#83: start worker processes
2024/01/28 00:04:56 [notice] 83#83: start worker process 241
172.30.32.2 - - [28/Jan/2024:00:04:57 +0100] "GET / HTTP/1.1" 200 1302 "http://192.168.15.20:8123/c8a990ad_wmbusmeters-ha-addon" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/120.0.0.0 Safari/537.36"
```

And my questions is:
1. What mean "...needs polling but has no pollinterval set.."?  Unortuneteller general in MQTT not appear new device.
2. Have Aparot vendor mbus device AT-MBUS-01, in repositories in src found drivers for apator:
  - aparot
  - apator162
  - apator172

Do you know what I should use?  Try all 3. but result for all the same.

General I see communication and especially commands + response using MBSheet application so I can help build better driver. If something need I can deliver communication data.

avandorp, 2024-03-12:
Did you set  pollinterval=60s in your meter configuration?

kzajac83, 2024-03-12:
Not set but in meantime continue topic under other subject:

https://github.com/wmbusmeters/wmbusmeters/issues/435#issuecomment-1980727988

Therefore current topic can be closed, is need build new driver and today create request (hope provide all need data)

https://github.com/wmbusmeters/wmbusmeters/issues/1211#issue-2181599839

closed 2024-03-12 by kzajac83

kzajac83, 2024-03-14:
@avandorp maybe can help and tell me where HA addon keep wmbusmeters file?  Cannot found, compile new driver, general working but how update addon with my driver? 

https://github.com/wmbusmeters/wmbusmeters/issues/1211#issuecomment-1996187253

avandorp, 2024-03-14:
I don't know the HA addon (I use MQTT). But from looking at the source code you would configure it directly in the add-on configuration (instead of having a wmbusmeters.conf and a file per meter). If the add-ons wmbusmeter is too old you might have to build your own: https://github.com/wmbusmeters/wmbusmeters-ha-addon

avandorp, 2024-03-14:
> #1211 (comment)

It's a docker image. The docker file defines what is installed where inside
the image.

kzajac83, 2024-03-15:
Now understood.  When login into SSH I'm in some container, because see only config of HA and addons then this is HA container.

But when go to physical machine then see CLI of HAOS but this is some overlap on BASH/Linux system. Now need understood this CLI....