wmbusmeters/wmbusmeters #1163: No longer working after update

jackwilsdon, 2024-01-30:
### Type of request

Bug

### HAOS version

11.4

### wmbusmeters version

1.14.0-101-1

### Your message goes here

After updating from 1.14.0-77 to 1.14.0-101-1, the wmbusmeters addon no longer reports water meter readings.

<details>
<summary>Logs from 1.14.0-77</summary>

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/nginx.sh
cont-init: info: /etc/cont-init.d/nginx.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun nginx (no readiness notification)
s6-rc: info: service legacy-services successfully started
[17:41:14] INFO: CONFIG_CONF ...
[17:41:14] INFO: {"device":"rtlwmbus","format":"json","logfile":"/dev/stdout","loglevel":"normal","logtelegrams":false,"shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[17:41:14] INFO: CONFIG_METERS ...
[17:41:14] INFO: [{"name":"Water Meter","driver":"izar","id":"<my_id>","key":"NOKEY"}]
[17:41:14] INFO: Syncing wmbusmeters configuration ...
[17:41:15] INFO: Registering meters ...
[17:41:15] INFO: Adding meter-0001 ...
[17:41:15] INFO: Generating MQTT configuration ... 
[17:41:16] INFO: Broker core-mosquitto will be used.
[17:41:16] INFO: 
MQTT Discovery ...
[17:41:17] INFO:  Adding meter: Water Meter ...
[17:41:17] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/<my_id>_current_alarms/config
[17:41:18] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/<my_id>_last_month_measure_date/config
[17:41:18] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/<my_id>_last_month_total_m3/config
[17:41:18] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/<my_id>_previous_alarms/config
[17:41:19] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/<my_id>_rssi_dbm/config
[17:41:19] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/<my_id>_timestamp/config
[17:41:19] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/<my_id>_total_m3/config
[17:41:19] INFO: MQTT Discovery cleanup...
[17:41:20] INFO: Starting web configuration service.
[17:41:20] INFO: Running wmbusmeters ...
Started config rtlwmbus[00000001] listening on none
[17:41:22] INFO: Starting Nginx...
2024/01/30 17:41:22 [notice] 83#83: using the "epoll" event method
2024/01/30 17:41:22 [notice] 83#83: nginx/1.24.0
2024/01/30 17:41:22 [notice] 83#83: OS: Linux 6.1.63-haos-raspi
2024/01/30 17:41:22 [notice] 83#83: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/01/30 17:41:22 [notice] 83#83: start worker processes
2024/01/30 17:41:22 [notice] 83#83: start worker process 284
```
</details>

<details>
<summary>Logs from 1.14.0-101-1</summary>

```
6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/nginx.sh
cont-init: info: /etc/cont-init.d/nginx.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun nginx (no readiness notification)
s6-rc: info: service legacy-services successfully started
[17:45:12] INFO: CONFIG_CONF ...
[17:45:12] INFO: {"device":"rtlwmbus","format":"json","logfile":"/dev/stdout","loglevel":"normal","logtelegrams":false,"shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[17:45:12] INFO: CONFIG_METERS ...
[17:45:12] INFO: [{"name":"Water Meter","driver":"izar","id":"<my_id>","key":"NOKEY"}]
[17:45:12] INFO: Syncing wmbusmeters configuration ...
[17:45:12] INFO: Registering meters ...
[17:45:12] INFO: Adding meter-0001 ...
[17:45:12] INFO: Generating MQTT configuration ... 
[17:45:13] INFO: Broker core-mosquitto will be used.
[17:45:14] INFO: 
MQTT Discovery ...
[17:45:14] INFO:  Adding meter: Water Meter ...
[17:45:15] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/<my_id>_current_alarms/config
[17:45:15] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/<my_id>_last_month_measure_date/config
[17:45:15] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/<my_id>_last_month_total_m3/config
[17:45:15] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/<my_id>_previous_alarms/config
[17:45:16] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/<my_id>_rssi_dbm/config
[17:45:16] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/<my_id>_timestamp/config
[17:45:16] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/<my_id>_total_m3/config
[17:45:16] INFO: MQTT Discovery cleanup...
[17:45:17] INFO: Starting web configuration service.
[17:45:17] INFO: Running wmbusmeters ...
Started config rtlwmbus[00000001] listening on none
[17:45:18] INFO: Starting Nginx...
2024/01/30 17:45:18 [notice] 83#83: using the "epoll" event method
2024/01/30 17:45:18 [notice] 83#83: nginx/1.24.0
2024/01/30 17:45:18 [notice] 83#83: OS: Linux 6.1.63-haos-raspi
2024/01/30 17:45:18 [notice] 83#83: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/01/30 17:45:18 [notice] 83#83: start worker processes
2024/01/30 17:45:18 [notice] 83#83: start worker process 282
Lost 00000001 closing rtlwmbus[00000001]
Started config rtlwmbus[00000001] listening on none
Lost 00000001 closing rtlwmbus[00000001]
Started config rtlwmbus[00000001] listening on none
Lost 00000001 closing rtlwmbus[00000001]
Started config rtlwmbus[00000001] listening on none
<last two lines repeat forever, appearing every 2 seconds or so>
```
</details>

jackwilsdon, 2024-01-30:
The RTL-SDR dongle I'm using is an RTL-SDR Blog V3 (RTL2832U R820T2 TCXO+BIAS T+HF). My meter is an IZAR RC 868 I R4 PL.

weetmuts, 2024-01-31:
Can you try to upgrade to latest ha-addon-edge? I have pushed fixes that will log warning if the data-pipeline stalls
and then try to restart it. So when you upgrade to ha-addon-edge check the logs to see what happens.

jackwilsdon, 2024-02-01:
It seems like it has started working again after updating to 1.14.0-106. There's nothing obvious in the logs indicating why:

<details>
<summary>Logs</summary>

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/nginx.sh
cont-init: info: /etc/cont-init.d/nginx.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun nginx (no readiness notification)
s6-rc: info: service legacy-services successfully started
[21:46:23] INFO: CONFIG_CONF ...
[21:46:23] INFO: {"device":"rtlwmbus","format":"json","logfile":"/dev/stdout","loglevel":"normal","logtelegrams":false,"shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[21:46:23] INFO: CONFIG_METERS ...
[21:46:23] INFO: [{"name":"Water Meter","driver":"izar","id":"<my_id>","key":"NOKEY"}]
[21:46:23] INFO: Syncing wmbusmeters configuration ...
[21:46:23] INFO: Registering meters ...
[21:46:23] INFO: Adding meter-0001 ...
[21:46:24] INFO: Generating MQTT configuration ... 
[21:46:25] INFO: Broker core-mosquitto will be used.
[21:46:25] INFO: 
MQTT Discovery ...
[21:46:26] INFO:  Adding meter: Water Meter ...
[21:46:26] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/<my_id>_current_alarms/config
[21:46:26] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/<my_id>_last_month_measure_date/config
[21:46:27] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/<my_id>_last_month_total_m3/config
[21:46:27] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/<my_id>_previous_alarms/config
[21:46:27] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/<my_id>_rssi_dbm/config
[21:46:27] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/<my_id>_timestamp/config
[21:46:28] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/<my_id>_total_m3/config
[21:46:28] INFO: MQTT Discovery cleanup...
[21:46:29] INFO: Starting web configuration service.
[21:46:29] INFO: Running wmbusmeters ...
Started config rtlwmbus[00000001] listening on none
[21:46:30] INFO: Starting Nginx...
2024/01/31 21:46:30 [notice] 83#83: using the "epoll" event method
2024/01/31 21:46:30 [notice] 83#83: nginx/1.24.0
2024/01/31 21:46:30 [notice] 83#83: OS: Linux 6.1.63-haos-raspi
2024/01/31 21:46:30 [notice] 83#83: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/01/31 21:46:30 [notice] 83#83: start worker processes
2024/01/31 21:46:30 [notice] 83#83: start worker process 287
```
</details>

closed 2024-02-01 by jackwilsdon