wmbusmeters/wmbusmeters #1164: BusyBox mktemp doesn't support --suffix

majherek, 2024-01-31:
### Type of request

Bug

### OS version

Ubuntu 22.04.3 LTS

### wmbusmeters version

docker image: 82abe39f45d1

### Your message goes here

Hi, 

I try to use official docker image (latest):
```wmbusmeters/wmbusmeters    latest               82abe39f45d1   7 days ago       39.7MB```

Also compile my own today.

```
532963c0a842:/wmbusmeters# mktemp  --suffix=__test
mktemp: unrecognized option: suffix=__test
BusyBox v1.36.1 (2023-11-07 18:53:09 UTC) multi-call binary.

Usage: mktemp [-dt] [-p DIR] [TEMPLATE]

Create a temporary file with name based on TEMPLATE and print its name.
TEMPLATE must end with XXXXXX (e.g. [/dir/]nameXXXXXX).
Without TEMPLATE, -t tmp.XXXXXX is assumed.

	-d	Make directory, not file
	-q	Fail silently on errors
	-t	Prepend base directory name to TEMPLATE
	-p DIR	Use DIR as a base directory (implies -t)
	-u	Do not create anything; print a name
```
From logs in debug mode:

```
(rtlwmbus) using command: ERRFILE=$(mktemp --suffix=_wmbusmeters_rtlsdr) ; echo ERRFILE=$ERRFILE ; date -Iseconds > $ERRFILE ; tail -f $ERRFILE & /usr/bin/rtl_sdr  -d 0 -f 868.9M -s 1.6e6 - 2>>$ERRFILE | /usr/bin/rtl_wmbus  -f
(rtlwmbus) opening 012345671
(serial) EVENT thread interrupted
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "ERRFILE=$(mktemp --suffix=_wmbusmeters_rtlsdr) ; echo ERRFILE=$ERRFILE ; date -Iseconds > $ERRFILE ; tail -f $ERRFILE & /usr/bin/rtl_sdr  -d 0 -f 868.9M -s 1.6e6 - 2>>$ERRFILE | /usr/bin/rtl_wmbus  -f"
(serialcmd) opened /bin/sh pid 55 fd 12 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) TIMER thread interrupted
(serial) EVENT thread interrupted
(serial) received ascii "mktemp: unrecognized option: suffix=_wmbusmeters_rtlsdr<0A>BusyBox v1.36.1 (2023-11-07 18:53:09 UTC) multi-call binary.<0A><0A>Usage: mktemp [-dt] [-p DIR] [TEMPLATE]<0A><0A>Create a temporary file with name based on TEMPLATE and print its name.<0A>TEMPLATE must end with XXXXXX (e.g. [/dir/]nameXXXXXX).<0A>Without TEMPLATE, -t tmp.XXXXXX is assumed.<0A><0A><09>-d<09>Make directory, not file<0A><09>-q<09>Fail silently on errors<0A><09>-t<09>Prepend base directory name to TEMPLATE<0A><09>-p DIR<09>Use DIR as a base directory (implies -t)<0A><09>-u<09>Do not create anything; print a name<0A><0A>Base directory is: -p DIR, else $TMPDIR, else /tmp<0A>ERRFILE=<0A>/bin/sh: can't create : nonexistent directory<0A>/bin/sh: can't create : nonexistent directory<0A>rtl_wmbus: monitoring flow<0A>"
```

**So busybox mktemp doesn't support --suffix.**

majherek, 2024-01-31:
Quick workaround. 

Create patch file containing:
```
--- wmbus_rtlwmbus.cc	2024-01-31 17:39:42.573611206 +0100
+++ wmbus_rtlwmbus_new.cc	2024-01-31 17:35:20.935778963 +0100
@@ -180,11 +180,11 @@
 
             if (!force_freq)
             {
-                command = "ERRFILE=$(mktemp --suffix=_wmbusmeters_rtlsdr) ; echo ERRFILE=$ERRFILE ;  date -Iseconds > $ERRFILE ; tail -f $ERRFILE & "+rtl_sdr+" "+ppm+" -d "+to_string(id)+" -f "+freq+" -s 1.6e6 - 2>>$ERRFILE | "+rtl_wmbus+" -s"+add_f;
+                command = "ERRFILE=$(mktemp) ; echo ERRFILE=$ERRFILE ;  date -Iseconds > $ERRFILE ; tail -f $ERRFILE & "+rtl_sdr+" "+ppm+" -d "+to_string(id)+" -f "+freq+" -s 1.6e6 - 2>>$ERRFILE | "+rtl_wmbus+" -s"+add_f;
             }
             else
             {
-                command = "ERRFILE=$(mktemp --suffix=_wmbusmeters_rtlsdr) ; echo ERRFILE=$ERRFILE ; date -Iseconds > $ERRFILE ; tail -f $ERRFILE & "+rtl_sdr+" "+ppm+" -d "+to_string(id)+" -f "+freq+" -s 1.6e6 - 2>>$ERRFILE | "+rtl_wmbus+" "+add_f;
+                command = "ERRFILE=$(mktemp) ; echo ERRFILE=$ERRFILE ; date -Iseconds > $ERRFILE ; tail -f $ERRFILE & "+rtl_sdr+" "+ppm+" -d "+to_string(id)+" -f "+freq+" -s 1.6e6 - 2>>$ERRFILE | "+rtl_wmbus+" "+add_f;
             }
         }
         verbose("(rtlwmbus) using command: %s\n", command.c_str());
```
Name it e.g.: **patch_by_majherek.patch** and place in `docker` folder. 

Then modify `Dockerfile`:
- add this line after git clone:
```COPY patch_by_majherek.patch ./```
- add this line before `RUN make`
```RUN patch src/wmbus_rtlwmbus.cc ../patch_by_majherek.patch```

Build your own docker image with:
```docker build -t majherek/wmbusmeters:20240131-03 .```

Now it's working:
```
(rtlwmbus) using command: ERRFILE=$(mktemp) ; echo ERRFILE=$ERRFILE ; date -Iseconds > $ERRFILE ; tail -f $ERRFILE & /usr/bin/rtl_sdr  -d 0 -f 868.9M -s 1.6e6 - 2>>$ERRFILE | /usr/bin/rtl_wmbus  -f
(rtlwmbus) opening 012345671
(serial) EVENT thread interrupted
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "ERRFILE=$(mktemp) ; echo ERRFILE=$ERRFILE ; date -Iseconds > $ERRFILE ; tail -f $ERRFILE & /usr/bin/rtl_sdr  -d 0 -f 868.9M -s 1.6e6 - 2>>$ERRFILE | /usr/bin/rtl_wmbus  -f"
(serialcmd) opened /bin/sh pid 19 fd 6 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
(serial) received ascii "ERRFILE=/tmp/tmp.OPDkcM<0A>rtl_wmbus: monitoring flow<0A>2024-01-31T18:10:17+01:00<0A>"
(rtlwmbus) checkRTLWMBusFrame "ERRFILE=/tmp/tmp.OPDkcM<0A>rtl_wmbus: monitoring flow<0A>2024-01-31T18:10:17+01:00<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "rtl_wmbus: monitoring flow<0A>2024-01-31T18:10:17+01:00<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "2024-01-31T18:10:17+01:00<0A>"
(rtlwmbus) only text
[...]
```

weetmuts, 2024-01-31:
Arrrggghhhh. Oh well. Should this command work? I think I forgot the -t because I want the file in /tmp
so that it disappears when the machine is rebooted.

mktemp -t wmbusmeters_rtlsdr.XXXXXXXXXX

These temporary files are a fix for the bug in rtl_sdr. Which is fixed in version 2.0.1 of rtlsdr....

majherek, 2024-01-31:
Thanks. 

`-t` should work.

```
$ docker exec -it wmbusmeters  /bin/sh
/wmbusmeters # mktemp -t wmbusmeters_rtlsdr.XXXXXXXXXX
/tmp/wmbusmeters_rtlsdr.XXXXhmIbbC
/wmbusmeters # 
/wmbusmeters # mktemp -t wmbusmeters_rtlsdr.XXXXXXXXXX
/tmp/wmbusmeters_rtlsdr.XXXXDaHBBg
```

weetmuts, 2024-01-31:
Fix pushed. ha-addon-edge is rebuilding. Please test when its completed. :-)
https://github.com/wmbusmeters/wmbusmeters-ha-addon/actions/runs/7731552791

majherek, 2024-01-31:
Problem solved. It works. Great. Thanks.

closed 2024-01-31 by majherek