wmbusmeters/wmbusmeters #1166: decryption issue, apator162

john172426, 2024-02-03:
found in LOG file: Started config im871a[00104436] on /dev/ttyUSB0 listening on t1,c1
(meter) newly created meter (Vodnik 03758055 apator162) did not handle telegram!
(TPL) warning: aes-cbc-iv decryption received less bytes than expected for decryption! Got 87 bytes but expected at least 96 bytes since num encr blocks was 6.
(TPL) warning: decryption received non-multiple of 16 bytes! Got 87 bytes shrinking message to 80 bytes.
(apator162) telegram contains a register (97) with unknown size.
Please open an issue at https://github.com/wmbusmeters/wmbusmeters/
and report this telegram: 654401068777750305077A9A0060852F2F0F8163C298180200437403840397CC0D97CC01010000101FC702007B012DC5020086A90200AD8C0200FE7402009F5A02006C420200DF260200131202004FF50100FBD80100E6BD010070A20100A0EA345D75974020
 

https://wmbusmeters.org/analyze/654401068777750305077a9a0060852f2f0f8163c298180200437403840397cc0d97cc01010000101fc702007B012dc5020086a90200ad8c0200fe7402009f5a02006c420200df260200131202004ff50100fBd80100e6Bd010070a20100a0ea345d75974020

weetmuts, 2024-02-13:
Thank you ! This is a problem in the receiving of the telegram, strange. Unfortunately the necessary information is not visible above, you have to enable loglevel=debug to see the actual data from the im871a dongle.

If this happens often, then enable debug and capture an error and post here.

john172426, 2024-02-21:
I found a config mistake at my side. Bus meter No. 03758055 was configured with driver=qcaloric (no encryption) but in reality it is NOT "Heat Cost Allocator" from manufacturer (QDS) Qundis, Germany (0x4493) but "water meter" (with encryption) by manufacturer (APA) Apator, Poland (0x601) instead. Sorry for confusion, perhaps slightly different wording in (TPL) warning could hint the user ie. incorrect driver...etc

closed 2024-05-22 by github-actions[bot]