wmbusmeters/wmbusmeters #1169: latest docker image 40b78bff5395, is broken.

itcsoft54, 2024-02-05:
### Type of request

Bug

### OS version

debian 12

### wmbusmeters version

1.14.0-106-g440ea26

### Your message goes here

when use the latest wmbusmeters image (40b78bff5395), something is broken :

```
./wmbusmeters rtlwmbus MyWaterMeter auto 12345678 NOKEY
Started config rtlwmbus[00000001] listening on none
Warning! Detected rtl_wmbus exit due to stopped data flow. Resetting pipeline!
(wmbus) resetting 00000001:rtlwmbus[00000001]
(wmbus) reset completed 00000001:rtlwmbus[00000001]
Warning! Detected rtl_wmbus exit due to stopped data flow. Resetting pipeline!
(wmbus) resetting 00000001:rtlwmbus[00000001]
^C(wmbus) reset completed 00000001:rtlwmbus[00000001]
/wmbusmeters # ./wmbusmeters -h
wmbusmeters version: 1.14.0-106-g440ea26
```

using rtl_433 directly work :
```
rtl_433 -f 868.625M
rtl_433 version 23.11-37-g0ae39cc5 branch master at 202401281113 inputs file rtl_tcp RTL-SDR

New defaults active, use "-Y classic -s 250k" if you need the old defaults

Detached kernel driver
Found Rafael Micro R820T tuner
[SDR] Using device 0: Nooelec, NESDR SMArt v5, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
```

using : wmbusmeters is broken

```
./wmbusmeters --verbose rtlwmbus MyWaterMeter auto 83800549 NOKEY
(wmbusmeters) version: 1.14.0-106-g440ea26
(config) using device: rtlwmbus 
(config) number of meters: 1
Started config rtlwmbus[00000001] listening on none
(rtlwmbus) using command: ERRFILE=$(mktemp -t wmbusmeters_rtlsdr.XXXXXXXXXXXX) ; echo ERRFILE=$ERRFILE ;  date -Iseconds > $ERRFILE ; tail -f $ERRFILE & /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>>$ERRFILE | /usr/bin/rtl_wmbus -s -f
(serialcmd) opened /bin/sh pid 25 fd 6 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
Warning! Detected rtl_wmbus exit due to stopped data flow. Resetting pipeline!
(serialcmd) closed /bin/sh pid=25 fd=6 (rtlwmbus)
(wmbus) resetting 00000001:rtlwmbus[00000001]
(serialcmd) opened /bin/sh pid 39 fd 6 (rtlwmbus)
(wmbus) reset completed 00000001:rtlwmbus[00000001]
Warning! Detected rtl_wmbus exit due to stopped data flow. Resetting pipeline!
(serialcmd) closed /bin/sh pid=39 fd=6 (rtlwmbus)
(wmbus) resetting 00000001:rtlwmbus[00000001]
^C(serialcmd) opened /bin/sh pid 53 fd 6 (rtlwmbus)
(wmbus) reset completed 00000001:rtlwmbus[00000001]
(serial) event loop stopped!
(serialcmd) closed /bin/sh pid=53 fd=6 (rtlwmbus)
```

the command parameter "-f" rtl_wmbus in seems to be the root cause of broken stream and so broken wmbusmeters run.

I am using : Nooelec, NESDR SMArt v5, SN: 00000001, "Generic RTL2832U OEM"

weetmuts, 2024-02-05:
So the new -f options is to exit rtl_wmbus if the pipeline stalls due to a hung rlt_sdr dongle or stole usb port.
The latest rtl_wmbus should be part of the docker image and therefore the -f options should be there.

Can you run the rtl_wmbus inside the docker image somehow?
If no arguments it should print:
Usage ./build/rtl_wmbus:
	-a accelerate (use an inaccurate atan version)
	-r 0 to disable run length algorithm
	-t 0 to disable time2 algorithm
	-d 2 set decimation rate to 2 (defaults to 2 if omitted)
	-v show used algorithm in the output
	-V show version
	-s receive S1 and T1/C1 datagrams simultaneously. rtl_sdr _MUST_ be set to 868.625MHz (-f 868.625M)
	-p [T,S] to disable processing T1/C1 or S1 mode.
	-f exit if flow of incoming data stops.
	-h print this help

weetmuts, 2024-02-05:
Perhaps you run an rtl_wmbus outside of the docker image that has not been upgraded to the latest version yet?

weetmuts, 2024-02-05:
Can you run wmbusmeters with --debug then we will see the communication with rtl_wmbus

itcsoft54, 2024-02-05:
@weetmuts all commands are running from inside of docker using --entrypoint=/bin/sh to bypass docker entrypoint and have a shell.

itcsoft54, 2024-02-05:
> Can you run wmbusmeters with --debug then we will see the communication with rtl_wmbus

I run with debug when i have access to host and post result here.

weetmuts, 2024-02-06:
How often does it print?
Warning! Detected rtl_wmbus exit due to stopped data flow. Resetting pipeline!
(wmbus) resetting 00000001:rtlwmbus[00000001]

If it prints every 2 seconds, then the pipeline is actually stalled, and you experience the rtl_sdr dongle docker problem
that it is supposed to find...but you say that rtl_433 works.....hmmm.

If it prints much much faster, then its something else.

itcsoft54, 2024-02-06:
@weetmuts, i believe that the problem is here : https://github.com/weetmuts/rtl-wmbus/blob/20cafdcecf28121cb4d5546cfe9cbc1822a70a03/rtl_wmbus.c#L1252-L1259.
when -f is use if buffer is empty, then break but because rtl_sdr is software, it can take up to 3 second (in my case) before first data is available :

I use `time /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - | grep -qE '.*'` to know how much it's take to initialize and send first data in pipe

the part `| grep -qE '.*'` exit after first bytes of data is read on pipe, and send SIGPIPE to rtl_sdr process.
```
time /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - | grep -qE '.*'
Found 1 device(s):
  0:  Nooelec, NESDR SMArt v5, SN: 00000001

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sampling at 1600000 S/s.
Tuned to 868625000 Hz.
Tuner gain set to automatic.
Reading samples in async mode...
Signal caught, exiting!
Short write, samples lost, exiting!

User cancel, exiting...
Reattached kernel driver
real	0m 2.66s
user	0m 0.02s
sys	0m 0.04s
```
`alarm(2)` as define, stop fread before the first data come (signal interrupt fread).

I can make a PR (and open issue) to weetmuts/rtl-wmbus to improve buffer read and broken stream detection if needed.

itcsoft54, 2024-02-06:
@weetmuts it's not a docker problem, it's a rtl_wmbus problem, maybe we have to rename the issue.

itcsoft54, 2024-02-06:
`rtl_433 -R 0 -f 868.95M -s 1600k -w - | rtl_wmbus -p s -a` don't output anything except initialization output :
```
/wmbusmeters # rtl_433 -R 0 -f 868.95M -s 1600k -w - | rtl_wmbus -p s -a
rtl_433 version 23.11-37-g0ae39cc5 branch master at 202401281113 inputs file rtl_tcp RTL-SDR
Disabling all device decoders.

New defaults active, use "-Y classic -s 250k" if you need the old defaults

Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
```
run about 10 minutes.

@testuser7 why use 868.95M frequency instead of 868.625M ? It seems produce T1 output only when use 868.625M frequency and with -s param.

testuser7, 2024-02-06:
I don't really know. I'm using this frequency and it works.
Some more info about frequency: https://github.com/wmbusmeters/wmbusmeters/issues/290#issuecomment-853093993

And does this work `rtl_433 -f 868.95M`?

itcsoft54, 2024-02-06:
yes it's work :
```
/wmbusmeters # rtl_433 -f 868.95M 
rtl_433 version 23.11-37-g0ae39cc5 branch master at 202401281113 inputs file rtl_tcp RTL-SDR

New defaults active, use "-Y classic -s 250k" if you need the old defaults

Found Rafael Micro R820T tuner
[SDR] Using device 0: Nooelec, NESDR SMArt v5, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-02-06 18:52:12
model     : Wireless-MBus Mode      : T            Manufacturer: DME         ID        : [REDACTED]
Version   : 123          Device Type: 0x07         Device Type String: Water Control   : 0x44          Data Length: 31           Data      : [REDACTED] Integrity : CRC
Control Info: 0x7A       Access number: 0xDB       Device Type: 0x00         Configuration Word: 0x0510 Payload Encrypted: 1
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
t
```

testuser7, 2024-02-06:
Try this one: ```rtl_433 -R 0 -f 868.95M -w - | rtl_wmbus -p s -a```

itcsoft54, 2024-02-06:
don't output anything except initialization output :
```
/wmbusmeters # rtl_433 -R 0 -f 868.95M -w - | rtl_wmbus -p s -a
rtl_433 version 23.11-37-g0ae39cc5 branch master at 202401281113 inputs file rtl_tcp RTL-SDR
Disabling all device decoders.

New defaults active, use "-Y classic -s 250k" if you need the old defaults

Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!

```

itcsoft54, 2024-02-06:
> Do you have binary data output: `rtl_433 -R 0 -f 868.95M -w -`?
yes

testuser7, 2024-02-06:
Can you try this docker image: https://github.com/testuser7/wmbusmeters/pkgs/container/wmbusmeters ?

itcsoft54, 2024-02-06:
running  : `rtl_433 -R 0 -f 868.95M -w - | rtl_wmbus -p s -a` don't output anything except initialization output :
```
root@home-services:~# podman pull ghcr.io/testuser7/wmbusmeters:dev
Trying to pull ghcr.io/testuser7/wmbusmeters:dev...
Getting image source signatures
Copying blob 02f2207cdf9f done   | 
Copying blob 8fac0863c724 done   | 
Copying blob 3ccd4dff06c6 done   | 
Copying blob 52fdf6e0c38e done   | 
Copying blob 4abcf2066143 done   | 
Copying blob 10aeb642efd9 done   | 
Copying blob 3f4a4dea08f0 done   | 
Copying config 8bee28b748 done   | 
Writing manifest to image destination
8bee28b748f284a90773cdfc7b2b913c6e0e24d1819f89e9e8ac008ec51874a5
root@home-services:~# podman run -it --rm --privileged -v /dev/swradio0:/dev/swradio0 -v /dev/bus/usb/:/dev/bus/usb/ -v /tmp/outdir:/tmp/outdir --entrypoint=/bin/sh wmbusmeters:dev
/wmbusmeters # rtl_433 -R 0 -f 868.95M -w - | rtl_wmbus -p s -a
rtl_433 version v3.19.0_rc2-183-gae19996b542 branch master at 202312040647 inputs file rtl_tcp RTL-SDR with TLS
Disabling all device decoders.

New defaults active, use "-Y classic -s 250k" if you need the old defaults

Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
```

weetmuts, 2024-02-06:
>when -f is use if buffer is empty, then break but because rtl_sdr is software, it can take up to 3 second (in my case) before first data is available :

@itcsoft54 Yikes, if it take 3 seconds to fill up the buffer, then its a major problem that the rtl_wmbus flow detector
will abort in 2 seconds. The timeout in rtl_wmbus is for getting any data at all, ie it s the read from the file descriptor,
so the question is, why does it take 3 seconds. It should start sending data immediately to rtl_wmbus....right?

itcsoft54, 2024-02-06:
@weetmuts dirty fix alarm(2) to alarm(5) fix the problem. (i recompile and rebuild image for testing)
I mean my sdr usb key has "slow start" and take up to 3 sec to init before send any data in the pipe. Root cause may be the hardware where the usb key is connect (a 2 cpu and 4go ram virtualbox vm).

The usb sdr key i use : https://amzn.eu/d/8D8nMso

itcsoft54, 2024-02-07:
commit https://github.com/weetmuts/rtl-wmbus/commit/6c4f95fccb7b9059372f3af8573621d766f1816f fix this issue.

weetmuts, 2024-02-08:
Thank you very much @itcsoft54 for reporting this issue and help fix it! Unforseen hardware problems is really hard to prevent against and you assistance was very valuable. I really thought that it would start sending samples in a few milliseconds.....so 2 seconds should be "enough" ! :-D

itcsoft54, 2024-02-18:
Ok no more problems with the new image. So issue can be closed.

closed 2024-02-18 by itcsoft54

asbachb, 2024-08-02:
@weetmuts Do you plan to push the fix to the upstream repo? I'm currently facing exactly the same situation.

Maybe another possibility would be to allow a parameter for `-f` so that it defaults to `2` but can be overwritten with an other value.

weetmuts, 2024-08-03:
@asbachb Thanks for the heads up! I completely forgot to push to upstream.
I saw your pull request that perhaps was accidentally closed? So I created a new one upstream.

asbachb, 2024-08-03:
@weetmuts hehe I already opened up a new one (https://github.com/xaelsouth/rtl-wmbus/pull/53) as the previous one had merge conflicts (like now yours)

I took your commit and rebased it to most recent master. So you're getting the credit anyways :)

If you want to fix your PR I'd close mine if you wish.

weetmuts, 2024-08-03:
Thanks for fixing! I will close mine! :-)