wmbusmeters/wmbusmeters #1171: nanoCUL-Device not recognized at startup of the container. Gets detected when container is restarted.

O491dogan, 2024-02-06:
### Type of request

Question

### OS version

LXC with Debian 12

### wmbusmeters version

latest - 06.02.2024

### Your message goes here

Hello everyone, 

I have a minor problem with the use of the docker image.
At first, I used the method to build from source inside a debian lxc and I managed to get it all working very good.

Then I wanted to give the docker image a try and it also works quite well except for the case, when the container is starting.
It doesn't recognize the nanoCul-Stick at first and outputs:

```
Expected cul on /dev/ttyUSB0 but did not find it! Ignoring tty!
No wmbus device detected, waiting for a device to be plugged in.
```

When I restart the wmbusmeters container in portainer, the stick is recognized and the listening starts as expected:
`Started config cul on /dev/ttyUSB0 listening on c1`

Does someone experience the same behavior and has a hint what the problem may be?

**My setup:**

Debian 12 in a non-privileged LXC on a Proxmox Host.
I have passed through the nanoCul-Stick successfully as it works with the wmbusmeters build from source and also when I restart the container after it has already started up.

My **wmbusmeters-config** in /opt/wmbusmeters/etc/wmbusmeters.conf looks like this:

```
loglevel=normal
device=/dev/ttyUSB0:cul:c1
logtelegrams=false
format=json
shell=/usr/bin/mosquitto_pub -h 192.168.xxx.xx -p 1883 -t wmbusmeters/$METER_ID/total -m "$METER_JSON"
```

The **docker compose** I have used looks like this (docker runs as root inside the lxc):

```
version: "2"
services:
  wmbusmeters:
    image: wmbusmeters/wmbusmeters
    container_name: wmbusmeters
    restart: always
    volumes:
      - /opt/wmbusmeters:/wmbusmeters_data
      - /etc/localtime:/etc/localtime:ro
      #- /dev/:/dev/                       # <-- passing /dev/:/dev/ instead of /dev/tty/USB0 results in the same behaviour
      - /dev/ttyUSB0:/dev/ttyUSB0
```
A delayed start of the lxc container did not change anything regarding this behaviour.

I can provide additional information if necessary and if someone tells me what to provide.

P.S.: Nice piece of software, thanks! :-)

O491dogan, 2024-02-07:
Hey @weetmuts , 

thank you for taking the time!
I changed the loglevel from "normal" to "debug" in my config-file.

Sorry for the long post because of the logs.

Here's the output when the container get's initially started.

```
2024-02-07T09:30:00.532448919Z (config) "\00\00\00\00\00\00~2024-02-07T09:32:06.604983097Z (config) "# Remember to change auto here to the device you are going to use in production." ""
2024-02-07T09:32:06.605268769Z (config) "#device=auto:t1" ""
2024-02-07T09:32:06.605277371Z (config) "#device=rtlwmbus:CMD(rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a)" ""
2024-02-07T09:32:06.605283415Z (config) "device" "/dev/ttyUSB0:cul:c1"
2024-02-07T09:32:06.605286215Z (config) "#logfile=/var/log/wmbusmeters/wmbusmeters.log" ""
2024-02-07T09:32:06.605288738Z (config) "# Set to true to capture all received telegrams in log file." ""
2024-02-07T09:32:06.605291222Z (config) "logtelegrams" "false"
2024-02-07T09:32:06.605293595Z (config) "format" "json"
2024-02-07T09:32:06.605296037Z (config) "# Enable the meterfiles to write telegrams to disk." ""
2024-02-07T09:32:06.605298240Z (config) "#meterfiles=/var/lib/wmbusmeters/meter_readings" ""
2024-02-07T09:32:06.605300508Z (config) "#meterfilesaction=overwrite" ""
2024-02-07T09:32:06.605302643Z (config) "# Enable execution of a shell command for each received telegram. For example: curl or mqtt" ""
2024-02-07T09:32:06.605304954Z (config) "#shell=/usr/bin/mosquitto_pub -h localhost -t wmbusmeters/ -m """ ""
2024-02-07T09:32:06.605307250Z (config) "#shell=psql water -c "insert into consumption values ('',,'') "" ""
2024-02-07T09:32:06.605309587Z (config) "shell" "/usr/bin/mosquitto_pub -h 192.168.178.31 -p 1883 -t wmbusmeters/$METER_ID/total -m "$METER_JSON""
2024-02-07T09:32:06.605312013Z (config) "# The alarmshell is executed when a problem with the receiving radio hardware is detected." ""
2024-02-07T09:32:06.605990126Z (wmbusmeters) version: 1.14.0-106-g440ea26
2024-02-07T09:32:06.606003398Z (config) using device: /dev/ttyUSB0:cul:c1 
2024-02-07T09:32:06.606007276Z (config) number of meters: 4
2024-02-07T09:32:06.606085696Z (lookup) with file/hex "/dev/ttyUSB0" cul
2024-02-07T09:32:06.606092831Z (serial) check if /dev/ttyUSB0 can be accessed
2024-02-07T09:32:06.606095563Z (serial) tty /dev/ttyUSB0 can be accessed
2024-02-07T09:32:06.606177206Z (serial) EVENT thread interrupted
2024-02-07T09:32:06.606489662Z (serial) before config: /dev/ttyUSB0 speed(B9600) input(ICRNL|IXON) output(ONLCR|OPOST) control(CLOCAL|CREAD|CSIZE|HUPCL) local(ECHO|ECHOCTL|ECHOE|ECHOK|ECHOKE|ICANON|IEXTEN|ISIG) special_chars(3,28,127,21,4,1,17,19,26,18,15,23,22,18,78,195,101,1,107,100,55) modem(DTR|RTS)
2024-02-07T09:32:06.606680459Z (serial) after config:  /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(18,78,195,101,16) modem(DTR|RTS)
2024-02-07T09:32:06.606711162Z (serialtty) opened /dev/ttyUSB0 fd 3 (detect cul)
2024-02-07T09:32:06.606714046Z (serial /dev/ttyUSB0) sent "560D0A"
2024-02-07T09:32:06.606717171Z (serial) EVENT thread interrupted
2024-02-07T09:32:06.806939614Z (serial) received binary "7B67C4C131316121100131218471082111B83121C48610004A252121216121312121AC"
2024-02-07T09:32:06.806955873Z (cul) probe response "{g<C4><C1>11a!<10><01>1!<84>q<08>!<11><B8>1!<C4><86><10><00>J%!!!a!1!!<AC>"
2024-02-07T09:32:07.307238961Z (serial /dev/ttyUSB0) sent "560D0A"
2024-02-07T09:32:07.307265513Z (serial) EVENT thread interrupted
2024-02-07T09:32:07.507538359Z (serial) received binary ""
2024-02-07T09:32:07.507564048Z (cul) probe response ""
2024-02-07T09:32:08.007798085Z (serial /dev/ttyUSB0) sent "560D0A"
2024-02-07T09:32:08.007823864Z (serial) EVENT thread interrupted
2024-02-07T09:32:08.208113907Z (serial) received binary ""
2024-02-07T09:32:08.208133410Z (cul) probe response ""
2024-02-07T09:32:08.708737772Z (serialtty) closed /dev/ttyUSB0 (detect cul)
2024-02-07T09:32:08.708759053Z (cul) are you there? no
2024-02-07T09:32:08.708762516Z Expected cul on /dev/ttyUSB0 but did not find it! Ignoring tty!
2024-02-07T09:32:08.708764988Z (verbose) ignoring device c1
2024-02-07T09:32:08.708767173Z No wmbus device detected, waiting for a device to be plugged in.
2024-02-07T09:32:08.708769296Z (shell) exec (capture output) "/bin/pidof"
2024-02-07T09:32:08.708771472Z (shell) arg "wmbusmetersd"
2024-02-07T09:32:08.708773506Z (serial) EVENT thread interrupted
2024-02-07T09:32:08.709442165Z (shell) output: >>><<<
2024-02-07T09:32:08.709455574Z (shell) return code 1
2024-02-07T09:32:08.709458865Z (shell) 10 exited
2024-02-07T09:32:08.709461204Z (shell) exec (capture output) "/bin/pidof"
2024-02-07T09:32:08.709463701Z (shell) arg "wmbusmeters"
2024-02-07T09:32:08.709465878Z (serial) EVENT thread interrupted
2024-02-07T09:32:08.709946196Z (shell) output: >>>7
2024-02-07T09:32:08.709958670Z <<<
2024-02-07T09:32:08.709962151Z (shell) return code 0
2024-02-07T09:32:08.709964558Z (shell) 11 exited
2024-02-07T09:32:08.709966581Z (serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
2024-02-07T09:32:08.709968638Z (serial) waiting for stop
2024-02-07T09:32:08.709970496Z (serial) EVENT thread interrupted
2024-02-07T09:32:08.709991727Z (serial) TIMER thread interrupted
[... 85 lines cut ...]
2024-02-07T11:24:01.524747513Z (wmbus) ff a dll crc mid 66-81 f5ac ok
2024-02-07T11:24:01.524749210Z (wmbus) ff a dll crc final 84-93 7f9e ok
[... cut ...]
2024-02-07T11:24:01.524782801Z (meter) FlatHeatEnergy: not for me: not my id
2024-02-07T11:24:01.524784482Z (meter) FlatWarmWater: for me? 35511848 in 35436121
2024-02-07T11:24:01.524786148Z (meter) FlatWarmWater: not for me: not my id
2024-02-07T11:24:01.524787832Z (meter) FlatColdWater: for me? 35511848 in 35537715
2024-02-07T11:24:01.524789497Z (meter) FlatColdWater: not for me: not my id
2024-02-07T11:24:01.524791174Z (wmbus) telegram from 35511848 ignored by all configured meters!
````

It seems like the cul is not responding properly because of some interrupts.

O491dogan, 2024-02-14:
Does someone have a clue why this might happen?

weetmuts, 2024-02-14:
In the first probe we receive garbage:
(cul) probe response "{g<C4><C1>11a!<10><01>1!<84>q<08>!<11><B8>1!<C4><86><10><00>J%!!!a!1!!<AC>"
and the second attempt nothing.

After the restart we get the expected response after the probe:
(cul) probe response "? (`V is unknown) Use one of B b C e F G K l M R T t V W X x<0D><0A>"

It is changing the serial speed from 9600 to 38400 the first time, and the garbage looks like a speed mismatch.
The second time, then the serial port is already 38400.

Can you search for serial speed setting problems and docker?
You can set the speed manually with: 
stty -F /dev/ttyUSB0 38400
can you use this to fix the first container?

O491dogan, 2024-02-15:
Hey, thank you again for analyzing the logs!

Indeed it looks like the speed is getting changed.
For me it looks like the speed gets changed to 38400 before the "560D0A" is sent which then leads to the receiving garbage (at least as I understand the logfile - not sure if the speed get's really changed though).

I have experimented a little bit:

**1. Booting up my proxmox host, no lxc, no wmbusmeters, no docker running**  
```
**root@eq12**:~# tty -F /dev/ttyUSB0 
speed 9600 baud; line = 0;
-brkint -imaxbel
```

**2. Starting wmbusmeters docker container inside lxc**
Host:
```
**root@eq12:**~# stty -F /dev/ttyUSB0 
speed 38400 baud; line = 0;
intr = <undef>; quit = <undef>; erase = <undef>; kill = <undef>; eof = <undef>; start = <undef>; stop = <undef>; susp = <undef>; rprnt = <undef>;
werase = <undef>; lnext = <undef>; discard = <undef>; min = 0; time = 0;
-brkint -icrnl -imaxbel
-opost -onlcr
-isig -icanon -iexten -echo -echoe -echok -echoctl -echoke
```
LXC (I have mapped the /dev/ttyUSB0 with an udev-rule also to /dev/nanoCUL which is passed to the container):
```
**root@rtl-test:~**# stty -F /dev/nanoCUL 
speed 38400 baud; line = 0;
intr = <undef>; quit = <undef>; erase = <undef>; kill = <undef>; eof = <undef>; start = <undef>; stop = <undef>; susp = <undef>; rprnt = <undef>; werase = <undef>;
lnext = <undef>; discard = <undef>; min = 0; time = 0;
-brkint -icrnl -imaxbel
-opost -onlcr
-isig -icanon -iexten -echo -echoe -echok -echoctl -echoke
```
**3. Restarting the docker container**
Host:
```
**root@eq12:**~# stty -F /dev/ttyUSB0 
speed 38400 baud; line = 0;
intr = <undef>; quit = <undef>; erase = <undef>; kill = <undef>; eof = <undef>; start = <undef>; stop = <undef>; susp = <undef>; rprnt = <undef>;
werase = <undef>; lnext = <undef>; discard = <undef>; min = 0; time = 0;
-brkint -icrnl -imaxbel
-opost -onlcr
-isig -icanon -iexten -echo -echoe -echok -echoctl -echoke
```

LXC:
```
**root@rtl-test:**~# stty -F /dev/nanoCUL 
speed 38400 baud; line = 0;
intr = <undef>; quit = <undef>; erase = <undef>; kill = <undef>; eof = <undef>; start = <undef>; stop = <undef>; susp = <undef>; rprnt = <undef>; werase = <undef>;
lnext = <undef>; discard = <undef>; min = 0; time = 0;
-brkint -icrnl -imaxbel
-opost -onlcr
-isig -icanon -iexten -echo -echoe -echok -echoctl -echoke
```

**4. A fresh start of the proxmox host setting the speed to 38400; no running LXC and wmbusmeters yet**
```
**root@eq12:**~# stty -F /dev/ttyUSB0 
speed 9600 baud; line = 0;
-brkint -imaxbel
root@eq12:~# stty -F /dev/ttyUSB0 38400
root@eq12:~# stty -F /dev/ttyUSB0 
speed 38400 baud; line = 0;
-brkint -imaxbel
```

**5.  Wmbusmeters docker log output first start**
[... 27 lines cut ...]
2024-02-15T20:13:56.993093615Z (verbose) ignoring device c1
2024-02-15T20:13:56.993095353Z No wmbus device detected, waiting for a device to be plugged in.
2024-02-15T20:13:56.993097092Z (shell) exec (capture output) "/bin/pidof"
2024-02-15T20:13:56.993098837Z (shell) arg "wmbusmetersd"
2024-02-15T20:13:56.993100493Z (serial) EVENT thread interrupted
2024-02-15T20:13:56.993102095Z (shell) output: >>><<<
2024-02-15T20:13:56.993103944Z (shell) return code 1
2024-02-15T20:13:56.993105566Z (shell) 10 exited
2024-02-15T20:13:56.993127092Z (shell) exec (capture output) "/bin/pidof"
2024-02-15T20:13:56.993130225Z (shell) arg "wmbusmeters"
2024-02-15T20:13:56.993132060Z (shell) output: >>>7
2024-02-15T20:13:56.993133930Z <<<
2024-02-15T20:13:56.993135630Z (serial) EVENT thread interrupted
2024-02-15T20:13:56.993137249Z (shell) return code 0
2024-02-15T20:13:56.993138790Z (shell) 11 exited
2024-02-15T20:13:56.993140431Z (serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
2024-02-15T20:13:56.993142128Z (serial) waiting for stop
2024-02-15T20:13:56.993143729Z (serial) EVENT thread interrupted
2024-02-15T20:13:56.993145334Z (serial) TIMER thread interrupted
```

Unfortunatley, the behaviour hasn't changed yet. The stick doesn't respond.
I guess this is not the scope of the software anymore, but if you have another suggestion, I'd be very happy. 

Do you think this is caused by the stick's firmware? I'm currently on the linked "nanoCUL_r568_mbus_c1t1_bufsize300", I guess.

O491dogan, 2024-04-26:
Reinstalled Proxmox and recreated the container - works as expected!
I think I have messed up tings with the previously non-docker install on the same system.

Thank you again for taking the time to investigate!

closed 2024-07-26 by github-actions[bot]