wmbusmeters/wmbusmeters #1173: Is there a way to use the template and discovery option of the ha addon with a standalone wmbusmeters server ?

uncle-buzz, 2024-02-08:
### Type of request

Question

### HAOS version

11.5

### wmbusmeters version

1.14.0

### Your message goes here

because of the need for the antenna's proximity to my main water meter, I have to use another computer to host the wmbusmeters service than the computer running Home Assistant, so I'm unable to run the addon.
Is there a way to benefit your work on the ha addon for wmbusmters without running wmbusmeters locally?
If not, that could be a good enhancement!
At least, how could I use with manual configuration the already set templates for my meters?
Thanks for your help !

BIBOLV, 2024-02-08:
There is no out-of-the-box ready solution AFAIK. But if you are tinkerer and looking for small footprint device at edge then there could be some options like [this](https://github.com/maciekn/izar-wmbus-esp) and [this](https://github.com/chester4444/esp-multical21) and [this](https://github.com/SzczepanLeon/esphome-components). Then capture wmbus traffic and forward it to wmbusmeters instance over network. 
If having PC close to meter is not an issue, then depending from your wmbus receiver, you can either write some small app that captures all traffic and forwards it to central instance or use already existing solutions for that.

uncle-buzz, 2024-02-09:
Thank you for your reply.
Running wmbusmeters on the HA computer with a network pipe to read incoming DATA seems to be the easiest way (if you know how to install missing dependencies on HA OS to compile wmbusmeters ^^ )
But I would like to keep each service independent, so I prefer to keep wmbusmeters away from HA and just query the MQTT server from HA independently from wmbusmeters, I wonder if there is there a way to format JSON messages with the template in the mqtt_discovery directory before sending the JSON to the MQTT service? (I mean on the wmbusmetes server, not related to HA).
Should I write a script to format the final JSON message to call in the command line to push the MQTT message (depending on the meter type of course)?

BIBOLV, 2024-02-09:
There is no need to compile anything - there is [add-on](https://github.com/wmbusmeters/wmbusmeters-ha-addon) ready.
There has been some [discussions](https://github.com/wmbusmeters/wmbusmeters/issues/610) to implement templates in standalone version of wmbusmeters.
If you have few meters and they do not change that often, then IMO simplest solution is to forward readings to MQTT and setup card for meter in HA - fast and simple.

uncle-buzz, 2024-02-12:
Does the addon include the binary of wmbusmeters ? when I tried it, wmbusmetres didn't start but I thought the addon only included configuration of wmbusmeters driven by HA, so I didn't search why it failed to start since I wanted to run it on another server.
Thank you for the link to the previous discussion.

closed 2024-05-13 by github-actions[bot]