wmbusmeters/wmbusmeters #1175: Do Not Receive any Telegramm from Meter

Paroxxx, 2024-02-10:
### Type of request

Question

### HAOS version

Core 2024.2.1 / OS 11.5

### wmbusmeters version

1.14.0-106

### Your message goes here

Hi,

HaOs is intalled on Asustor NAS in a Virtual Machine (Virtual Box V6.1.12)

With wmbusmeters installed on a RPI, it works well with MQTT and HaOs

With HaAddon everything ### **seems** ### to be fine but nothing is litenning.

Any Help ? :)

Regards

`s6-rc: info: service legacy-services successfully started
[15:36:23] INFO: CONFIG_CONF ...
[15:36:23] INFO: {"device":"auto:t1","format":"json","logfile":"/dev/stdout","loglevel":"normal","logtelegrams":false,"shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[15:36:23] INFO: CONFIG_METERS ...
[15:36:23] INFO: []
[15:36:23] INFO: Syncing wmbusmeters configuration ...
[15:36:23] INFO: Registering meters ...
[15:36:23] INFO: Generating MQTT configuration ... 
[15:36:26] INFO: Broker core-mosquitto will be used.
[15:36:27] INFO: 
MQTT Discovery ...
[15:36:28] INFO: MQTT Discovery cleanup...
[15:36:29] INFO: Starting web configuration service.
[15:36:29] INFO: Running wmbusmeters ...
Started auto rtlwmbus[00000001] listening on t1
No meters configured. Printing id:s of all telegrams heard!
2024/02/10 15:36:32 [notice] 83#83: using the "epoll" event method
2024/02/10 15:36:32 [notice] 83#83: nginx/1.24.0
2024/02/10 15:36:32 [notice] 83#83: OS: Linux 6.1.74-haos
2024/02/10 15:36:32 [notice] 83#83: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
[15:36:32] INFO: Starting Nginx...
2024/02/10 15:36:32 [notice] 83#83: start worker processes
2024/02/10 15:36:32 [notice] 83#83: start worker process 249
172.30.32.2 - - [10/Feb/2024:15:37:32 +0100] "GET / HTTP/1.1" 200 1292 "http://192.168.1.220:8123/c8a990ad_wmbusmeters-ha-addon-edge" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/121.0.0.0 Safari/537.36"
172.30.32.2 - - [10/Feb/2024:15:37:32 +0100] "GET /bootstrap.min.css HTTP/1.1" 200 221221 "http://192.168.1.220:8123/api/hassio_ingress/MFbV_i5hjNL7PRkWP2KYHWlN4cTtvmgHPIBZovoILMw/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/121.0.0.0 Safari/537.36"`

BIBOLV, 2024-02-13:
Hi,
Likely related to known [issue](https://github.com/wmbusmeters/wmbusmeters/issues/989) that hopefully is fixed in latest edge version - so please update and test.

Paroxxx, 2024-02-17:
The last update solved the problem.

Thanks a lot !

(Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/121.0.0.0 Safari/537.36"
Received telegram from: 00301638
          manufacturer: (DME) DIEHL Metering, Germany (0x11a5)
                  type: Water meter (0x07) encrypted
                   ver: 0x7b
                device: rtlwmbus[00000001]
                  rssi: 43 dBm
                driver: dme_07

closed 2024-02-17 by Paroxxx