wmbusmeters/wmbusmeters #1176: Register of unknown size

ssuukk, 2024-02-10:
### Type of request

Bug

### OS version

n/a

### wmbusmeters version

built from current master

### Your message goes here

Please open an issue at https://github.com/wmbusmeters/wmbusmeters/
and report this telegram: 5144B4091828912218078C00A87A3D0040052F2F066DA405130A32000C13135214000F05000000005A37002F1000E61000FB14007318002F1C00622000762400972800CE2C00783000EC3300002F2F2F2F2F

weetmuts, 2024-02-12:
Are you sure you are reporting the correct telegram? That telegram decodes fine, and is not an apator162 instead it is a iwmtx5 B-Meters water meter.

The error you are referring to are for apator162 water meters.

ssuukk, 2024-02-13:
I have more than 200 w bus devices in the air, this was just a random log
message I noticed, so this might be indeed the case you describe Ed.

closed 2024-05-15 by github-actions[bot]