wmbusmeters/wmbusmeters #1179: request support for "Wehrle MODULARIS M-Bus-Modul" / "Stark M-Bus Kaltwasserzähler"

Marc084, 2024-02-14:
### Type of request

New device support

### Meter description

Online Shop: https://www.stark-elektronik.de/p/m-bus-kaltwasserzaehler-qn-2-5-dn-20-baulaenge-130mm-r3-4-g1-mid-geeicht

original Device (mBus Modul: 
https://wehrle.de/metering/fernauslesung/aufsatzmodule/
>Datasheet: https://wehrle.de/wp-content/uploads/2020/11/Modularis_M-Bus_Modul.pdf
>Manual: https://wehrle.de/wp-content/uploads/2020/11/Kurzanleitung-M-BUS-Modul-08_single.pdf
>mBus Protocol Description: https://wehrle.de/wp-content/uploads/2020/11/M-Bus_Protokollbeschreibung.pdf
>mBus Specifikation: https://wehrle.de/wp-content/uploads/2020/11/MBus_Protokoll_Spezifikationen.pdf

meter is not recognized either with driver set to "auto", "sensostar:MAIN:mbus" or "c5isf:MAIN:mbus"
Scanned meter with mBusSheet:
[image]

please add meter. 
In case you need more information, let me know

### Logged telegrams from meter in decrypted format

```shell
export from Console (HA Addon):
------------------------------------------
(meter) WasserOG: meter detection did not match the selected driver sensostar! correct driver is: unknown!
(meter) Not printing this warning again for id: 00000000 mfct: (@@@) Unknown (0x00) type: Other (0x00) ver: 0x00
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|6834346808007215054328475F031646000000|04135F000000046D05160E32426C000044130000000042EC7E073B0C78985403610F010000|1D16|+4
telegram=|6834346808007215054328475F031647000000|04135F000000046D06160E32426C000044130000000042EC7E073B0C78985403610F010000|1F16|+20
telegram=|6834346808007215054328475F031648000000|04135F000000046D06160E32426C000044130000000042EC7E073B0C78985403610F010000|2016|+36
telegram=|6834346808007215054328475F031649000000|04135F000000046D06160E32426C000044130000000042EC7E073B0C78985403610F010000|2116|+52
telegram=|6834346808007215054328475F03164A000000|04135F000000046D06160E32426C000044130000000042EC7E073B0C78985403610F010000|2216|+67
telegram=|6834346808007215054328475F03164B000000|04135F000000046D07160E32426C000044130000000042EC7E073B0C78985403610F010000|2416|+84
telegram=|6834346808007215054328475F03164C000000|04135F000000046D07160E32426C000044130000000042EC7E073B0C78985403610F010000|2516|+100
telegram=|6834346808007215054328475F03164D000000|04135F000000046D07160E32426C000044130000000042EC7E073B0C78985403610F010000|2616|+116
```

kzajac83, 2024-03-13:
@Marc084 can you confirm I requested yesterday for the same driver? 

https://github.com/wmbusmeters/wmbusmeters/issues/1210#issue-2181564653

If yes then I will close my ticket as double of you.  BTW - who is the producer? There are many sellers (brands?) but you know who real product it? Wehrle, NZR, other?

Marc084, 2024-03-20:
> @Marc084 can you confirm I requested yesterday for the same driver?
> 
> [#1210 (comment)](https://github.com/wmbusmeters/wmbusmeters/issues/1210#issue-2181564653)
> 
> If yes then I will close my ticket as double of you. BTW - who is the producer? There are many sellers (brands?) but you know who real product it? Wehrle, NZR, other?

Sry for delay.
I think we have the same module. 
In my package a manula and warranty papaer of wehrle was included. Also comparing the pictures of NZR and Wehrle i tend it's produced by Wehrle.

As long as there is no 100% driver you you can try to use "minomess:MAIN:mbus". Using this driver i got at least the current consumption.

kzajac83, 2024-03-20:
Thx.

Regarding minomess driver then yes I know and more I create build-in driver named "modularis" based on minomess, please read 2nd comment for more details:

https://github.com/wmbusmeters/wmbusmeters/issues/1210#issuecomment-1999313556

Taking your telegram the output is:

```
root@wyse:~# wmbusmeters --debug --format=json --ppjson "6834346808007215054328475F03164600000004135F000000046D05160E32426C000044130000000042EC7E073B0C78985403610F0100001D16" licznik modularis 28430515 NOKEY
(drivers) added builtin driver elster
(drivers) added builtin driver iperl
(wmbusmeters) version: 1.16.1-12-g1f2cd10
(config) using device: <6834346808007215054328475F03164600000004135F000000046D05160E32426C000044130000000042EC7E073B0C78985403610F0100001D16>:any
(config) number of meters: 1
(config) template licznik 28430515 modularis
(lookup) with file/hex "6834346808007215054328475F03164600000004135F000000046D05160E32426C000044130000000042EC7E073B0C78985403610F0100001D16" unknown
(lookup) driver: simulation hex
(main) opening <6834346808007215054328475F03164600000004135F000000046D05160E32426C000044130000000042EC7E073B0C78985403610F0100001D16>:any
Started config simulation listening on any
(simulation) in
(main) regular reset of simulation  will happen every 82800 seconds
(main) added  to files
(wmbus) no alarm (expected activity) for simulation
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(simulation) from file "6834346808007215054328475F03164600000004135F000000046D05160E32426C000044130000000042EC7E073B0C78985403610F0100001D16"
(mbus) checkMBUSFrame
 "6834346808007215054328475F03164600000004135F000000046D05160E32426C000044130000000042EC7E073B0C78985403610F0100001D16"
(wmbus) checkWMBUSFrame "6834346808007215054328475F03164600000004135F000000046D05160E32426C000044130000000042EC7E073B0C78985403610F0100001D16"
(simulator) is_mbus=true is_wmbus=false
(simulator) is mbus telegram.
(meter) no meter handled  checking 1 templates.
(mbus) parse MBUS DLL @0 58
(mbus)  "6834346808007215054328475F03164600000004135F000000046D05160E32426C000044130000000042EC7E073B0C78985403610F0100001D16"
(dvparser) found new format "0413046D426C441342EC7E0C78" with hash b00d, remembering!
(meter) licznik: for me? p0.M=@@@.V=00.T=00,28430515.M=WZG.V=03.T=16 in 28430515
(meter) licznik: yes for me
(meter) created licznik modularis 28430515 encrypted
(meter) used meter template licznik 28430515 modularis to match p0.M=@@@.V=00.T=00,28430515.M=WZG.V=03.T=16
(meter) started meter 1 (licznik 28430515 modularis) identity mode: id R28430515
(mbus) parse MBUS DLL @0 58
(mbus)  "6834346808007215054328475F03164600000004135F000000046D05160E32426C000044130000000042EC7E073B0C78985403610F0100001D16"
(meter) licznik: for me? p0.M=@@@.V=00.T=00,28430515.M=WZG.V=03.T=16 in 28430515
(meter) licznik: yes for me
(meter) licznik(1) modularis  handling telegram from 28430515.M=WZG.V=03.T=16
(meter) licznik 28430515.M=WZG.V=03.T=16 "6834346808007215054328475F03164600000004135F000000046D05160E32426C000044130000000042EC7E073B0C78985403610F0100001D16"
(mbus) parse MBUS DLL @0 58
(mbus)  "6834346808007215054328475F03164600000004135F000000046D05160E32426C000044130000000042EC7E073B0C78985403610F0100001D16"
(meters) trying field info fabrication_no(Text)[0]...
(meters) using field info fabrication_no(Text)[0] to extract 0C78 at offset 48
(meters) trying field info operating_time(Time)[1]...
(meters) trying field info on_time(Time)[2]...
(meters) trying field info on_time_at_error(Time)[3]...
(meters) trying field info meter_date(Text)[4]...
(meters) trying field info meter_datetime(Text)[5]...
(meters) using field info meter_datetime(Text)[5] to extract 046D at offset 27
(meters) trying field info total(Volume)[6]...
(meters) using field info total(Volume)[6] to extract 0413 at offset 21
(meter) Volume total_m3 decoded m3 default m3 value 0.095 (scale 1)
(meters) trying field info total_backward(Volume)[7]...
(meters) trying field info volume_flow(Flow)[8]...
(meters) trying field info target(Volume)[9]...
(meters) trying field info target_date(Text)[10]...
(meters) trying field info target(Volume)[11]...
(meters) using field info target(Volume)[11] to extract 4413 at offset 37
(meter) Volume target_m3 decoded m3 default m3 value 0 (scale 1)
(meters) trying field info target_date(Text)[12]...
(meters) using field info target_date(Text)[12] to extract 426C at offset 33
(meters) trying field info some_data(Text)[13]...
(meters) using field info some_data(Text)[13] to extract 42EC7E at offset 44
[... 39 lines cut ...]
    "name":"licznik",
    "id":"28430515",
    "target_m3":0,
    "total_m3":0.095,
    "fabrication_no":"61035498",
    "meter_datetime":"2024-02-14 22:05",
    "some_data":"ERROR_FLAGS_3B07",
    "target_date":"2000-00-00",
    "timestamp":"2024-03-20T09:47:15Z"
}
(serial) stopping manager
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) event loop stopped!
(serial) TIMER thread interrupted
(wmbus) deleted simulation

```
If you can encode error flag "3B07" and manufacturer specific data "010000" or have some documentation or have original reading software and based on reverse engineering provide usefull data then I can deliver better driver.

Please note my overlay have error flag "2BB9" and manufacturer specific the same 010000 

M-Tool from producer give me this (similar outpu in my isseu from "M-Bus Sheet"), additional don't know how get historical data in wmbumeters, don't found in telegram or better say don't know how encoded it.

[image]

closed 2024-06-19 by github-actions[bot]