wmbusmeters/wmbusmeters #1180: ERROR: Got unexpected response from the API: Service not enabled

Skysurfer-14, 2024-02-16:
### Type of request

Question

### HAOS version

11.4

### wmbusmeters version

1.15.0

### Your message goes here

After updating Wmbusmeters HA Addin to version 1.15.0 and adding the new Custom MQTT configuration I am getting the error message "[10:21:32] ERROR: Got unexpected response from the API: Service not enabled", although the addon works correctly.
What can be the reason for this and how can I solve the issue?

    cont-init: info: /etc/cont-init.d/nginx.sh exited 0
    s6-rc: info: service legacy-cont-init successfully started
    s6-rc: info: service legacy-services: starting
    services-up: info: copying legacy longrun nginx (no readiness notification)
    s6-rc: info: service legacy-services successfully started
    [10:21:28] INFO: CONFIG_CONF ...
    [10:21:28] INFO: {"device":"auto:c1","donotprobe":"/dev/ttyAMA0","format":"json","logfile":"/dev/stdout","loglevel":"normal","logtelegrams":true,"shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
    [10:21:28] INFO: CONFIG_METERS ...
    [10:21:28] INFO: [{"driver":"hcae2","id":"23412691","key":"FFEEDDCCBBAA00998877665544332211","name":"HCA_Keller"},{"driver":"hcae2","id":"23412690","key":"FFEEDDCCBBAA00998877665544332211","name":"HCA_EG_Wintergarten"},{"driver":"hcae2","id":"23412960","key":"FFEEDDCCBBAA00998877665544332211","name":"HCA_EG_Wohnzimmer"},{"driver":"hcae2","id":"23412961","key":"FFEEDDCCBBAA00998877665544332211","name":"HCA_EG_Küche"},{"driver":"hcae2","id":"23412962","key":"FFEEDDCCBBAA00998877665544332211","name":"HCA_EG_Schlafzimmer"},{"driver":"hcae2","id":"23412963","key":"FFEEDDCCBBAA00998877665544332211","name":"HCA_EG_Bad"},{"driver":"hcae2","id":"23412964","key":"FFEEDDCCBBAA00998877665544332211","name":"HCA_EG_Arbeitszimmer"},{"driver":"hcae2","id":"23412965","key":"FFEEDDCCBBAA00998877665544332211","name":"HCA_OG_Wohnzimmer"},{"driver":"hcae2","id":"23412966","key":"FFEEDDCCBBAA00998877665544332211","name":"HCA_OG_Küche"},{"driver":"hcae2","id":"23412967","key":"FFEEDDCCBBAA00998877665544332211","name":"HCA_OG_Schlafzimmer"},{"driver":"hcae2","id":"23412968","key":"FFEEDDCCBBAA00998877665544332211","name":"HCA_OG_Bad"},{"driver":"hcae2","id":"23412969","key":"FFEEDDCCBBAA00998877665544332211","name":"HCA_OG_Arbeitszimmer"}]
    [10:21:28] INFO: Syncing wmbusmeters configuration ...
    [10:21:29] INFO: Registering meters ...
    [10:21:29] INFO: Adding meter-0001 ...
    [10:21:29] INFO: Adding meter-0002 ...
    [10:21:29] INFO: Adding meter-0003 ...
    [10:21:29] INFO: Adding meter-0004 ...
    [10:21:29] INFO: Adding meter-0005 ...
    [10:21:30] INFO: Adding meter-0006 ...
    [10:21:30] INFO: Adding meter-0007 ...
    [10:21:30] INFO: Adding meter-0008 ...
    [10:21:30] INFO: Adding meter-0009 ...
    [10:21:30] INFO: Adding meter-0010 ...
    [10:21:30] INFO: Adding meter-0011 ...
    [10:21:31] INFO: Adding meter-0012 ...
    [10:21:31] INFO: Generating MQTT configuration ... 
    [10:21:31] INFO: Broker 192.168.1.101 will be used.
    [10:21:32] ERROR: Got unexpected response from the API: Service not enabled
    [10:21:32] INFO: 
    MQTT Discovery ...
    [10:21:32] INFO:  Adding meter: HCA_Keller ...
    [10:21:33] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/23412691_current_consumption_hca/config
    [10:21:33] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/23412691_rssi_dbm/config
    [10:21:33] INFO:   Add/update topic: homeassistant/binary_sensor/wmbusmeters/23412691_status_ERROR/config
    [10:21:33] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/23412691_timestamp/config
    [10:21:34] INFO:  Adding meter: HCA_EG_Wintergarten ...
    [10:21:34] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/23412690_current_consumption_hca/config
    [10:21:34] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/23412690_rssi_dbm/config
    [10:21:34] INFO:   Add/update topic: homeassistant/binary_sensor/wmbusmeters/23412690_status_ERROR/config
    [10:21:34] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/23412690_timestamp/config
    [10:21:35] INFO:  Adding meter: HCA_EG_Wohnzimmer ...
    [10:21:35] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/23412960_current_consumption_hca/config
    [10:21:35] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/23412960_rssi_dbm/config
    [10:21:35] INFO:   Add/update topic: homeassistant/binary_sensor/wmbusmeters/23412960_status_ERROR/config
    [10:21:36] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/23412960_timestamp/config
    [10:21:36] INFO:  Adding meter: HCA_EG_Küche ...
    [10:21:36] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/23412961_current_consumption_hca/config
    [10:21:36] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/23412961_rssi_dbm/config
    [10:21:37] INFO:   Add/update topic: homeassistant/binary_sensor/wmbusmeters/23412961_status_ERROR/config
    [10:21:37] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/23412961_timestamp/config
    [10:21:37] INFO:  Adding meter: HCA_EG_Schlafzimmer ...
    [10:21:37] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/23412962_current_consumption_hca/config
    [10:21:38] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/23412962_rssi_dbm/config
    [10:21:38] INFO:   Add/update topic: homeassistant/binary_sensor/wmbusmeters/23412962_status_ERROR/config
    [10:21:38] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/23412962_timestamp/config
[... 22 lines cut ...]
    [10:21:43] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/23412967_rssi_dbm/config
    [10:21:43] INFO:   Add/update topic: homeassistant/binary_sensor/wmbusmeters/23412967_status_ERROR/config
    [10:21:44] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/23412967_timestamp/config
    [10:21:44] INFO:  Adding meter: HCA_OG_Bad ...
    [10:21:44] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/23412968_current_consumption_hca/config
    [10:21:44] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/23412968_rssi_dbm/config
    [10:21:45] INFO:   Add/update topic: homeassistant/binary_sensor/wmbusmeters/23412968_status_ERROR/config
    [10:21:45] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/23412968_timestamp/config
    [10:21:45] INFO:  Adding meter: HCA_OG_Arbeitszimmer ...
    [10:21:45] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/23412969_current_consumption_hca/config
    [10:21:46] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/23412969_rssi_dbm/config
    [10:21:46] INFO:   Add/update topic: homeassistant/binary_sensor/wmbusmeters/23412969_status_ERROR/config
    [10:21:46] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/23412969_timestamp/config
    [10:21:46] INFO: MQTT Discovery cleanup...
    [10:21:47] INFO: Starting web configuration service.
    [10:21:47] INFO: Running wmbusmeters ...
    [10:21:48] INFO: Starting Nginx...
    2024/02/16 10:21:48 [notice] 85#85: using the "epoll" event method
    2024/02/16 10:21:48 [notice] 85#85: nginx/1.24.0
    2024/02/16 10:21:48 [notice] 85#85: OS: Linux 6.1.71-haos
    2024/02/16 10:21:48 [notice] 85#85: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
    2024/02/16 10:21:48 [notice] 85#85: start worker processes
    2024/02/16 10:21:48 [notice] 85#85: start worker process 688
    Started auto im871a[00103945] on /dev/ttyUSB1 listening on c1
    telegram=|CE44C5146729412331087A8400C0252F2F_046D152A1032036E6D0000426CFF2C436E95000082016CDE2B83016E000000C2016E000082026E0000C2026E000082036E0000C2036E000082046E0000C2046E000082056E0000C2056E000082066E0000C2066E000082076E0000C2076E000082086E0000C2086E000082096E0000C2096E0000820A6E0000C20A6E0000820B6E0000C20B6E0000820C6E0000C20C6E1100820D6E1F00C20D6E3B00820E6E2A00C20E6E8DFF820F6E2A00C20F6E2100317F00346D002000002F2F2F2F2F|+10

BIBOLV, 2024-02-16:
Duplicate of https://github.com/wmbusmeters/wmbusmeters/issues/1078

CommanderROR9, 2024-02-20:
I seem to be having the same issue since yesterday or so. Started with 1.15 I think.

CommanderROR9, 2024-02-21:
Luckily a full reboot of the HA Server seems to have fixed whatever this was...

closed 2024-05-22 by github-actions[bot]