wmbusmeters/wmbusmeters #1182: Qundis Qheat 5.5R meter walk-by measures off by factor 10

bilbo-b, 2024-02-20:
my Qundis Qheat 5.5R sends 2 kinds of readings according to https://qundis.de/wp-content/uploads/2022/11/Q-heat-5.5-R-IST-Datenblatt-de-221114.pdf page 4 
Sendeverhalten C-Mode
Q walk-by every 112 Seconds
Q AMR every 7,5 Minutes

the AMRs are correctly decoded when forced to use qheat driver, but the walk-bys are off by factor 10, 

the following telegram decodes to "*** 49000000 ("total_energy_consumption_kwh":4.9)", while the AMR readings (and the display of the meter) show 49 kWh.

https://wmbusmeters.org/analyze/5844934475149167463778077975149167934446040dff5f3500823d0000810007c006ffff49000000ff2c000000001f3111000000008000800080008000800080008000800080000000000B002f02fd170000046d390d1432488408:qheat 

the following telegram is a AMR telegram, decoding correctly to 49 kWh:

414493447514916746377275149167934446044D000020_0C06490000004C0600000000426CFF2CCC080611000000C2086C1F3102FD170000326CFFFF046D330F1432|+495271

weetmuts, 2024-02-23:
Yep, the walk by telegram has a proprietary manufacturer specific field for which we do not know the unit.
Which totally sucks. We thought it was one type of unit, but apparently it is not always that unit.
If we change the unit, then the existing test cases will break.

The AMR telegram uses proper wmbus standard fields, which always encode the unit.

The original commit 0a7f8a8e2be0f54dee41335059eaf0941d938cf7 was gracefully provided by @stigger 

1) Perhaps there are more bits in the manufacturer specific telegram that tells us the scale/unit?
But someone has to have the documentation to know this.

2) The meter can be configured to send in different units, but we cannot know which unit it uses from telegram.
Then you can add a formula to correct the value, ie add to the meter file:
calculate_total_kwh='total_energy_consumption_kwh*10 factor'
or to the command line
--calculate_total_kwh='total_energy_consumption_kwh*10 factor'

weetmuts, 2024-02-23:
For other meters collected using AMR telegrams you can use: 
--calculate_total_kwh='total_energy_consumption_kwh'
to just copy the correct value to the new field total_kwh.

It seems like it is possible to override the field as well.....
--calculate_total_energy_consumption_kwh='total_energy_consumption_kwh*10 factor'

bilbo-b, 2024-02-23:
my problem is that I can not differentiate between the 2 telegram types. When I define the meter-id, I get both types of telegrams with the same name: 

[image]

the resulting total_energy_consumption_kwh alternates between 4.9 and 49, which makes it unusable to calculate e.g. daily consumption. I would be totally fine completely ignoring walk-by telegrams to get usable results, I just don't see how I can calculate / ignore the walk-bys without doing the same to the AMR-telegrams. Is there an option to do this?

weetmuts, 2024-03-03:
I will not have time to fix the proprietary driver. But I have another fix for you! Let us filter out the non-AMR telegrams.

I have now implemented a much more advanced address selection for the meter templates, which is of course
backwards compatible. Its shoud be in edge now.

If you write an id like: `67911475`
then all telegrams for this id will go to the same meter object. Irrespective of manufacturer, version and type.

If you write an id like: `67911475.T02`
then only telegrams with this id and that have the type=0x02  will be accepted.
The AMR telegrams have this type, so you can use this to select only these telegrams.

Try it!

You can specify even more:
`67911475.M=QDS.V=46.T=02`
but in your case, the only thing that is differen is the type.

At the same time I fixed the primary mbus addressing, so that you can now write p0-p250: 
`wmbusmeters --format=json --pollinterval=5s /dev/ttyUSB1:mbus:2400 TEMP piigth:mbus p0 NOKEY`
to poll mbus meter with primary id 0 on the mbus instead of the full secondary 12345678 address.

You can also create much more complicated rules for example:
wildcards `*.M=ABC.T=16`  all telegrams matching this manufacturer and which are cold water 0x16.
wildcards with filter out `*.M=ABC.T=16,!*.M=ABC.V=44` all those telegrams except those with version 44.
any of these ids '11111111,22222222'
etc.

In your case we could not use the filter-out rule since the proprietary telegram dll looks exactly like the dll in the AMR telegram.

bilbo-b, 2024-03-03:
thank you, Fredrik,
I probably do not yet fully comprehend how to configure it:

I tried with version: 1.16.1-9-g23779cb in HA-addon, but the type does not seem to be different :(

$ wmbus-logs.sh -ft |  grep 67911475
2024-03-03T15:11:37.979033000Z [16:11:37] INFO: [{"driver":"qwater","id":"14707094","key":"NOKEY","name":"KuecheKalt"},{"driver":"qwater","id":"14341598","key":"NOKEY","name":"KuecheWarm"},{"driver":"qwater","id":"14707088","key":"NOKEY","name":"BadKalt"},{"driver":"qwater","id":"14536811","key":"NOKEY","name":"BadWarm"},{"driver":"auto","id":"67911475.T=04","key":"NOKEY","name":"Waerme"}]
2024-03-03T15:11:48.623776000Z [16:11:48] INFO:  Removing topic: homeassistant/sensor/wmbusmeters/67911475.T=37_rssi_dbm/config
2024-03-03T15:11:48.639353000Z [16:11:48] INFO:  Removing topic: homeassistant/sensor/wmbusmeters/67911475.T=37_total_energy_consumption_kwh/config
2024-03-03T15:12:33.341702000Z (meter) created Waerme qheat 67911475.T=04,67911475 encrypted
2024-03-03T15:12:33.343017000Z (meter) used meter template Waerme 67911475.T=04 auto to match 67911475.M=QDS.V=46.T=37,67911475.M=QDS.V=46.T=04
2024-03-03T15:12:33.343361000Z (meter) started meter 3 (Waerme 67911475.T=04 auto) identity mode: id 67911475
2024-03-03T15:12:33.343728000Z (meter) Waerme(3) qheat  handling telegram from 67911475.M=QDS.V=46.T=04
2024-03-03T15:12:33.344039000Z (telegram) DLL L=41 C=44 (from meter SND_NR) M=4493 (QDS) A=67911475 VER=46 TYPE=37 (Radio converter (meter side)) (driver qheat) DEV=im871a[00104228] RSSI=-63
2024-03-03T15:14:09.340660000Z (meter) Not printing this warning again for id: 67911475 mfct: (QDS) Qundis, Germany (0x4493) type: Radio converter (meter side) (0x37) ver: 0x46
2024-03-03T15:14:09.341733000Z (meter) Waerme(3) qheat  handling telegram from 67911475.M=QDS.V=46.T=37
2024-03-03T15:14:09.342033000Z (telegram) DLL L=58 C=44 (from meter SND_NR) M=4493 (QDS) A=67911475 VER=46 TYPE=37 (Radio converter (meter side)) (driver unknown!) DEV=im871a[00104228] RSSI=-64

it shows T=04 and T=37, but no matter if I put "67911475.T=04" into the config-field "id" or "67911475.T=37" still both telegrams get sent to mqtt.

https://wmbusmeters.org/analyze/414493447514916746377275149167934446044E0000200C06490000004C0600000000426CFF2CCC080649000000C2086C1D3202FD170000326CFFFF046D3B0F0333 has the correct value alternating with the wrong value

https://wmbusmeters.org/analyze/5844934475149167463778077975149167934446040dff5f3500824e00000e0007c006ffff49000000ff2c000000001d324900000000800080008000800080008000800080000000000B0026002f02fd170000046d06100333:qheat

one is auto-detected as qheat the other is not, but no matter if I put "auto" or qheat into config for this id, I get alternating values.

weetmuts, 2024-03-03:
Yes, you are right! Fix pushed! Please test edge when it has built.

bilbo-b, 2024-03-04:
Thank you, had to wait until morning as walk-by telegrams are only sent through business hours :)

It works correctly now with .T=04 (for the correct AMR telegrams) or .T=37 (to filter just for the walk-bys off by factor 10)

one side effect:
when not filtering (config: qheat,67911475,NOKEY,Waerme) mqtt-discovery works, when filtering for .T=04 (or other) I do not get mqtt-discovery to work in Home Assistant.

directly specifying the mqtt-sensor in configuration.yaml works.

bilbo-b, 2024-03-04:
probably that's why?

2024-03-04T11:53:56.089240000Z [12:53:56] INFO:  Adding meter: Waerme ...
2024-03-04T11:53:56.476614000Z [12:53:56] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/67911475.T=04_rssi_dbm/config
2024-03-04T11:53:56.746284000Z [12:53:56] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/67911475.T=04_total_energy_consumption_kwh/config
202

I suspect the topic should likely not include the filter expression

closed 2024-06-03 by github-actions[bot]