wmbusmeters/wmbusmeters #1183: wmbusmeters/wmbusmeters:release-1.15.0 missing in Docker Hub

ChristianHeinemann, 2024-02-21:
### Type of request

Bug

### OS version

linux/amd64

### wmbusmeters version

1.15.0

### Your message goes here

The Docker image tag was not created properly. There is a tag "[release-](https://hub.docker.com/layers/wmbusmeters/wmbusmeters/release-/images/sha256-b5b426d585097e55f664c34c4ecc06c32df9cb3d1cfb3d6f8eca626bfee5b892?context=explore)" (without the version suffix) created in [pipeline run](https://github.com/wmbusmeters/wmbusmeters/actions/runs/7905568384) for the [1.15.0](https://github.com/wmbusmeters/wmbusmeters/releases/tag/1.15.0) release, but "release-1.15.0" is missing.

BIBOLV, 2024-02-22:
@testuser7 could you please check?

weetmuts, 2024-02-22:
Yeah, I think it was this line that got lost:
ADD https://api.github.com/repos/wmbusmeters/wmbusmeters/git/refs/heads/master version.json
it has been re-added to wmbusmeters/docker/Dockerfile

Should I build a 1.16.0 release?

testuser7, 2024-02-22:
This join in workflow doesn't seem to work: `join('release-', github.ref_name)`

weetmuts, 2024-02-22:
@testuser7 So it does not have anything to do with the ADD version.json line?

weetmuts, 2024-02-22:
Oh, well, rebuilding with new tag. Perhaps it starts working again.....

testuser7, 2024-02-22:
Fixed: https://github.com/wmbusmeters/wmbusmeters/pull/1186

testuser7, 2024-02-22:
One more PR: https://github.com/wmbusmeters/wmbusmeters/pull/1187

weetmuts, 2024-02-22:
Many thanks @testuser7 ! It looks like 1.16.1 was properly released on dockerhub. Right?

testuser7, 2024-02-22:
Yes, the new tag is correct.

closed 2024-02-23 by weetmuts