wmbusmeters/wmbusmeters #1184: Feature request: expose an env var containing the encrypted telegram

MedaiP90, 2024-02-21:
### Type of request

Enhancement

### OS version

Running in yours docker container

### wmbusmeters version

1.13.1

### Your message goes here

It would be useful to expose an environment variable containing the raw encrypted telegram (the bytes coming from the meters before applying the key provided in the config file). An user can use this data for sending to some API service that manages the decryption process for itself. Suggested name: `$METER_RAW`

MedaiP90, 2024-02-21:
> Yes, this has been asked several times, but I do not have any interest in adding this. It is usually requested by employees at companies that do not want to put knowledge in wmbusmeters on how to decode the telegrams, they just want to use wmbusmeters for its collecting capabilities.
> 
> Why do >you< want it to relay encrypted telegrams?

Unfortunately for me, I would need this feature for the same reason you have mentioned

weetmuts, 2024-02-21:
Hey, I like your courage to admit it straight away! You can 1) change the code yourself  and maintain the patched version for your company or 2) have your company contact me, we might come to an agreement.

Cheers!

MedaiP90, 2024-02-22:
Thankyou @weetmuts, we're going to fork the project and proceed for ourself for now. We'll evalute the possibility of a collaboration if we encounter some difficulty.

closed 2024-02-22 by weetmuts