wmbusmeters/wmbusmeters #1188: Wmbusmeters on Armbian

ondras76, 2024-02-22:
### Type of request

Question

### OS version

Armbian

### wmbusmeters version

1.15.0

### Your message goes here

Hello, can you tell me please where should be the problem?
log
"root@orangepizero:/etc# sudo systemctl start  wmbusmeters
root@orangepizero:/etc# sudo systemctl status  wmbusmeters
× wmbusmeters.service - "wmbusmeters service"
     Loaded: loaded (/lib/systemd/system/wmbusmeters.service; disabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Thu 2024-02-22 21:06:57 CET; 4s ago
       Docs: https://github.com/wmbusmeters/wmbusmeters
             man:wmbusmeters(1)
    Process: 1827 ExecStartPre=/bin/mkdir -p /var/lib/wmbusmeters/meter_readings (code=exited, status=0/SUCCESS)
    Process: 1828 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/lib/wmbusmeters/meter_readings (code=exited, status=0/SUCCESS)
    Process: 1829 ExecStartPre=/bin/mkdir -p /var/log/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 1830 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/log/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 1831 ExecStartPre=/bin/mkdir -p /run/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 1832 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /run/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 1833 ExecStart=/usr/sbin/wmbusmetersd /run/wmbusmeters/wmbusmeters.pid (code=exited, status=0/SUCCESS)
   Main PID: 1834 (code=exited, status=1/FAILURE)
        CPU: 382ms

Feb 22 21:06:57 orangepizero systemd[1]: wmbusmeters.service: Scheduled restart job, restart counter is at 3.
Feb 22 21:06:57 orangepizero systemd[1]: Stopped "wmbusmeters service".
Feb 22 21:06:57 orangepizero systemd[1]: wmbusmeters.service: Start request repeated too quickly.
Feb 22 21:06:57 orangepizero systemd[1]: wmbusmeters.service: Failed with result 'exit-code'.
Feb 22 21:06:57 orangepizero systemd[1]: Failed to start "wmbusmeters service".
root@orangepizero:/etc# "

BIBOLV, 2024-03-15:
Hi,
Try setting loglevel to debug and try to start and check logs for clues.

ondras76, 2024-03-18:
> Hi,
> 
> Try setting loglevel to debug and try to start and check logs for clues. 

thans, I'll try.

ondras76, 2024-03-20:
> Hi, Try setting loglevel to debug and try to start and check logs for clues.
it helps, I found the problem. 
Just a question, I don't see any message on MQTT explorer. Can you tell me witch command to use.
1/ shell=/usr/bin/mosquitto_pub -h 10.0.0.120 -t wmbusmeters/ -m ""
or
2/ shell=mosquitto_pub -h 10.0.0.120 -p 1883 -u xxx -P xxx -t "wmbusmeters/$METER_NAME" "$METER_JSON"

I use both, but nothing. it is necessary to use user and pass.

I can see telegram if I use command in terminal, but if I running like deamon I dont see, 

debug:
"(wmbusmeters) logging started 2024-03-20 15:52:05 1.16.1-2-gc21efd1
(wmbusmeters) version: 1.16.1-2-gc21efd1
(config) store meter files in: "/var/lib/wmbusmeters/meter_readings"
(config) using device: rtlwmbus
(config) number of meters: 1
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers

Usage:   -f frequency_to_tune_to [Hz]
        [-s samplerate (default: 2048000 Hz)]
        [-d device_index (default: 0)]
        [-g gain (default: 0 for auto)]
        [-p ppm_error (default: 0)]
        [-b output_block_size (default: 16 * 16384)]
        [-n number of samples to read (default: 0, infinite)]
        [-S force sync output (default: async)]
        [-D enable direct sampling (default: off)]
        filename (a '-' dumps samples to stdout)

<<<
(shell) return code 1
(shell) 886 exited
(main) rtl_srd found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(serial) EVENT thread interrupted
(serial) EVENT thread interrupted
(shell) output: >>><<<
(shell) return code 0
(shell) 893 exited
(main) rtl_wmbus found in path
(main) rtlsdr device 00000001 not currently used.
(main) found specified device (rtlwmbus) that matches detected device (:rtlwmbus[00000001]:0/0)
(main) opening rtlwmbus
[2024-03-20_15:52:06] Started config rtlwmbus[00000001] listening on none
[2024-03-20_15:52:06] (rtlwmbus) error: when starting as daemon, wmbusmeters looked for /rtl_sdr and /usr/bin/rtl_sdr, but found neither!
(serial) stopping manager
(wmbusmeters) logging started 2024-03-20 15:52:07 1.16.1-2-gc21efd1
(wmbusmeters) version: 1.16.1-2-gc21efd1
(config) store meter files in: "/var/lib/wmbusmeters/meter_readings"
(config) using device: rtlwmbus
(config) number of meters: 1
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers"

Thank you for your help.

closed 2024-06-19 by github-actions[bot]