wmbusmeters/wmbusmeters #1208: DIEHL METERING HELP

IAmOrion, 2024-03-12:
I moved into a new build back in Feb 2023.  The water meter installed is a **DIEHL METERING IZAR RC 868 i W R4** from what I can gather (blocked out meter serial number)

[image]

[image]

I've tried using ESPHome ESP32 + CC1101 setups, I've tried DIY NanoCUL using multiple firmwares, Also tried a Nano Every with the everycul firmware as well as using wmbusmetering directly using Fedora 39 and for the life of me I can not get any information and I'm max 2 meters away from the device.

I've tried 5 different CC1101 modules and triple checked wiring each time.  2 x CC1101 cheap modules from AliExpress, 1 x CC1101 cheap module from Amazon UK, 1 x CC1101 cheap module from eBay UK, 1 x E07 900m10s from Amazon UK

[https://www.diehl.com/cms/files/IZAR-RC-868-i-W-R4-2.pdf](https://www.diehl.com/cms/files/IZAR-RC-868-i-W-R4-2.pdf)

The builtin LED (not to be mistaken with the power led) on the Arduino (when trying NanoCUL firmwares) starts flashing away once the commands activate/trigger it and remains flashing indefinitely 

I've tried these commands but they just sit and do nothing forever after the initial init stuff.  

`
sudo wmbusmeters --debug --format=json auto:t1 MyWater izar '*' NOKEY
sudo wmbusmeters --debug --format=json auto:c1 MyWater izar '*' NOKEY
sudo wmbusmeters --debug auto:c1
sudo wmbusmeters --debug auto:t1
`

T1 gives me this:

`
[sudo] password for iamorion: 
(drivers) added builtin driver elster
(drivers) added builtin driver iperl
(wmbusmeters) version: rpm-build_1.16.1
(config) number of meters: 1
(main) device /dev/ttyS0 not currently used, detect contents...
(serial) check if /dev/ttyS0 can be accessed
(serial) tty /dev/ttyS0 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyS0 speed(B115200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS|CD|DSR)
(serial) after config:  /dev/ttyS0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS|CD|DSR)
(serialtty) opened /dev/ttyS0 fd 3 (detect amb8465/amb3665)
(serial) received binary ""
(amb8465/3665) sending 6 bytes attempt 0
(serial /dev/ttyS0) sent "FF0A02008077"
(serial) EVENT thread interrupted
(amb8465/3665) sent 6 bytes OK
(amb8465/3665) reading response... 1
(serial) received binary ""
(amb8465/3665) reading response... 2
(serial) received binary ""
(amb8465/3665) reading response... 3
(serial) received binary ""
(amb8465/3665) are you there? no.
(serial) EVENT thread interrupted
(serialtty) closed /dev/ttyS0 (detect amb8465/amb3665)
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyS0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS|CD|DSR)
(serial) after config:  /dev/ttyS0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS|CD|DSR)
(serialtty) opened /dev/ttyS0 fd 3 (detect im871a)
(serial) received binary ""
(serial /dev/ttyS0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary ""
(im871a/im170a) are you there? no.
(serial) EVENT thread interrupted
(serialtty) closed /dev/ttyS0 (detect im871a)
(serial) before config: /dev/ttyS0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS|CD|DSR)
(serial) EVENT thread interrupted
(serial) after config:  /dev/ttyS0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS|CD|DSR)
(serialtty) opened /dev/ttyS0 fd 3 (detect rc1180)
(serial /dev/ttyS0) sent "00"
(serial) EVENT thread interrupted
(serial) received binary ""
[... 246 lines cut ...]
Started auto cul on /dev/ttyUSB0 listening on c1
(cul) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (cul)
(main) regular reset of cul  on /dev/ttyUSB0 will happen every 82800 seconds
(cul) set link mode c
(serial /dev/ttyUSB0) sent "6272630D0A"
(serial) EVENT thread interrupted
(serial) received binary "434D4F44450D0A"
(cul) checkCULFrame "CMODE<0D><0A>"
(cul) no leading 'b' so it is text and no frame
(cul) received "CMODE"(serial /dev/ttyUSB0) sent "5832310D0A"
(wmbus) no alarm (expected activity) for cul
(serial) EVENT thread interrupted
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop

`

And again will sit for hours and never give any telegram id's 

What am I doing wrong!? It's driving me insane!

closed 2024-03-14 by IAmOrion

labuwx, 2025-02-18:
@IAmOrion If I may ask, is your meter from IWNL? If so, did you manage to receive a reading?

IAmOrion, 2025-02-18:
> [@IAmOrion](https://github.com/IAmOrion) If I may ask, is your meter from IWNL? If so, did you manage to receive a reading?

SInce I have no idea what IWNL is I'm gonna presume no ha.

FWIW I changed the 868mhz board I was using for the E07 module on AliExpress and that got it working for me.
I've now got it connected via ESPHome so the meter reading is in my Home Assistant dashboard

labuwx, 2025-02-18:
Thank you for the information.

IWNL = Independent Water Networks Limited, a UK water company
Their meters and transmitters (same as yours) are available on eBay, but before making a purchase, I would like to find out if I could decode their transmission.

IAmOrion, 2025-02-18:
> Thank you for the information.
> 
> IWNL = Independent Water Networks Limited, a UK water company Their meters and transmitters (same as yours) are available on eBay, but before making a purchase, I would like to find out if I could decode their transmission.

Ahh ok, I see.  I have no idea who installed the meters, I live in a new build area (well, 2 years old now) and so all properties here were fitted with them by the contractors/builders as they were built.

This is what I used with mine: https://vi.aliexpress.com/item/1005003256614835.html

[image]

If it's any help to you - this is my standalone arduino code.

[Diehl_868MHz_Water_Meter_Reader_v3.zip](https://github.com/user-attachments/files/18855640/Diehl_868MHz_Water_Meter_Reader_v3.zip)

[SmartRC-CC1101-Driver-Lib-JT.zip](https://github.com/user-attachments/files/18855626/SmartRC-CC1101-Driver-Lib-JT.zip)