wmbusmeters/wmbusmeters #1215: Apator NA-1 driver

gitwoy, 2024-03-15:
I want back to 
[https://github.com/wmbusmeters/wmbusmeters/issues/793](https://github.com/wmbusmeters/wmbusmeters/issues/793)

Mine is 
Apator NA-1 water meter
Meter ID 0x06414357
key: 00000000000000000000000000000000 
unknown water counter readings, but for sure will not correspond with wmbus readings. Below is the actual time of the readings, so the counter should be identical or with a slight increase.

09:35:32 | [D] | [wmbus:176] | Meter ID [0x06414357] RSSI: -70 dBm LQI: 128 Frame: T1 A not found in configuration T: 1C440106574341061407A1E3004B277391923EE776069C110CE6F29BCB (29)
09:36:02 | [D] | [mbus:035] | Processing T1 A frame
09:36:02 | [D] | [wmbus:176] | Meter ID [0x06414357] RSSI: -74 dBm LQI: 128 Frame: T1 A not found in configuration T: 1C440106574341061407A1E6007BC6D26699199EDB64A663ABBBDB7238 (29)
09:36:11 | [D] | [mbus:035] | Processing T1 A frame
09:36:11 | [D] | [wmbus:176] | Meter ID [0x06414357] RSSI: -75 dBm LQI: 129 Frame: T1 A not found in configuration T: 1C440106574341061407A0E700474AF1742F5394F05F36AC1A6D4CC005 (29)
09:36:21 | [D] | [mbus:035] | Processing T1 A frame
09:36:22 | [D] | [wmbus:176] | Meter ID [0x06414357] RSSI: -74 dBm LQI: 128 Frame: T1 A not found in configuration T: 1C440106574341061407A0E800DC25BF6DAC4FD576D39FE6ED9056D0C0 (29)
09:36:51 | [D] | [mbus:035] | Processing T1 A frame
09:36:51 | [D] | [wmbus:176] | Meter ID [0x06414357] RSSI: -85 dBm LQI: 128 Frame: T1 A not found in configuration T: 1C440106574341061407A0EB006A5C5DB9AFDFF6F25111456769D71FA4 (29)
09:37:01 | [D] | [mbus:035] | Processing T1 A frame
09:37:01 | [D] | [wmbus:176] | Meter ID [0x06414357] RSSI: -63 dBm LQI: 128 Frame: T1 A not found in configuration T: 1C440106574341061407A1EC00FB831D6A1C5532022ED86262BD62A9CB

kzajac83, 2024-03-15:
How do you know that the key is all zeros?

Maybe try to contact Apator Telemetria from Słupsk (producer of all the overlays) about key or more information, this is division Apator Powogaz (water, heat gas meters) from Poznań/Jaryszki, I try for wired overlay but hard get some information but maybe you will be lucky.

gitwoy, 2024-03-15:
Up to now all Apator's overlays (other than this one) I'm using have zeros.

gitwoy, 2024-03-18:
According support the key is generated based of the serial no and for this overlay is B3E933C6F2681EA66CE76D082A057153.
But even with this key none of the drivers work.

kzajac83, 2024-03-25:
Last days bought USB dongle with E07-900M10S chip (CC1101, inter alia to read 868MHz band) and found my APT-WMBUS-NA-1 main water meter.    

Telegram from 24.03.2024 below:

```
[16:26:21][D][wmbus:182]: Meter ID [0x04913581] RSSI: -75 dBm LQI: 128 Frame: T1 A not found in configuration T: 1C440106813591041407A0B000266A705474DDB80D9A0EB9AE2EF29D96 (29)

[16:26:50][D][wmbus:182]: Meter ID [0x04913581] RSSI: -75 dBm LQI: 128 Frame: T1 A not found in configuration T: 1C440106813591041407A0B3003FA550900BDAF3D49FD2A13CA81A61ED (29)

[16:26:59][D][wmbus:182]: Meter ID [0x04913581] RSSI: -77 dBm LQI: 128 Frame: T1 A not found in configuration T: 1C440106813591041407A1B4009C87A8034F02E25E3CC161FC171A2D39 (29)

[16:27:19][D][wmbus:182]: Meter ID [0x04913581] RSSI: -75 dBm LQI: 128 Frame: T1 A not found in configuration T: 1C440106813591041407A0B600E63F50206F4062F253D80C4F762CAB51 (29)

[16:28:17][D][wmbus:182]: Meter ID [0x04913581] RSSI: -75 dBm LQI: 128 Frame: T1 A not found in configuration T: 1C440106813591041407A0BC002CA9A55981BE989CDBA3326154583F6B (29)

[16:30:04][D][wmbus:182]: Meter ID [0x04913581] RSSI: -76 dBm LQI: 128 Frame: T1 A not found in configuration T: 1C440106813591041407A0C700A1F6918F83A34FC2486AA4F57A57E9B9 (29)

```

ID 04913581 and from support.powogaz@apator.com got AES key 

```
1F25361CDB14B6BC370CD77B2EC6517D

1F 25 36 1C DB 14  B6 BC 37 0C D7 7B 2E C6 51 7D

```

but results the same, somehow decoded:

https://wmbusmeters.org/analyze/1C440106813591041407A0B000266A705474DDB80D9A0EB9AE2EF29D96:auto:1F25361CDB14B6BC370CD77B2EC6517D

PS. What is interested the overlay have NFC and Android app from NXP "TagWriter" cannot read tag (phone read but data probably is also encrypted because no data visible).

kzajac83, 2024-03-26:
Nothing tell me this values, for my example telegrams there was about 345.19 m3 on counter for 24.03.2024 , in below table the same telegram plus some extra telegrams for midnight 25.03.2024 (unfortunately don't know what is value m3 for this date).

In table below 1st 11. bytes are (w)m-bus header of the telegram, bytes 12-29 is manufacturer specification.  Also 11th byte sometime "A0" and another time "A1". 

Can you tell me how you extract values "00E3 00E6 00E7 00EB 00EC" ?

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/legion/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/legion/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

  | 0 | 1 | 2 | 3 | 4 | 5 | 6 | 7 | 8 | 9 | 10 | 11 | 12 | 13 | 14 | 15 | 16 | 17 | 18 | 19 | 20 | 21 | 22 | 23 | 24 | 25 | 26 | 27 | 28 | 29
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
2024-03-24T15:26:21Z | 1C440106813591041407A0B000266A705474DDB80D9A0EB9AE2EF29D96 | 1C | 44 | 01 | 06 | 81 | 35 | 91 | 04 | 14 | 07 | A0 | B0 | 00 | 26 | 6A | 70 | 54 | 74 | DD | B8 | 0D | 9A | 0E | B9 | AE | 2E | F2 | 9D | 96
2024-03-24T15:26:50Z | 1C440106813591041407A0B3003FA550900BDAF3D49FD2A13CA81A61ED | 1C | 44 | 01 | 06 | 81 | 35 | 91 | 04 | 14 | 07 | A0 | B3 | 00 | 3F | A5 | 50 | 90 | 0B | DA | F3 | D4 | 9F | D2 | A1 | 3C | A8 | 1A | 61 | ED
2024-03-24T15:26:59Z | 1C440106813591041407A1B4009C87A8034F02E25E3CC161FC171A2D39 | 1C | 44 | 01 | 06 | 81 | 35 | 91 | 04 | 14 | 07 | A1 | B4 | 00 | 9C | 87 | A8 | 03 | 4F | 02 | E2 | 5E | 3C | C1 | 61 | FC | 17 | 1A | 2D | 39
2024-03-24T15:27:19Z | 1C440106813591041407A0B600E63F50206F4062F253D80C4F762CAB51 | 1C | 44 | 01 | 06 | 81 | 35 | 91 | 04 | 14 | 07 | A0 | B6 | 00 | E6 | 3F | 50 | 20 | 6F | 40 | 62 | F2 | 53 | D8 | 0C | 4F | 76 | 2C | AB | 51
2024-03-24T15:28:17Z | 1C440106813591041407A0BC002CA9A55981BE989CDBA3326154583F6B | 1C | 44 | 01 | 06 | 81 | 35 | 91 | 04 | 14 | 07 | A0 | BC | 00 | 2C | A9 | A5 | 59 | 81 | BE | 98 | 9C | DB | A3 | 32 | 61 | 54 | 58 | 3F | 6B
2024-03-24T15:30:04Z | 1C440106813591041407A0C700A1F6918F83A34FC2486AA4F57A57E9B9 | 1C | 44 | 01 | 06 | 81 | 35 | 91 | 04 | 14 | 07 | A0 | C7 | 00 | A1 | F6 | 91 | 8F | 83 | A3 | 4F | C2 | 48 | 6A | A4 | F5 | 7A | 57 | E9 | B9
2024-03-25T23:16:39Z | 1C440106813591041407A00400B43972592028BDAF1873A9F93E214C0B | 1C | 44 | 01 | 06 | 81 | 35 | 91 | 04 | 14 | 07 | A0 | 04 | 00 | B4 | 39 | 72 | 59 | 20 | 28 | BD | AF | 18 | 73 | A9 | F9 | 3E | 21 | 4C | 0B
2024-03-25T23:19:40Z | 1C440106813591041407A007007FEFD0A82910610A927AC951DAABB6C5 | 1C | 44 | 01 | 06 | 81 | 35 | 91 | 04 | 14 | 07 | A0 | 07 | 00 | 7F | EF | D0 | A8 | 29 | 10 | 61 | 0A | 92 | 7A | C9 | 51 | DA | AB | B6 | C5
2024-03-25T23:22:41Z | 1C440106813591041407A00A00E92BE6146D1AEC80EB59B53484C1FFEB | 1C | 44 | 01 | 06 | 81 | 35 | 91 | 04 | 14 | 07 | A0 | 0A | 00 | E9 | 2B | E6 | 14 | 6D | 1A | EC | 80 | EB | 59 | B5 | 34 | 84 | C1 | FF | EB
2024-03-25T23:25:41Z | 1C440106813591041407A00D001851DB180314A9CFB2A0FEF15425D21E | 1C | 44 | 01 | 06 | 81 | 35 | 91 | 04 | 14 | 07 | A0 | 0D | 00 | 18 | 51 | DB | 18 | 03 | 14 | A9 | CF | B2 | A0 | FE | F1 | 54 | 25 | D2 | 1E
2024-03-25T23:26:41Z | 1C440106813591041407A10E00F10E584D50811D2DBB1C643B489828D4 | 1C | 44 | 01 | 06 | 81 | 35 | 91 | 04 | 14 | 07 | A1 | 0E | 00 | F1 | 0E | 58 | 4D | 50 | 81 | 1D | 2D | BB | 1C | 64 | 3B | 48 | 98 | 28 | D4
2024-03-25T23:27:41Z | 1C440106813591041407A00F00C29392E42D9DAEA6686D1BF5D150186C | 1C | 44 | 01 | 06 | 81 | 35 | 91 | 04 | 14 | 07 | A0 | 0F | 00 | C2 | 93 | 92 | E4 | 2D | 9D | AE | A6 | 68 | 6D | 1B | F5 | D1 | 50 | 18 | 6C
2024-03-25T23:?Z+60s | 1C440106813591041407A10E00F10E584D50811D2DBB1C643B489828D4 | 1C | 44 | 01 | 06 | 81 | 35 | 91 | 04 | 14 | 07 | A1 | 0E | 00 | F1 | 0E | 58 | 4D | 50 | 81 | 1D | 2D | BB | 1C | 64 | 3B | 48 | 98 | 28 | D4
2024-03-25T23:?Z+60s | 1C440106813591041407A00F00C29392E42D9DAEA6686D1BF5D150186C | 1C | 44 | 01 | 06 | 81 | 35 | 91 | 04 | 14 | 07 | A0 | 0F | 00 | C2 | 93 | 92 | E4 | 2D | 9D | AE | A6 | 68 | 6D | 1B | F5 | D1 | 50 | 18 | 6C
2024-03-25T23:?Z+60s | 1C440106813591041407A022004658CA87A76B5B37069B26BD49F975E2 | 1C | 44 | 01 | 06 | 81 | 35 | 91 | 04 | 14 | 07 | A0 | 22 | 00 | 46 | 58 | CA | 87 | A7 | 6B | 5B | 37 | 06 | 9B | 26 | BD | 49 | F9 | 75 | E2
2024-03-25T23:?Z+60s | 1C440106813591041407A024008403BA640D80016D0C71302EA96A43D3 | 1C | 44 | 01 | 06 | 81 | 35 | 91 | 04 | 14 | 07 | A0 | 24 | 00 | 84 | 03 | BA | 64 | 0D | 80 | 01 | 6D | 0C | 71 | 30 | 2E | A9 | 6A | 43 | D3

</body>

</html>

kzajac83, 2024-03-27:
Made one step forward, successfully encode AES 128 data from manufacturer spec. block. 

General bytes in my cause:
1-10 -  (w)M-Bus header
11 - values A0 or A1 only, part of header? Maybe alternative values every send telegram??
12 - Manf block - counter sent telegrams. Every telegram inc value +1 to FF and jump to 00
13 - Manuf block - always zeros 
14-29 - AES 128 decoded, KEY=16 x double zeros (KEY not from producer)

Encoded 14-29 bytes in last column. Now this block have sense, need focus on in below table.  

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/legion/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/legion/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

14 | 15 | 16 | 17 | 18 | 19 | 20 | 21 | 22 | 23 | 24 | 25 | 26 | 27 | 28 | 29
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
2E | 06 | CF | 61 | 91 | 28 | A4 | 96 | 9D | 91 | B9 | 15 | 9B | B0 | B0 | 30
2E | 05 | CF | 61 | 91 | 4D | A4 | 96 | 9E | 92 | BB | 15 | 98 | B3 | B3 | 33
2E | 02 | CF | 61 | 91 | 57 | A4 | 96 | 99 | 95 | BD | 12 | 9F | B4 | B4 | 34
2E | 01 | CF | 61 | 91 | 62 | A4 | 96 | 9B | 97 | BE | 11 | 9D | B6 | B6 | 36
2E | 08 | CF | 61 | 91 | A4 | A4 | 96 | 91 | 9D | B7 | 14 | 97 | BC | BC | 3C
2E | 06 | CE | 61 | 91 | 08 | A5 | 96 | EA | E6 | CE | 6E | EC | C7 | C7 | 47
2E | 0B | CD | 60 | 91 | 61 | D5 | 93 | 29 | 5D | 0E | A4 | 2B | 04 | 04 | 84
2E | 09 | CD | 60 | 91 | 1E | D6 | 93 | 2A | 5E | 0F | A6 | 28 | 07 | 07 | 87
2E | 06 | CC | 60 | 91 | CB | D6 | 93 | 27 | 53 | 03 | AB | 25 | 0A | 0A | 8A
2E | 06 | CC | 60 | 91 | 87 | D7 | 93 | 20 | 54 | 04 | AC | 22 | 0D | 0D | 8D
2E | 06 | CC | 60 | 91 | BB | D7 | 93 | 23 | 57 | 07 | AF | 21 | 0E | 0E | 8E
2E | 06 | CC | 60 | 91 | FF | D7 | 93 | 22 | 56 | 06 | AE | 20 | 0F | 0F | 8F
2E | 06 | CC | 60 | 91 | BB | D7 | 93 | 23 | 57 | 07 | AF | 21 | 0E | 0E | 8E
2E | 06 | CC | 60 | 91 | FF | D7 | 93 | 22 | 56 | 06 | AE | 20 | 0F | 0F | 8F
2E | 05 | CC | 60 | 91 | 6E | DC | 93 | 0F | 7B | 2A | 80 | 0D | 22 | 22 | A2
2E | 05 | CC | 60 | 91 | E5 | DC | 93 | 09 | 7D | 2C | 86 | 0B | 24 | 24 | A4
2E | 0B | 46 | 60 | 91 | FE | 12 | 91 | E9 | B4 | CE | 48 | F5 | C4 | C4 | 44
2E | 0B | 46 | 60 | 91 | 31 | 13 | 91 | E8 | B5 | CF | 49 | F4 | C5 | C5 | 45
2E | 0B | 46 | 60 | 91 | 24 | 1C | 91 | E4 | B9 | C3 | 45 | F8 | C9 | C9 | 49
2E | 0B | 46 | 60 | 91 | 5F | 1C | 91 | E7 | BA | C0 | 46 | FB | CA | CA | 4A
2E | 0B | 46 | 60 | 91 | 92 | 1C | 91 | E6 | BB | C1 | 47 | FA | CB | CB | 4B
2E | 0B | 46 | 60 | 91 | D5 | 1C | 91 | E1 | BC | C6 | 40 | FD | CC | CC | 4C
2E | 0B | 46 | 60 | 91 | 0F | 1D | 91 | E0 | BD | C7 | 41 | FC | CD | CD | 4D
2E | 0B | 46 | 60 | 91 | 42 | 1D | 91 | E3 | BE | C4 | 42 | FF | CE | CE | 4E
2E | 0B | 46 | 60 | 91 | B8 | 1D | 91 | FD | A0 | DA | 5C | E1 | D0 | D0 | 50
2E | 09 | 46 | 60 | 91 | 37 | 1E | 91 | FF | A2 | DA | 5F | E3 | D2 | D2 | 52
2E | 0A | 49 | 60 | 91 | 92 | 1F | 91 | F5 | A8 | D1 | 48 | E9 | D8 | D8 | 58
2E | 0A | 49 | 60 | 91 | D5 | 1F | 91 | F4 | A9 | D0 | 49 | E8 | D9 | D9 | 59
2E | 0A | 49 | 60 | 91 | 08 | 18 | 91 | F7 | AA | D3 | 4A | EB | DA | DA | 5A
2E | 0A | 49 | 60 | 91 | FE | 18 | 91 | F3 | AE | D7 | 4E | EF | DE | DE | 5E
2E | 0A | 49 | 60 | 91 | 31 | 19 | 91 | F2 | AF | D6 | 4F | EE | DF | DF | 5F

</body>

</html>

And all data in tables:
[... 33 lines cut ...]
2024-03-25T23:26:41Z | 1C440106813591041407A10E00F10E584D50811D2DBB1C643B489828D4 | 1C | 44 | 01 | 06 | 81 | 35 | 91 | 04 | 14 | 07 | A1 | 0E | 00 | F1 | 0E | 58 | 4D | 50 | 81 | 1D | 2D | BB | 1C | 64 | 3B | 48 | 98 | 28 | D4 | F10E584D50811D2DBB1C643B489828D4 | 2E06CC6091BBD793235707AF210E0E8E | 2E | 06 | CC | 60 | 91 | BB | D7 | 93 | 23 | 57 | 07 | AF | 21 | 0E | 0E | 8E
2024-03-25T23:27:41Z | 1C440106813591041407A00F00C29392E42D9DAEA6686D1BF5D150186C | 1C | 44 | 01 | 06 | 81 | 35 | 91 | 04 | 14 | 07 | A0 | 0F | 00 | C2 | 93 | 92 | E4 | 2D | 9D | AE | A6 | 68 | 6D | 1B | F5 | D1 | 50 | 18 | 6C | C29392E42D9DAEA6686D1BF5D150186C | 2E06CC6091FFD793225606AE200F0F8F | 2E | 06 | CC | 60 | 91 | FF | D7 | 93 | 22 | 56 | 06 | AE | 20 | 0F | 0F | 8F
2024-03-25T23:?Z + 60s | 1C440106813591041407A10E00F10E584D50811D2DBB1C643B489828D4 | 1C | 44 | 01 | 06 | 81 | 35 | 91 | 04 | 14 | 07 | A1 | 0E | 00 | F1 | 0E | 58 | 4D | 50 | 81 | 1D | 2D | BB | 1C | 64 | 3B | 48 | 98 | 28 | D4 | F10E584D50811D2DBB1C643B489828D4 | 2E06CC6091BBD793235707AF210E0E8E | 2E | 06 | CC | 60 | 91 | BB | D7 | 93 | 23 | 57 | 07 | AF | 21 | 0E | 0E | 8E
2024-03-25T23:?Z + 60s | 1C440106813591041407A00F00C29392E42D9DAEA6686D1BF5D150186C | 1C | 44 | 01 | 06 | 81 | 35 | 91 | 04 | 14 | 07 | A0 | 0F | 00 | C2 | 93 | 92 | E4 | 2D | 9D | AE | A6 | 68 | 6D | 1B | F5 | D1 | 50 | 18 | 6C | C29392E42D9DAEA6686D1BF5D150186C | 2E06CC6091FFD793225606AE200F0F8F | 2E | 06 | CC | 60 | 91 | FF | D7 | 93 | 22 | 56 | 06 | AE | 20 | 0F | 0F | 8F
2024-03-25T23:?Z + 60s | 1C440106813591041407A022004658CA87A76B5B37069B26BD49F975E2 | 1C | 44 | 01 | 06 | 81 | 35 | 91 | 04 | 14 | 07 | A0 | 22 | 00 | 46 | 58 | CA | 87 | A7 | 6B | 5B | 37 | 06 | 9B | 26 | BD | 49 | F9 | 75 | E2 | 4658CA87A76B5B37069B26BD49F975 [... cut ...] 091FE1891F3AED74EEFDEDE5E | 2E | 0A | 49 | 60 | 91 | FE | 18 | 91 | F3 | AE | D7 | 4E | EF | DE | DE | 5E
2024-03-26T22:52:24Z | 1C440106813591041407A0DF0048911DE141FBB0AB5E224172DE9B4E7F | 1C | 44 | 01 | 06 | 81 | 35 | 91 | 04 | 14 | 07 | A0 | DF | 00 | 48 | 91 | 1D | E1 | 41 | FB | B0 | AB | 5E | 22 | 41 | 72 | DE | 9B | 4E | 7F | 48911DE141FBB0AB5E224172DE9B4E7F | 2E0A496091311991F2AFD64FEEDFDF5F | 2E | 0A | 49 | 60 | 91 | 31 | 19 | 91 | F2 | AF | D6 | 4F | EE | DF | DF | 5F

</body>

</html>

abusca, 2024-03-27:
Hi @kzajac83

I have an APT-WMBUS-NA-1 connected to a meter that I can manipulate. I can paste my read telegrams if it helps.

kzajac83, 2024-03-27:
@abusca 
Today upgrade my excel spreadsheet to easier encode telegrams (earlier it was long manual work) so please send your telegrams, I will easy encode manufacturing specification part. I assume key = 16. zeros, if will be different I will let you know.
Additional useful will be timestamp (data + time), value m3 (or liters), flow condition (on / off), magnets put or not (overlay recognize magnetic field) to each telegram. Maybe based on this information will be possible recognize encoded manuf. spec.  My overlay is not my and I have hard access to it (in cellar) so your data will be helpful.  

Have you maybe access to Inkasoid/Inkasent Apator app PC/Android? 
.   
BTW. Because other Apator's overlay has been recognize then maybe answer is in other Apator's drivers, not checking this thread.

kzajac83, 2024-03-27:
PS. Many useful information is stored in specification: https://api.apator.com/uploads/oferta/woda-i-cieplo/systemy/radiowy/apt-wmbus-na-1/apt-wmbus-na-1-manual.pdf  

[apt-wmbus-na-1-manual.pdf](https://github.com/wmbusmeters/wmbusmeters/files/14780389/apt-wmbus-na-1-manual.pdf)

eg. regarding encryption

> All RF data communication with the module is encrypted

> The encryption of data exchanged over ISM 868 MHz RF transmission is based on AES-CBC with
> a static 128-bit encryption key with a dynamic trigger vector. To enable correct data communication
> with the universal RF module, the software/firmware of the external device must use the encryption
> key assigned to the same module.

> Unidirectional (T1): metered usage (actual reading + 1 historical reading
> (the last saved one)); event flags

and the same in other part of document:

> RF interface-based data reading and writing
> The universal RF module transmits a spontaneous T1 data frame every 10 seconds between 5:00
> and 21:00 and every 60 seconds between 21:00 and 5:00. **The data frame contents follow:
> - Actual water meter indication and actual device date and time;
> - The last historical water meter indication with its save date and time;
> - Event flags (see details further in the Manual).**

> The universal RF module can store up to 12 previous water meter indications, and the storage
> period is configurable (by choosing between one week, one month, or one year). Each
> spontaneous data frame (T1 mode) includes the last water meter indication. The full history of
> readings can be read (and includes 12 historical indications) in T2 mode (bidirectional
> communication) if triggered by an RF command or over NFC

Can try NFC but probably Inkasoid app is need to encode, when I try read NFC 3th part app then nothing read (there was "beep" for read but app nothing read). Via NFC you can check:

> Some data exchanged over NFC is not encrypted, which includes:
> - Actual water meter indication
> - Actual device date and time
> - The last 12 historical water meter indications with their save date and time
> - Event flags
> - Event details
> - Device configuration

> The following data is encrypted and requires an encryption key for processing:
> - Diagnostic data

There are also information events and type of it.

abusca, 2024-03-27:
I'm using a modified version of wmbusmeters, because I'm developing an standalone embedded data recorder, so following logs may not follow standard wmbusmeters logs.

The water meter is not tied to any pipe. I have it in my hands, so I need to blow in order to change the reading...

First scenario: APT-WMBUS-NA-1 connected to water meter. Water meter shows 00000,102m3.
```
2024-03-27 20:21:08 --> 1C4401061426170414073D21A1B6002053209A6E49346F69A2B789559D5C384F1DECE300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

2024-03-27 20:21:18 --> 1C4401061426170414073D21A0B700F37D6A28AA9671B41200EFD0A0ABEC9813D0A1E100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

2024-03-27 20:21:37 --> 1C4401061426170414073D21A1B900F1158D30356DCE283ADE9D53134A387EBA59C2BB00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

2024-03-27 20:21:57 --> 1C4401061426170414073D21A0BB00692C2D9104F72ED93A029D9A9B1EF9F66784004600000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
```

Second scenario: APT-WMBUS-NA-1 connected to water meter. Water meter shows 00000,122m3.
```
2024-03-27 20:36:31 --> 1C4401061426170414073D21A1130014CBBA62E87B2E10ECED48F13B3BB4E9260969D200000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

2024-03-27 20:36:51 --> 1C4401061426170414073D21A015008B9B7838D12317CF565B027ECB979F5E6446984E00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

2024-03-27 20:37:01 --> 1C4401061426170414073D21A1160012092099529918A97958995D07FFD75E9FCDE82100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

2024-03-27 20:37:31 --> 1C4401061426170414073D21A11900AA56EABE62B9DCA316330EF7BCEF1FC04006C42D00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
```

Third scenario: APT-WMBUS-NA-1 NOT connected to water meter.
```
2024-03-27 20:42:17 --> 1C4401061426170414073D21A03600EEA0884A7DF9A501BB726D7AE68039AF9AB1EB9600000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

2024-03-27 20:42:27 --> 1C4401061426170414073D21A13700659B027CB08951F7E1783B97AD48476666DEEAFB00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

2024-03-27 20:42:37 --> 1C4401061426170414073D21A03800D7DAF054E0B983A1EAB4B7D443EA69012FF021BE00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

2024-03-27 20:42:47 --> 1C4401061426170414073D21A0390035EBDAE7916E09BD80954409581C22E7D4E4ED3000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
```

jacek27, 2024-03-27:
Those devices at least in principle should NOT have a 0 key. While having the proper key, please try to use the ultrimis driver, it should work at least for the current volume.

kzajac83, 2024-03-27:
@abusca  Yes probably not standard, in bytes your telegram long is 35 bytes, I have 29 bytes total. Wmbusmeters decode mfct part to 33 bytes total (last 2 bytes is omitted, maybe to close telegram / communication ??). However this is still 4 bytes longer than my telegram and author of the issue. General I have 16 bytes decoded part, you have 20 bytes.  

Therefore now don't know how encode longer longer telegram than 16 bytes in excel, sorry cannot help now, I need work on it.   

Also is interested why you have longer telegram, I know 100% that my overlay have no errors and because you have removed the overlay then I'm sure appear error flag in your overlay and last flags seems are error flags.

kzajac83, 2024-03-27:
@jacek27 Key is not 0 but "00000000000000000000000000000000" in my cause. Maybe I wrong understood but key is string, not value.  Also when use KEY from supplier then after AES encode got the same nonsense bytes when comparing telegram_n VS telegram_n-1 and telegram_n+1. Additional ultrimis driver not work for me, no any data encoded with KEY from Apator and zeros.

jacek27, 2024-03-28:
@kzajac83 Maybe some clarification is needed, Powogaz had 2 devices named NA-1 [APT-WMBUS-NA1](https://www.apator.com/nasze-rozwiazania/woda-i-cieplo/system-zdalnego-odczytu-mediow/system-radiowy/apt-wmbus-na-1#pliki-do-pobrania) and APT-OMS-NA-1, those are quite different devices. Based on info you provided this thread might not adress your device.

kzajac83, 2024-03-28:
@jacek27 we are speaking about this:

https://api.apator.com/thumbnails/gallery_post_main/uploads/oferta/woda-i-cieplo/systemy/radiowy/apt-wmbus-na-1/apt-wmbus-na-1.jpg

kzajac83, 2024-04-03:
Maybe someone help.

Looking into decoded AES values from manuf. spec. seems bytes 2 to 4 (5?) is current value (in liters?) from overlay. I take the data and using hex2byte convert hex values into binary string, next split the byte into two 4 bites and create columns 2a, 2b, 3a, 3b, 4a, 4b (and 5? - const hex value 91) for better view, is visible increasing value in sections, but cannot extract correct value form this data.  From top form below table value on overlay is ~345.19 m3 and going down in the table value in the meter increase.

Value under 2a column should be last digit of the value (in liters?). Next digits (middle) of the value is under byte 3 (a+b) and first digits of the value are under byte 4 (a+b). Maybe 5 is also part of value because is constant (expect one occure 33 but maybe this is mistake). 

Please note decoded manuf. spec. in the table is not one-by-one digest telegrams, the data come from many days for few minutes, sometime pieces of decoded values is one-by-one, but not always.

Additional manuf, spec. bytes (5?) 6 to 8 seems is timestamp but now leave it.

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/legion/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/legion/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

2 | 3 | 4 | 5 | 6 | 7 | 8 | 9 | 10 | 11 | 12 | 13 | 14 | 15 | 16 |   | 2a | 2b | 3a | 3b | 4a | 4b | 5
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
06 | CF | 61 | 91 | 28 | A4 | 96 | 9D | 91 | B9 | 15 | 9B | B0 | B0 | 30 |   | 0000 | 0110 | 1100 | 1111 | 0110 | 0001 | 10010001
05 | CF | 61 | 91 | 4D | A4 | 96 | 9E | 92 | BB | 15 | 98 | B3 | B3 | 33 |   | 0000 | 0101 | 1100 | 1111 | 0110 | 0001 | 10010001
02 | CF | 61 | 91 | 57 | A4 | 96 | 99 | 95 | BD | 12 | 9F | B4 | B4 | 34 |   | 0000 | 0010 | 1100 | 1111 | 0110 | 0001 | 10010001
01 | CF | 61 | 91 | 62 | A4 | 96 | 9B | 97 | BE | 11 | 9D | B6 | B6 | 36 |   | 0000 | 0001 | 1100 | 1111 | 0110 | 0001 | 10010001
08 | CF | 61 | 91 | A4 | A4 | 96 | 91 | 9D | B7 | 14 | 97 | BC | BC | 3C |   | 0000 | 1000 | 1100 | 1111 | 0110 | 0001 | 10010001
06 | CE | 61 | 91 | 08 | A5 | 96 | EA | E6 | CE | 6E | EC | C7 | C7 | 47 |   | 0000 | 0110 | 1100 | 1110 | 0110 | 0001 | 10010001
0B | CD | 60 | 91 | 61 | D5 | 93 | 29 | 5D | 0E | A4 | 2B | 04 | 04 | 84 |   | 0000 | 1011 | 1100 | 1101 | 0110 | 0000 | 10010001
09 | CD | 60 | 91 | 1E | D6 | 93 | 2A | 5E | 0F | A6 | 28 | 07 | 07 | 87 |   | 0000 | 1001 | 1100 | 1101 | 0110 | 0000 | 10010001
06 | CC | 60 | 91 | CB | D6 | 93 | 27 | 53 | 03 | AB | 25 | 0A | 0A | 8A |   | 0000 | 0110 | 1100 | 1100 | 0110 | 0000 | 10010001
06 | CC | 60 | 91 | 87 | D7 | 93 | 20 | 54 | 04 | AC | 22 | 0D | 0D | 8D |   | 0000 | 0110 | 1100 | 1100 | 0110 | 0000 | 10010001
06 | CC | 60 | 91 | BB | D7 | 93 | 23 | 57 | 07 | AF | 21 | 0E | 0E | 8E |   | 0000 | 0110 | 1100 | 1100 | 0110 | 0000 | 10010001
06 | CC | 60 | 91 | FF | D7 | 93 | 22 | 56 | 06 | AE | 20 | 0F | 0F | 8F |   | 0000 | 0110 | 1100 | 1100 | 0110 | 0000 | 10010001
06 | CC | 60 | 91 | BB | D7 | 93 | 23 | 57 | 07 | AF | 21 | 0E | 0E | 8E |   | 0000 | 0110 | 1100 | 1100 | 0110 | 0000 | 10010001
06 | CC | 60 | 91 | FF | D7 | 93 | 22 | 56 | 06 | AE | 20 | 0F | 0F | 8F |   | 0000 | 0110 | 1100 | 1100 | 0110 | 0000 | 10010001
05 | CC | 60 | 91 | 6E | DC | 93 | 0F | 7B | 2A | 80 | 0D | 22 | 22 | A2 |   | 0000 | 0101 | 1100 | 1100 | 0110 | 0000 | 10010001
05 | CC | 60 | 91 | E5 | DC | 93 | 09 | 7D | 2C | 86 | 0B | 24 | 24 | A4 |   | 0000 | 0101 | 1100 | 1100 | 0110 | 0000 | 10010001
0B | 46 | 60 | 91 | FE | 12 | 91 | E9 | B4 | CE | 48 | F5 | C4 | C4 | 44 |   | 0000 | 1011 | 0100 | 0110 | 0110 | 0000 | 10010001
0B | 46 | 60 | 91 | 31 | 13 | 91 | E8 | B5 | CF | 49 | F4 | C5 | C5 | 45 |   | 0000 | 1011 | 0100 | 0110 | 0110 | 0000 | 10010001
0B | 46 | 60 | 91 | 24 | 1C | 91 | E4 | B9 | C3 | 45 | F8 | C9 | C9 | 49 |   | 0000 | 1011 | 0100 | 0110 | 0110 | 0000 | 10010001
0B | 46 | 60 | 91 | 5F | 1C | 91 | E7 | BA | C0 | 46 | FB | CA | CA | 4A |   | 0000 | 1011 | 0100 | 0110 | 0110 | 0000 | 10010001
0B | 46 | 60 | 91 | 92 | 1C | 91 | E6 | BB | C1 | 47 | FA | CB | CB | 4B |   | 0000 | 1011 | 0100 | 0110 | 0110 | 0000 | 10010001
0B | 46 | 60 | 91 | D5 | 1C | 91 | E1 | BC | C6 | 40 | FD | CC | CC | 4C |   | 0000 | 1011 | 0100 | 0110 | 0110 | 0000 | 10010001
0B | 46 | 60 | 91 | 0F | 1D | 91 | E0 | BD | C7 | 41 | FC | CD | CD | 4D |   | 0000 | 1011 | 0100 | 0110 | 0110 | 0000 | 10010001
0B | 46 | 60 | 91 | 42 | 1D | 91 | E3 | BE | C4 | 42 | FF | CE | CE | 4E |   | 0000 | 1011 | 0100 | 0110 | 0110 | 0000 | 10010001
0B | 46 | 60 | 91 | B8 | 1D | 91 | FD | A0 | DA | 5C | E1 | D0 | D0 | 50 |   | 0000 | 1011 | 0100 | 0110 | 0110 | 0000 | 10010001
09 | 46 | 60 | 91 | 37 | 1E | 91 | FF | A2 | DA | 5F | E3 | D2 | D2 | 52 |   | 0000 | 1001 | 0100 | 0110 | 0110 | 0000 | 10010001
0A | 49 | 60 | 91 | 92 | 1F | 91 | F5 | A8 | D1 | 48 | E9 | D8 | D8 | 58 |   | 0000 | 1010 | 0100 | 1001 | 0110 | 0000 | 10010001
[... cut ...]
0E | C3 | 6D | 91 | 95 | 2F | A7 | 59 | 37 | 77 | D8 | 77 | 74 | 74 | F4 |   | 0000 | 1110 | 1100 | 0011 | 0110 | 1101 | 10010001
0F | C3 | 6D | 91 | 9E | 2F | A7 | 58 | 36 | 75 | D8 | 76 | 75 | 75 | F5 |   | 0000 | 1111 | 1100 | 0011 | 0110 | 1101 | 10010001
0F | C3 | 6D | 91 | A0 | 2F | A7 | 5B | 35 | 76 | DB | 75 | 76 | 76 | F6 |   | 0000 | 1111 | 1100 | 0011 | 0110 | 1101 | 10010001
0F | C3 | 6D | 91 | A9 | 2F | A7 | 5A | 34 | 77 | DA | 74 | 77 | 77 | F7 |   | 0000 | 1111 | 1100 | 0011 | 0110 | 1101 | 10010001

</body>

</html>

kzajac83, 2024-04-05:
@abusca I'm stuck then focus on your examples and according WM-Bus/OMS spec and eg. this page:

https://doc.lobaro.com/display/PUB/Wireless+M-BUS

> L-Field
> The first byte of the first block is the length field. The field specifies the number of subsequent user bytes, including control and address bytes, but not CRC bytes. If ((L-9) MOD 16) is not zero, then the last block must contain ((L-9) MOD 16) data bytes +2 CRC bytes. All other blocks, except the first block, must always contain 16 data bytes + 2 CRC bytes.
> 
> CRCs
> First crc after 10 bytes
> Then every 16 bytes one CRC of length 2 (= 18)
> Plus one crc at the end (unless ending on 16 byte block)

So your telegram is with CRCs

`1C4401061426170414073D21A1B6002053209A6E49346F69A2B789559D5C384F1DECE300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000`

then cutting zeros and erase two CRC bytes after 10 byte get:

```
1C440106142617041407>>3D 21A << 1B6002053209A6E49346F69A2B789559D5C384F1DECE3`
1c440106813591041407a022004658ca87a76b5b37069b26bd49f975e2
```
After 16 bytes again 2 bytes for CRC and finsed CRC too and finally you get: 

```
1C 44 01 06 14 26 17 04 14 07 >>3D 21 << A1 B6 00 20 53 20 9A 6E 49 34 6F 69 A2 B7 89 55 >> 9D 5C << 38 4F 1D >> EC E3 <<
1C 44 01 06 14 26 17 04 14 07                      A1 B6 00 20 53 20 9A 6E 49 34 6F 69 A2 B7 89 55                        38 4F 1D
```

Therfore below 29 bytes output.

```
First scenario: APT-WMBUS-NA-1 connected to water meter. Water meter shows 00000,102m3.

2024-03-27 20:21:08 --> 1c440106142617041407a1b6002053209a6e49346f69a2b78955384f1d

2024-03-27 20:21:18 --> 1c440106142617041407a0b700f37d6a28aa9671b41200efd0a09813d0

2024-03-27 20:21:37 --> 1c440106142617041407a1b900f1158d30356dce283ade9d53137eba59

2024-03-27 20:21:57 --> 1c440106142617041407a0bb00692c2d9104f72ed93a029d9a9bf66784

```
```
Second scenario: APT-WMBUS-NA-1 connected to water meter. Water meter shows 00000,122m3.

2024-03-27 20:36:31 --> 1c440106142617041407a1130014cbba62e87b2e10eced48f13be92609

2024-03-27 20:36:51 --> 1c440106142617041407a015008b9b7838d12317cf565b027ecb5e6446

2024-03-27 20:37:01 --> 1c440106142617041407a1160012092099529918a97958995d075e9fcd

2024-03-27 20:37:31 --> 1c440106142617041407a11900aa56eabe62b9dca316330ef7bcc04006

```
```
Third scenario: APT-WMBUS-NA-1 NOT connected to water meter.

2024-03-27 20:42:17 --> 1c440106142617041407a03600eea0884a7df9a501bb726d7ae6af9ab1

2024-03-27 20:42:27 --> 1c440106142617041407a13700659b027cb08951f7e1783b97ad6666de

2024-03-27 20:42:37 --> 1c440106142617041407a03800d7daf054e0b983a1eab4b7d443012ff0

2024-03-27 20:42:47 --> 1c440106142617041407a0390035ebdae7916e09bd8095440958e7d4e4

```

Decoded 16 zeros result is below, maybe encode values:

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/legion/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/legion/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

mfc   spec. 14-29 bytes | AES decoded 14-29 bytes | 14 | 15 | 16 | 17 | 18 | 19 | 20 | 21 | 22 | 23 | 24 | 25 | 26 | 27 | 28 | 29
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
2053209a6e49346f69a2b78955384f1d | 2E071226178B3F909BB1B4B6B6B6A6A2 | 2E | 07 | 12 | 26 | 17 | 8B | 3F | 90 | 9B | B1 | B4 | B6 | B6 | B6 | A6 | A2
f37d6a28aa9671b41200efd0a09813d0 | 2E071226179D3F909AB0B5B7B7B7A7A3 | 2E | 07 | 12 | 26 | 17 | 9D | 3F | 90 | 9A | B0 | B5 | B7 | B7 | B7 | A7 | A3
f1158d30356dce283ade9d53137eba59 | 2E07122617A83F9094BEBBB9B9B9A9AD | 2E | 07 | 12 | 26 | 17 | A8 | 3F | 90 | 94 | BE | BB | B9 | B9 | B9 | A9 | AD
692c2d9104f72ed93a029d9a9bf66784 | 2E07122617C43F9096BCB9BBBBBBABAF | 2E | 07 | 12 | 26 | 17 | C4 | 3F | 90 | 96 | BC | B9 | BB | BB | BB | AB | AF
  |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
14cbba62e87b2e10eced48f13be92609 | 2E031326172D3B903E14111613131307 | 2E | 03 | 13 | 26 | 17 | 2D | 3B | 90 | 3E | 14 | 11 | 16 | 13 | 13 | 13 | 07
8b9b7838d12317cf565b027ecb5e6446 | 2E03132617393B903812171015151501 | 2E | 03 | 13 | 26 | 17 | 39 | 3B | 90 | 38 | 12 | 17 | 10 | 15 | 15 | 15 | 01
12092099529918a97958995d075e9fcd | 2E03132617433B903B11141316161602 | 2E | 03 | 13 | 26 | 17 | 43 | 3B | 90 | 3B | 11 | 14 | 13 | 16 | 16 | 16 | 02
aa56eabe62b9dca316330ef7bcc04006 | 2E03132617613B90341E1B1C1919190D | 2E | 03 | 13 | 26 | 17 | 61 | 3B | 90 | 34 | 1E | 1B | 1C | 19 | 19 | 19 | 0D
  |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
eea0884a7df9a501bb726d7ae6af9ab1 | 2E031326178024901B31343336363222 | 2E | 03 | 13 | 26 | 17 | 80 | 24 | 90 | 1B | 31 | 34 | 33 | 36 | 36 | 32 | 22
659b027cb08951f7e1783b97ad6666de | 2E031326178A24901A30353237373323 | 2E | 03 | 13 | 26 | 17 | 8A | 24 | 90 | 1A | 30 | 35 | 32 | 37 | 37 | 33 | 23
d7daf054e0b983a1eab4b7d443012ff0 | 2E031326179C2490153F3A3D38383C2C | 2E | 03 | 13 | 26 | 17 | 9C | 24 | 90 | 15 | 3F | 3A | 3D | 38 | 38 | 3C | 2C
35ebdae7916e09bd8095440958e7d4e4 | 2E03132617A52490143E3B3C39393D2D | 2E | 03 | 13 | 26 | 17 | A5 | 24 | 90 | 14 | 3E | 3B | 3C | 39 | 39 | 3D | 2D

</body>

</html>

abusca, 2024-04-06:
Thank you @kzajac83.
Very valuable info.
I took recorded some frames at known values of water volume and decrypted them using your method.
```
00000,000 m3:
1C4401061426170414073D21A03B0317A709BEF8C162E4EA8FEEAEBCD876C934F18DE4
1C440106142617041407    A03B0317A709BEF8C162E4EA8FEEAEBC    C934F1
                              17A709BEF8C162E4EA8FEEAEBC    C934F1
                              2E 08 1B 78 48 5B D2 A4  16 D6 3A 23 3B 3B 3B 0D        <- Decrypted

1C4401061426170414073D21A13C036747298F44E34C4837F65191608566173179E968
1C440106142617041407    A13C036747298F44E34C4837F6519160    173179
                              6747298F44E34C4837F6519160    173179
                              2E 08 1B 78 48 6D D2 A4  11 D1 3D 24 3C 3C 3C 0A        <- Decrypted

99999,999 m3:
1C4401061426170414073D21A06503206BD2B5258F953EF2A09772C6BB06451C33A497
1C440106142617041407    A06503206BD2B5258F953EF2A09772C6    451C33
                              206BD2B5258F953EF2A09772C6    451C33
                              2E 0B 1B 78 48 F2 D3 A4  48 88 67 7D 65 65 65 53        <- Decrypted

1C4401061426170414073D21A1660325245DF44A2FDB45C9BB213714C66F37DD378CFE
1C440106142617041407    A1660325245DF44A2FDB45C9BB213714    37DD37
                              25245DF44A2FDB45C9BB213714    37DD37
                              2E 0B 1B 78 48 04 DC A4  4B 8B 64 7E 66 66 66 50        <- Decrypted

00000,001 m3:
1C4401061426170414073D21A085038B41A566DBD4744C68B71F350EC4EB5CF6FA3758
1C440106142617041407    A085038B41A566DBD4744C68B71F350E    5CF6FA
                              8B41A566DBD4744C68B71F350E    5CF6FA
                              2E 09 1B 78 48 23 DD A4  A8 68 85 9D 85 85 85 B3        <- Decrypted

1C4401061426170414073D21A086036C6977ABDE9B1246DD1CBB40B302DB2521C1113F
1C440106142617041407    A086036C6977ABDE9B1246DD1CBB40B3    2521C1
                              6C6977ABDE9B1246DD1CBB40B3    2521C1
                              2E 09 1B 78 48 35 DD A4  AB 6B 86 9E 86 86 86 B0        <- Decrypted

99999,998:
1C4401061426170414073D21A1B1039248DE0BDCC0B144A94CCF43657F16F2F2E10129
1C440106142617041407    A1B1039248DE0BDCC0B144A94CCF4365    F2F2E1
                              9248DE0BDCC0B144A94CCF4365    F2F2E1
                              2E 0A 1B 78 48 C9 DE A4  9C 5C B2 A9 B1 B1 B1 87        <- Decrypted
    
1C4401061426170414073D21A0B3032163F520BEA4D8394227E37475756086F5AC4D7F
1C440106142617041407    A0B3032163F520BEA4D8394227E37475    86F5AC
                              2163F520BEA4D8394227E37475    86F5AC
                              2E 0A 1B 78 48 E4 DE A4  9E 5E B0 AB B3 B3 B3 85        <- Decrypted

00000,002:
1C4401061426170414073D21A0D00356F8056ECCACED458E782EE6012FCE051554DBEE
1C440106142617041407    A0D00356F8056ECCACED458E782EE601    051554
                              56F8056ECCACED458E782EE601    051554
                              2E 06 14 26 17 F8 DF A4  FD 3E D3 C7 D0 D0 D0 E6        <- Decrypted

1C4401061426170414073D21A1D203FA78BBCE74B79FE15F88E2DED2D638BE0E9EB88E
1C440106142617041407    A1D203FA78BBCE74B79FE15F88E2DED2    BE0E9E
                              FA78BBCE74B79FE15F88E2DED2    BE0E9E
                              2E 06 14 26 17 0B D8 A4  FF 3C D1 C5 D2 D2 D2 E4        <- Decrypted
```
It looks like bytes 15-18 are part of the counter value, but I'm unable to find it...

kzajac83, 2024-04-06:
Agree, the same, cannot figured out how to decode it. 

General what I see is below, bytes no:
11 -> A0 and A1 only, don't know what represent
12 -> counter sent telegrams 
13 -> 00, now I have 01 value for more than 366m3
14 -> always E2
15-18 -> actual value of liters/m3
19-21 - current timestamp
27 -> the same as 12th byte
26-29 -> somehow connected together (each 4 bites of each byte is connected with 4 bits of next to byte).    

According documantation in T1 should be:

> - Actual water meter indication and actual device date and time;
> - The last historical water meter indication with its save date and time;
> - Event flags (see details further in the Manual).

then in fields 22-26/27 should be historian value and event falg but this values is changing every telegram, cannot found constant part for historian value and event flag.

kzajac83, 2024-04-06:
@abusca - BTW you should check based on CRC does teleram is correct, not just erase CRC bytes from telegram and assume no issue in communcation.

kzajac83, 2024-04-12:
Still don't understood how actual value in 15-18 bytes is coded in telegram but below some tips, hope someone focus on it and help decode. Let's have below data (add also 14 byte, it is constant 0x2E but don't know is or not part of actual value).
My analyse base on data from about one week (value about 367m3 plus/minus 10m3 for this period)

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/legion/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/legion/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

14 | 15 | 16 | 17 | 18
-- | -- | -- | -- | --
2E | 03 | 7E | 6E | 91
2E | 00 | 7E | 6E | 91
2E | 0D | 7E | 6E | 91
2E | 06 | 81 | 69 | 91
2E | 04 | 81 | 69 | 91
2E | 04 | 81 | 69 | 91

</body>

</html>

**Byte 14**

Constant 0x2E

**Byte 15**

Value that changed most faster in code, it is last digit/-s in actual value of m3/liters.  

1st octet is always 0. 
2nd octet assign all values 0-F

In my cause value change too fast to found some rule (overlay is on main water connection for multi-apartments building)

**Byte 16**

Increase value goes according below rule:

1st octet increase normal from 0 to F
2nd octet increase according below order

`1, 0, 3, 2, 5, 4, 7, 6, 9, 8, B, A, D, C, F, E`

Somehow connected with odd and even numbers, changed alternatives odd-even. 

Change value form 0x7E -> 0x81 cause change value in next byte this mean in byte 17. Therefore seems 0x81 is value "0" 

**Byte 17**

In my cause not many changes here but:

1st octet - too small data to found some rule (still 6 value in my cause)
2nd octet - changed according below order, partial data 

`1, 0, 3, D, C, F, E, 9,  ??????`

This mean 2nd octet changed in different order in byte 16 and byte 17, another logic of changes value. 

**Byte 18**

No data to analyse, in my cause constant 0x91

@abusca are you sure value from overlay is equal to values in water meter?  
Can I ask you to deliver minimum 16 telegrams for changes +1 liter each telegram? I'd like analyse changes for byte 15, in my cause changes is too big (my overlay is mounted on main water connection for building with 12 apartments, consumption of water is too high)

abusca, 2024-04-12:
Hi @kzajac83 
Here you have the telegrams:
```
00000,001 m3:
1C4401061426170414073D21A051038C69AAA65F765DCA4AB89055AC39CBE740495897
2E 09 1B 78 48 25 0A AB  7C D4 55 49 51 51 51 63

00000,002 m3:
1C4401061426170414073D21A07B037BC27A44248E36DF7F7762847F595CCD4A163E3A
2E 06 14 26 17 B0 0B AB  56 FE 7C 6C 7B 7B 7B 49

00000,003 m3:
1C4401061426170414073D21A185036616F62E534272A67CD26C982583FE831D045972
2E 07 14 26 17 17 34 AB  A8 00 83 92 85 85 85 B7

00000,004 m3:
1C4401061426170414073D21A18E03BC370AEB0CEBD644F161B77208C141024B419AB3
2E 04 14 26 17 6C 34 AB  A3 0B 8B 99 8E 8E 8E BC

00000,005 m3:
1C4401061426170414073D21A0980324A9EAB7D05E1E11DA7A3E8206404D9F08B59521
2E 05 14 26 17 CC 34 AB  B5 1D 9C 8F 98 98 98 AA

00000,006 m3:
1C4401061426170414073D21A0A40313CCBBE56498420E26656B72A07D6C1ECA26C0F7
2E 02 14 26 17 3F 35 AB  89 21 A3 B2 A4 A4 A4 96

00000,007 m3:
1C4401061426170414073D21A0AD03C9D5577CAE286D39EC111F914D6F73E4C83E2D4B
2E 03 14 26 17 96 35 AB  80 28 AB BB AD AD AD 9F

00000,008 m3:
1C4401061426170414073D21A0B3036E9A73A1402C8A51B0B863FF20502BE5020D1509
2E 00 14 26 17 C8 35 AB  9E 36 B6 A5 B3 B3 B3 81

00000,009 m3:
1C4401061426170414073D21A0B903CBD7840254ECC0CDFC975EF34459B2657B8AFD27
2E 01 14 26 17 02 36 AB  94 3C BD AF B9 B9 B9 8B

00000,010 m3:
1C4401061426170414073D21A0BF03A9F2ACC30334CACE358CBC2036EABEBAB0DC725B
2E 0E 14 26 17 45 36 AB  92 3A B8 AA BF BF BF 8D

00000,011 m3:
1C4401061426170414073D21A1C7031E30E6E7F02A242CE9E824B45FCA41254BA9D0C5
2E 0F 14 26 17 8B 36 AB  EA 42 C1 D2 C7 C7 C7 F5

00000,012 m3:
1C4401061426170414073D21A124039B77DC61A9CCE628165F0766A4138AAC6C238659
2E 0C 14 26 17 C9 5A AB  09 AC 21 31 24 24 24 16

00000,013 m3:
1C4401061426170414073D21A1270339A66D27DE527D22DFAFD77AA1D5A47BC22F0223
2E 0D 14 26 17 7B 5B AB  0A AE 23 32 27 27 27 15

00000,014 m3:
1C4401061426170414073D21A12A0310E81562072885CAD4C70689B8A5709568D75128
2E 0A 14 26 17 34 44 AB  07 A3 2D 3E 2A 2A 2A 18

00000,015 m3:
1C4401061426170414073D21A12D03A1ACFA7D61CAEF6FB0975977A7216EA979D99FAC
2E 0B 14 26 17 E6 44 AB  00 A4 2B 39 2D 2D 2D 1F

00000,016 m3:
1C4401061426170414073D21A02F03E33681507863DEB9473943A9DF0E28A10CA0E468
2E 08 14 26 17 5C 45 AB  02 A6 2A 3B 2F 2F 2F 1D
```

For reference, here are the real values of the meter for the first and last captures.
[image]
[image]

Let me know if you need a telegram for any particular value.

kzajac83, 2024-04-12:
thx but can you confirm that actual value in the water meter represent the same in overlay telegram? You are removed overlay earlier, in this time nothing change in water meter? Or has been not calibrated correctly in buying time?  Asking because can be offset 2 or 1 liters looking into 18th byte because is dif comparing to others in this byte. 
Additional changes in 16th byte (2nd octet) from "B" to "4" never happened in my cause tell me that some offset real vs overly is possible for you. 
 
Answering on above question will deliver answer where is total "0". Anyway again thx for data.

So **byte 15th** (2nd octet) changing according this order:

`6, 7, 4, 5, 2, 3, 0, 1, E, F, C, D, A, B, 8, 9 ` 

Change value "9" -> "6" in 2nd octet of 15th byte increase +1 byte 16th. Additional order is reversed comparing to 2nd octet of 16th byte.

Quickly cannot found some math formula here, maybe just simple mapping like:
6=0
7=1 
4=2 
5=3 etc???

General can be simple mapping for each byte because byte 17th seems is different order of 2nd octet in my cause comparing to 2nd order for bytes 15th and 16th. 

Now nothing more can deliver, I need buy WiFI camera to see my water meter with mounted overlay remotely (I have very difficult access to the water meter placed in cellar) 

Perhaps someone will look at these values and find a way to decode them ;)

sgrzmiel, 2024-04-13:
I have the same meter, would it help if I provide my telegrams as well? Or can I help in some other way?

kzajac83, 2024-04-13:
Any confirmed examples will help, if you can deliver some numbers of telegram+value from meter will be fine. 

I have confirmed my value in meter with telegram (Aquanet guy last days show me values in smartphone, RTC timestamp also I have aligned  with no error flags but 1st need buy WiFi cam to see what it's on water meter). 

PS. The best to have inkasoid.apk or newer SPIDAP app, maybe someone can share it.

abusca, 2024-04-14:
@kzajac83, you are right. I can't confirm wether the values are calibrated or not. I have no experience on water meters, so I missed the point of the needing a calibration, etc. If I have the chance of getting access to data from Apator's official software, I'll let you know.

saganos, 2024-06-08:
Assuming that the key is known or the default zeros were not changed:

@gitwoy
your telegram: 
`1C440106574341061407A1E3004B277391923EE776069C110CE6F29BCB`

002   : 0106 dll-mfct (APA) - vendor
004   : 57434106 dll-id (06414357) - serial 

Your iv (vendor + serial + zeros): 
`01065743410600000000000000000000`

encrypted part: `4B277391923EE776069C110CE6F29BCB`
key: `00000000000000000000000000000000`
decrypted: 
`2f01c520007ad781ceb9e58cf6e3e367`
```
[0] [1] [2] [3] [4] ....
2f  01  c5  20  00 ....
```
formula: `(0x01 & 0b00001111) | (0xc5 << 4) | (0x20 << 12) | (0x00 << 20)`
the reading is: `134225  (134,225)`

===

@kzajac83

`1C440106813591041407A0B000266A705474DDB80D9A0EB9AE2EF29D96`

002   : 0106 dll-mfct (APA)
004   : 81359104 dll-id (04913581)

iv: `01068135910400000000000000000000`
decrypted: 
`2f004e54002ca4969d91b9159bb0b030`

formula: `(0x00 & 0b00001111) | (0x4e << 4) | (0x54 << 12) | (0x00 << 20)`
the reading is: `345312  (345,312)`

===

`1c440106142617041407a1b6002053209a6e49346f69a2b78955384f1d`
reading: `97`

`1c440106142617041407a1130014cbba62e87b2e10eced48f13be92609`
reading: `117`

===

@abusca looks like your telegrams are different

===

It could be also decrypted with IV `00000000000000000000000000000000` and same key,
and XOR the result with proper IV after decryption.

kzajac83, 2024-06-09:
Thnak you fro your answer, can you tell me for my example:

`1C440106813591041407A0B000266A705474DDB80D9A0EB9AE2EF29D96`

encrypted part is:

266A705474DDB80D9A0EB9AE2EF29D96

Using AES-128 CBC  I got decrypted:

`2E06CF619128A4969D91B9159BB0B030`

but you provide:

`2F004E54002CA4969D91B9159BB0B030
`

1st part if defferent, 2nd part is the same.   Can you please explain what is souce of difference?

Probably IV is the key but in my decryptor has no IV option....

BR, Krzysiek

saganos, 2024-06-09:
@kzajac83 try the XOR hex operation (e.g. https://xor.pw)
```
2E06CF619128A4969D91B9159BB0B030
XOR
1068135910400000000000000000000
=
2f004e54002ca4969d91b9159bb0b030
```
or find a way to provide Initialization vector (IV) to your decryptor - check this online tool (pick mode CBC http://aes.online-domain-tools.com/)

please keep in mind that the IV formula is certainly not filled with zeros, at least it has counter(s) - but it's enough to read the value. The second part is the same because zeros in XOR keep the initial value in certain octets.

saganos, 2024-06-09:
@kzajac83 python example, requires `pycryptodome`

```python3
from Crypto.Cipher import AES
from binascii import unhexlify

frame = '1C440106813591041407A0B000266A705474DDB80D9A0EB9AE2EF29D96'
encrypted_part = frame[26:]
iv_part = frame[4:16]
iv_part_filled = iv_part + '00000000000000000000'
iv = unhexlify(iv_part_filled)
key = unhexlify('00000000000000000000000000000000')

ciphertext = unhexlify(encrypted_part)
cipher = AES.new(key, AES.MODE_CBC, iv)
plain = cipher.decrypt(ciphertext)

print('iv: ', iv_part_filled, 'encrypted:', encrypted_part, 'decrypted:', plain.hex())
print('plain part:', plain[4], plain[3], plain[2], plain[1])

print('result', plain[4] << 20 | plain[3] << 12 | plain[2] << 4 | (plain[1] & 0b00001111))
```

kzajac83, 2024-06-10:
@saganos respect and many thanks, seems general working for me but little difference value (my excel vs real value) I have:

[image]

[image]

Time on cam is OK, as I compare data and date&time with Apator inkasoid app it was also correct (~2 months ago), timestamp in excel is also correct (this is time arrive telegram to the reader, is fine checked) then 2nd decoded byte and operation in excel can be a issue:

`=BITAND(HEX2BIN(AS114813;8);"00001111")`

maybe the problem is on my site?  Can you please calculate few below telegram and what value you get? 

I have in excel 470,692  m3 vs real (from cam) 470,628 m3.  General check in deep in night values because almost no changes here and in my example for many hours no change in water meter

The difference is 64 liters this mean 2^6.  

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/legion/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/legion/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

TS | RAW Telegram (without CRC) | 2 | 3 | 4 | 5 | SUM(2:5)
-- | -- | -- | -- | -- | -- | --
07.06.2024 01:37 | 1C440106813591041407A186010BC6D6B2D336269B277A56BF8A60CE09 | 68 | 3680 | 466944 | 0 | 470,692
07.06.2024 01:40 | 1C440106813591041407A1890154C5B045FABF6073936C34908F3E3BDF | 68 | 3680 | 466944 | 0 | 470,692
07.06.2024 01:41 | 1C440106813591041407A08A0186780E9711DD6487614F913DD9511C3F | 68 | 3680 | 466944 | 0 | 470,692
07.06.2024 01:48 | 1C440106813591041407A09101645BE8B4E1624CB6C417FCC51BB276A6 | 68 | 3680 | 466944 | 0 | 470,692
07.06.2024 01:56 | 1C440106813591041407A09A01AE8F02A6247DB50660A2EE9603EC41A3 | 68 | 3680 | 466944 | 0 | 470,692
07.06.2024 01:58 | 1C440106813591041407A09C016B05AC79015782F5CD004BFA178F4D75 | 68 | 3680 | 466944 | 0 | 470,692
07.06.2024 01:59 | 1C440106813591041407A09D01C7A2CDD9725C18B4CC170707A598F36D | 68 | 3680 | 466944 | 0 | 470,692
07.06.2024 02:01 | 1C440106813591041407A09F013B26FF4C166FEDAD4B33315DE7D6EF38 | 68 | 3680 | 466944 | 0 | 470,692
07.06.2024 02:03 | 1C440106813591041407A1A1016DD5930C48D8C636C09D39787D957956 | 68 | 3680 | 466944 | 0 | 470,692

</body>

</html>

kzajac83, 2024-06-10:
Change from

`=BITAND(HEX2BIN(AS114811;8);"00001111")`

to
`=BITAND(HEX2BIN(AS114811;8);"00111111")`

working fine for me, now is align cam vs excel :)   @saganos  again many thanks

saganos, 2024-06-10:
@weetmuts could you help with the driver? I've tried but `decrypt_TPL_AES_CBC_IV` was not called and `processContent`  does not have decrypted data

kzajac83, 2024-06-12:
@weetmuts can you help one comment higher ? 

>  I've tried but decrypt_TPL_AES_CBC_IV was not called and processContent does not have decrypted data

saganos, 2024-08-12:
I will continue to work on this

kzajac83, 2024-08-13:
As I see Mr Fredrik focus on own XMQ depends language driver for OMS-compliant devices only, unfortunately Apator Telemetria from Słupsk not fully or wrong understood OMS standard hence Weetmust ignore us in this cause, on the other hand you just ask for C++ developer help not understood or decode telegram  frame because is done already. 

@saganos if need help please let me, I'm not C++ developer but simple changes can implement like showed in old driver build instruction or made test.  

BTW, XMQ drivers pushed out old drivers instruction from wmbusmeters pages, if help you can found in many places eg. here:

https://github.com/wmbusmeters/wmbusmeters/blob/debug_maccrap/doc/CreateDriver.md

kzajac83, 2024-08-14:
@teklabania need translate phyton code into C++ and fit to existing structure of driver code (use proper classes, function etc). So general we are know what we want achieve but (myself) don't know how to do. Hope developers C++ help here because is very close to build the driver. 
Sad is author of project don't want help us.

saganos, 2024-08-16:
https://github.com/wmbusmeters/wmbusmeters/pull/1346

kzajac83, 2024-08-16:
@saganos for me works, tested on more than 20 examples, thx for the driver

` ./build/wmbusmeters --debug --format=json --ppjson "1C440106813591041407A08E01CDD374ED8EE6335536996B9457090933" licznik apatorna1 04913581 NOKEY
`

output is

```
{
    "media":"water",
    "meter":"apatorna1",
    "name":"licznik",
    "id":"04913581",
    "total_m3":457.956,
    "timestamp":"2024-08-16T17:58:14Z"
}

```

kzajac83, 2024-08-16:
BTW 

Do you know how read from 1234/TCP port communication from device reading messages T1, next process telegram on local machine and finally send to Mosquito broker to see value in HA?  Trying this, see values but that all:  

```
root@wyse:~# wmbusmeters 'hex:CMD(nc -lk 1234 ; /usr/bin/mosquitto_sub -u USER -P PASSWD -h 192.168.15.20 -t rtl_433/device/devices/6/Wireless-MBus/+/data | tr -d "\n" )' licznik_glowny apatorna1 04913581 NOKEY &
[4] 1171998
root@wyse:~# Started config hextty listening on none using CMD(nc -lk 1234 ; /usr/bin/mosquitto_sub -u mqtt_user -P mqtt_user -h 192.168.15.20 -t rtl_433/device/devices/6/Wireless-MBus/+/data | tr -d "\n" )
licznik_glowny  04913581        589.216 m³      2024-08-17 01:07:45
licznik_glowny  04913581        589.216 m³      2024-08-17 01:08:44
licznik_glowny  04913581        589.224 m³      2024-08-17 01:15:44

```

MQTT cannot see this topic

[image]

However when send manually below:

```
root@wyse:~# mosquitto_pub -h 192.168.15.20 -t rtl_433/device/devices/6/Wireless-MBus/ -m "some_data" -u USER -P PASSWD -d
Client (null) sending CONNECT
Client (null) received CONNACK (0)
Client (null) sending PUBLISH (d0, q0, r0, m1, 'rtl_433/device/devices/6/Wireless-MBus/', ... (9 bytes))
Client (null) sending DISCONNECT
```

than I see msg in MQTT:
[image]

What I doing wrong? 

PS. @saganos I can put any value in KEY and always working because I have zeros-string as you.  But if someone have "real" key then what?

kzajac83, 2024-09-12:
~6h ago the driver has been merge, please testing.  

For me the driver working from 18.08.2024 until now 12.09.2024, more than 3 weeks and found one issue, not sure due to corrupted telegram somehow pass CRC validation or driver issue. 

In the issue data is between 599.036 and 599.04 decrease to 598.784 (difference ~150 l) for one sample and back to correct one.  I have backflow prevention valve (PL zawór antyskażeniowy) like this so should be no reverse flow.

[image]

[image]
 
And whole period
[image]

SzczepanLeon, 2024-09-12:
One issue is still issue 😢

kzajac83, 2024-09-12:
Unfortunately no store telegrams because storing telegrams replace by below code after appear the driver:

`nohup nc -lk 1234 | wmbusmeters --selectfields=total_m3 --shell='/usr/bin/mosquitto_pub -h 192.168.15.20 -t licznik_glowny/water -m "$METER_JSON" -u USER -P PASSWD' stdin:rtlwmbus  licznik_glowny apatorna1 04913581 NOKEY &`

And will continue using above until you update  your ESPhome device for new/missing drivers. After that I can start storing telegrams because will use ESPhome API for delivering data and will have free port for this. 

But do you know how change the command to store telegrams parallel ? 

Also cannot easy validate decoded data from overlay vs real because my main water meter is deep in cellars and I need again install IP camera on water meter. Also I know that telegram vs real is correct because water supply company read this data remonty and they see if something was wrong (eg. removed overlay from water meter or other manipulations).

kitor, 2024-11-07:
Just some feedback - I'm running this driver for a week now and not a single "broken" reading. But I'm sending data via python script to influxdb (not using HA integration) if this may have any difference.

saganos, 2024-11-08:
@kzajac83 did you manage to catch another broken reading? I've also tested the driver and no single broken reading occurred. For better testing, I've updated the example code: now it contains historic value, time info and all of the flags. 
@JaioSkura take a look at flags here.

```python
# pip3 install pycryptodome

from Crypto.Cipher import AES
from datetime import datetime

telegram = bytes.fromhex('1C440106813591041407A0B000266A705474DDB80D9A0EB9AE2EF29D96')

id = telegram[4:8].hex()
reversed_id = ''.join([id[i:i+2] for i in range(0, len(id), 2)][::-1])

part1 = telegram[8:13]

encrypted = telegram[13:]
iv = telegram[2:10] + bytes([telegram[11]] * 8)
key = bytes([0] * 16)

cipher = AES.new(key, AES.MODE_CBC, iv)
plain = cipher.decrypt(encrypted)

# print('encrypted:', encrypted.hex(), 'iv:', iv.hex(), 'decrypted:', plain.hex())

multiplier = 10 ** ((plain[1] & 0b00110000) >> 4)
#reading = (plain[4] << 24 | plain[3] << 16 | plain[2] << 8 | (plain[1] & 0b00001111) << 4) >> 4
reading = plain[4] << 20 | plain[3] << 12 | plain[2] << 4 | (plain[1] & 0b00001111)
#reading_historic = (plain[13] << 24 | plain[12] << 16 | plain[11] << 8 | (plain[10] & 0b00000011) << 6) >> 6
reading_historic = plain[13] << 18 | plain[12] << 10 | plain[11] << 2 | (plain[10] & 0b00000011)

volume = multiplier * reading
volume_historic = volume - multiplier * reading_historic

timestamp_2000 = 946684800  # timestamp at 2000-01-01 00:00:00
timestamp_offset = (plain[8] << 24 | plain[7] << 16 | plain[6] << 8 | plain[5])

data = {
    "id": reversed_id,
    "time": str(datetime.fromtimestamp(timestamp_2000 + timestamp_offset)),
    "volume": volume / 1000,
    "volume_historic": volume_historic / 1000,
    "historic_hours_ago": (plain[10] & 0b11111100) >> 2 << 8 | plain[9],
}

flags = {
    "F_summer_time": part1[4] & 0b00000001,  # Flaga okresu letniego / summer time in Poland: Sun, Mar 31, 2024  –  Sun, Oct 27, 2024
    "F_hist": part1[4] & 0b00000010,  # Bit znaku wskazania historycznego / Historical indication sign bit
    "F_max_speed_ex": part1[4] & 0b00000100,  # Flaga zapamiętana – przekroczenie maksymalnej prędkości obrotowej / maximum rotational speed exceeded
    "F_device_access_err": part1[4] & 0b00001000,  # Flaga zapamiętana – błąd dostępu do urządzenia / Flag remembered – device access error
    "F_two_way_com_block": part1[4] & 0b00010000,  # Flaga blokady komunikacji dwukierunkowej / Two-way communication block flag
    "F_install": part1[4] & 0b00100000,  # Flaga ramki instalacyjnej (wybudzenie z trybu magazynowego) / install flag
    "F_wrong_temp_error": part1[4] & 0b01000000,  # Flaga zapamiętana - nieprawidłowa temperatura pracy /  incorrect operating temperature
    "F_service_event": part1[4] & 0b10000000,  # Flaga zapamiętana - zdarzenie serwisowe (błąd wskazówki) / service event (pointer error)

    "F_min": plain[14] & 0b10000000, # Flaga aktualna - przepływ minimalny / Current flag - minimum flow
    "F_max": plain[14] & 0b01000000, # Flaga aktualna - przepływ maksymalny / Current flag - maximum flow
    "F_rev": plain[14] & 0b00100000, # Flaga aktualna - przepływ wsteczny / Current flag - reverse flow
    "F_cst": plain[14] & 0b00010000, # Flaga aktualna - pomiar bez zmian / Current flag - measurement unchanged
    "F_leak": plain[14] & 0b00001000, # Flaga aktualna - wyciek / Current flag - leak
    "F_disc": plain[14] & 0b00000100, # Flaga aktualna - odłączenie urządzenia / Current flag - disconnect device
    "F_mag": plain[14] & 0b00000010, # Flaga aktualna - wykrycie pola magnetycznego / Current flag - magnetic field detected
    "Fh_bat": plain[14] & 0b00000001, # Flaga zapamiętana - niski poziom baterii / Saved flag - low battery
    "Fh_min": plain[15] & 0b10000000, # Flaga zapamiętana - przepływ minimalny / Saved flag - minimum flow
    "Fh_max": plain[15] & 0b01000000, # Flaga zapamiętana - przepływ maksymalny / Saved flag - maximum flow
    "Fh_rev": plain[15] & 0b00100000, # Flaga zapamiętana - przepływ wsteczny / Saved flag - reverse flow
    "Fh_cst": plain[15] & 0b00010000, # Flaga zapamiętana - pomiar bez zmian / Saved flag - measurement unchanged
    "Fh_leak": plain[15] & 0b00001000, # Flaga zapamiętana - wyciek / Saved flag - leak
    "Fh_disc": plain[15] & 0b00000100, # Flaga zapamiętana - odłączenie urządzenia / Flag remembered - device disconnected
    "Fh_mag": plain[15] & 0b00000010, # Flaga zapamiętana - wykrycie pola magnetycznego / Flag remembered - magnetic field detected
    "Fh_lft": plain[15] & 0b00000001, # Flaga zapamiętana - przekroczenie progu zużycia baterii / Flag remembered - battery usage threshold exceeded
}

for flag in flags:
    if flags[flag]:
        data[flag] = 1

print(data)
```

example result: 
`{'id': '04913581', 'time': '2024-03-23 16:22:20', 'volume': 345.312, 'volume_historic': 300.619, 'historic_hours_ago': 545, 'Fh_min': 1}`

kzajac83, 2024-11-08:
@saganos I am at a loss for words for your contribution to this driver.    

Below query give me more than 15k records as I keep HA data one months before purge. 

`SELECT STATE, last_reported_ts FROM "states" where metadata_id='1669'  ORDER BY 'last_reported_ts'`

Based on SQL query I have bad news, found 3 discrepancies in period 30 days:

[image]

**Issue 1**

[image]

```
root@wyse:~# wmbusmeters --ppjson --format=json 1C440106813591041407A07F018D43B54E718E8718D5D2C8F939194847 licznik_glowny apatorna1 04913581 NOKEY
{
    "media":"water",
    "meter":"apatorna1",
    "name":"licznik_glowny",
    "id":"04913581",
    "total_m3":687.361,
    "timestamp":"2024-11-08T14:06:26Z"
}

root@wyse:~# wmbusmeters --ppjson --format=json 1C440106813591041407A17D01416562108AB98A0FE92FC06AA8122176 licznik_glowny apatorna1 04913581 NOKEY
{
    "media":"water",
    "meter":"apatorna1",
    "name":"licznik_glowny",
    "id":"04913581",
    "total_m3":687.104,
    "timestamp":"2024-11-08T14:05:02Z"
}

root@wyse:~# wmbusmeters --ppjson --format=json 1C440106813591041407A174013655E9B2B8085E2F256908D7498C402C licznik_glowny apatorna1 04913581 NOKEY
{
    "media":"water",
    "meter":"apatorna1",
    "name":"licznik_glowny",
    "id":"04913581",
    "total_m3":687.357,
    "timestamp":"2024-11-08T14:07:06Z"
```

**Issue 2**

[image]

```
root@wyse:~# wmbusmeters --ppjson --format=json 1C440106813591041407A0FF019E3790584B0876226B94ADC34FFABDED licznik_glowny apatorna1 04913581 NOKEY
{
    "media":"water",
    "meter":"apatorna1",
    "name":"licznik_glowny",
    "id":"04913581",
    "total_m3":689.403,
    "timestamp":"2024-11-08T14:20:07Z"
}
root@wyse:~# wmbusmeters --ppjson --format=json 1C440106813591041407A0020134813C00486734067DB0111C30188137 licznik_glowny apatorna1 04913581 NOKEY
{
    "media":"water",
    "meter":"apatorna1",
    "name":"licznik_glowny",
    "id":"04913581",
    "total_m3":689.152,
    "timestamp":"2024-11-08T14:22:29Z"
}
root@wyse:~# wmbusmeters --ppjson --format=json 1C440106813591041407A00C019334727003B3DA268CF11117D85B6D71  licznik_glowny apatorna1 04913581 NOKEY
{
    "media":"water",
    "meter":"apatorna1",
    "name":"licznik_glowny",
    "id":"04913581",
    "total_m3":689.415,
    "timestamp":"2024-11-08T14:22:56Z"
}

```

**Issue 3**

[image]

```
root@wyse:~# wmbusmeters --ppjson --format=json 1C440106813591041407A0080174159B7E36D8E6421E3AED55C76BA54A licznik_glowny apatorna1 04913581 NOKEY
{
    "media":"water",
    "meter":"apatorna1",
    "name":"licznik_glowny",
    "id":"04913581",
    "total_m3":696.575,
    "timestamp":"2024-11-08T14:31:03Z"
}
root@wyse:~# wmbusmeters --ppjson --format=json 1C440106813591041407A10C010AEF3C1D41BE7F7A33B9EB469924397C licznik_glowny apatorna1 04913581 NOKEY
{
    "media":"water",
    "meter":"apatorna1",
    "name":"licznik_glowny",
    "id":"04913581",
    "total_m3":696.32,
    "timestamp":"2024-11-08T14:31:18Z"
}
root@wyse:~# wmbusmeters --ppjson --format=json 1C440106813591041407A0130197DAF2A1C02F9B77B14ACD51CB20C41E licznik_glowny apatorna1 04913581 NOKEY
{
    "media":"water",
    "meter":"apatorna1",
    "name":"licznik_glowny",
    "id":"04913581",
    "total_m3":696.576,
    "timestamp":"2024-11-08T14:31:35Z"
}

```

[telegrams_na1.xlsx](https://github.com/user-attachments/files/17680160/telegrams_na1.xlsx) all telegrams related to my overlay

kzajac83, 2024-11-11:
@saganos 

1. Mention already, as you see I use NOKEY parameter, can you update your driver to use any AES key, not only 0x00000000000000000000000000000000?
2. Can you update your wmbusmeters driver with adding all missing fields discovered by you in Pyhon code?
3. Regarding the issue then please note I had in the time of issues ~45 WM-Bus devices, additional catch many signals from neighbors, and the measurement device in placed in big city so a lot of radio interference here (sometime I have correct read after 16h when device sending telegram every 5 min). 
4. Due to Ad3. from begging of November'24 I change 2/3 of my devices from WM-Bus -> wired M-Bus (a lot of physical work involved in putting down cables), now have only 15 WM-Bus devices (neighbors and interference are still), I still see corrupted telegrams in logs of my device but issued telegrams decreased but not to zero. 
6. Therefore I will add new reading WM-Bus telegrams logger device placed very close to Apator NA-1 overlay in basement, soon I will do that, maybe even toady.
7. I promised validate value from telegram vs real read from water meter but not yet do that ( I need put again camera as very hard physical access without wifi in cellars there, also need mount strong headlights, without this my camera not see water mater values. Also need put temporary F/UTP PoE cabel). I will let us know when do that.

Summary, please ping me after next 30 day, I will provide similar analyse but then I'll have logs from two reading device (main and 2nd very close the overlay). Maybe also I mount camera over overlay.
Additional ask you for two top points deliver. 

BTW - how you recognize flags?

kitor, 2024-11-11:
For the record, Apator datasheet mentions optional encryption.

kzajac83, 2024-12-17:
To be 100% sure I installed camera and cheeked, values from overlay and are the same what showing water meter.
Now have ~25 WM-Bus device (in meantime add more 10 but only temporary) plus some numbers from neighbors.  But this is less as have earlier. 
Main POE WM-Bus reader installed in middle of property for last 30 days not record any negative spikes, that is good, however is placed quite far from Apator NA-1 overlay comparing to other WM-Bus devices.

Additional installed WiFi WM-Bus reader near the Apator NA-1 overlay however there is very low WiFI and time to time no connection from few hours. 

What is bad that additional WiFI WM-Bus reader placed ~5 m from overlay recorded negative spikes, 8 examples for last 30 days:
[image]

Due to interrupted time to time WIFI connection assume real issues is twice more. 

CASE 1. 
17.11.2024 12:35:27

```
root@wyse:~# wmbusmeters 1c440106813591041407a0cc001d8eBca13056ce38387ad5462a33cB4c Apator apatorna1 04913581  00000000000000000000000000000000
Apator  04913581        745.215 m³      2024-12-17 00:50:17

root@wyse:~# wmbusmeters 1C440106813591041407A0CD001AB76C46F797F4CEC4BF1354AAB735A7 Apator apatorna1 049                                                                                                           13581  00000000000000000000000000000000
Apator  04913581        744.96 m³       2024-12-17 00:51:23

root@wyse:~# wmbusmeters 1C440106813591041407A0CF00F96785734EE20FE58882911D42C42F83 Apator apatorna1 049                                                                                                           13581  00000000000000000000000000000000
Apator  04913581        745.217 m³      2024-12-17 00:51:57
```

[image]
 
  
CASE 2.
27.11.2024 15:24:23

```
root@wyse:~# wmbusmeters 1C440106813591041407A15D00E414295342F23E3D5EE3299980D7C40D Apator apatorna1 04913581  00000000000000000000000000000000
Apator  04913581        762.878 m³      2024-12-17 01:05:22

root@wyse:~# wmbusmeters 1C440106813591041407A05F00AAA6FC941CD36CAFB3F67E9EF25F06F0 Apator apatorna1 04913581  00000000000000000000000000000000
Apator  04913581        762.624 m³      2024-12-17 01:05:32

root@wyse:~# wmbusmeters 1C440106813591041407A061000718DB890B4B0305E95791CDCD2BB0A5 Apator apatorna1 04913581  00000000000000000000000000000000
Apator  04913581        762.881 m³      2024-12-17 01:06:01
```

[image]

CASE 3.
03.12.2024 13:39:05

```
root@wyse:~# wmbusmeters 1C440106813591041407A0910081C3EBA4089D56A817062EC8B6DF6F72 Apator apatorna1 04913581  00000000000000000000000000000000
Apator  04913581        773.118 m³      2024-12-17 01:09:40

root@wyse:~# wmbusmeters 1C440106813591041407A092006F49D157D4FED8E03F81ED8714F89CB2 Apator apatorna1 04913581  00000000000000000000000000000000
Apator  04913581        772.864 m³      2024-12-17 01:10:10

root@wyse:~# wmbusmeters 1C440106813591041407A1930055A6A001B58EA43D77FA1BFFEB8C40A4 Apator apatorna1 04913581  00000000000000000000000000000000
Apator  04913581        773.121 m³      2024-12-17 01:12:37

```

[image]

CASE 4.
04.12.2024 10:22:39

```
root@wyse:~# wmbusmeters 1C440106813591041407A0CF00EF4CD02F35A55A081C79ADBBE48F73B0 Apator apatorna1 04913581  00000000000000000000000000000000
Apator  04913581        774.399 m³      2024-12-17 01:15:25

root@wyse:~# wmbusmeters 1C440106813591041407A1D00089C108BE42B9DF149F034B87A657E4B7 Apator apatorna1 04913581  00000000000000000000000000000000
Apator  04913581        774.144 m³      2024-12-17 01:15:45

root@wyse:~# wmbusmeters 1C440106813591041407A0D100CF5952DCD05960D8AB5B739117F429A4 Apator apatorna1 04913581  00000000000000000000000000000000
Apator  04913581        774.4 m³        2024-12-17 01:16:08

```

[image]

CASE 5. 
05.12.2024 10:21:47

```
root@wyse:~# wmbusmeters 1C440106813591041407A1BD00DCF26599B10D476B3D7CC2397A48046D Apator apatorna1 04913581  00000000000000000000000000000000
Apator  04913581        776.191 m³      2024-12-17 01:26:33

root@wyse:~# wmbusmeters 1C440106813591041407A0BE0099BFD028F9811C754031BCF230C66361 Apator apatorna1 04913581  00000000000000000000000000000000
Apator  04913581        775.936 m³      2024-12-17 01:26:36

root@wyse:~# wmbusmeters 1C440106813591041407A0BF00D49C9F4553369789FD32459E40356822 Apator apatorna1 04913581  00000000000000000000000000000000
Apator  04913581        776.192 m³      2024-12-17 01:26:49
```

[image]

 

**Summary**. 

Historically issue has been recognized by two WM-Bus readers devices so three possibilities to explain:
1. Broken Apator NA-1 overlay or firmware inside.  
2. Electromagnetic waves interference and happy CRC pass but too many causes so almost not possible. On the other hand found other NA-1 overlay in the air from neighbors so maybe...
3. Some special flags included in telegrams are not handle and driver not recognize it.

Please note always spikes are negative if nobody found similar cause hence assume my overlay is broken or have some old FW revision with the issue.  

Anybody have similar causes? If no issue can you deliver from sticker what revision FW you have?

Extra info.
WM-Bus reader placed near Apator overlay read more than ~38k telegram.
Main reader behind two floors record in the same period read more than 12k telegrams.
This is reason why main WM-Bus reader did not capture preoblematic telegrams.

PS. Mentioned NA-1  from neighbors in my range, some example:
root@wyse:~# wmbusmeters 1C440106867702031407A1600051E2B0474C47DE9E930E10220664CFF3 Apator apatorna1 03027786  00000000000000000000000000000000
Apator  03027786        1758.472 m³     2024-12-17 01:04:24

root@wyse:~# wmbusmeters 1C440106867702031407A08C00D41579B6530AA7C42C9487AF841D5731 Apator apatorna1 03027786  00000000000000000000000000000000
Apator  03027786        1758.472 m³     2024-12-17 01:27:50

teklabania, 2024-12-17:
Negative spikes occurs a few times a month in my case as well.

closed 2025-05-18 by github-actions[bot]