wmbusmeters/wmbusmeters #1216: Inverted order HEX data in debug mode

kzajac83, 2024-03-17:
### Type of request

Bug

### OS version

Current Debian

### wmbusmeters version

wmbusmeters: 1.16.1-12-g1f2cd10

### Your message goes here

Hello.

Try build new driver and in debug mode when you see telegrams there are part of related to structure or data like below

```
(atna01) 000   : 68 start
(atna01) 001   : 56 length (86 bytes)
(atna01) 002   : 56 length again (86 bytes)
(atna01) 003   : 68 start
(atna01) 091   : 16 end
(atna01) 090   : ED crc
(atna01) 004   : 08 dll-c (RSP_UD2)
(atna01) 005   : 01 dll-a primary (1)
(atna01) 006   : 72 tpl-ci (long header)
(atna01) 007   : 28170500 tpl-id (00051728)
(atna01) 011   : 0106 tpl-mfct (APA)
(atna01) 013   : 03 tpl-version
(atna01) 014   : 07 tpl-type (Water meter)
(atna01) 015   : 0f tpl-acc-field
(atna01) 016   : fa tpl-sts-field (ERROR PERMANENT_ERROR TEMPORARY_ERROR)(ERROR PERMANENT_ERROR TEMPORARY_ERROR UNKNOWN_E0)
(atna01) 017   : 0000 tpl-cfg 0000 ( )
(atna01) 019   : 0C dif (8 digit BCD Instantaneous value)
(atna01) 020   : 78 vif (Fabrication no)
(atna01) 021 C!: 28170500 ("fabrication_no":"00051728")
(atna01) 025   : 04 dif (32 Bit Integer/Binary Instantaneous value)
(atna01) 026   : 6D vif (Date and time type)
(atna01) 027 C!: 0C2F1133 ("meter_datetime":"2024-03-17 15:12")
(atna01) 031   : 04 dif (32 Bit Integer/Binary Instantaneous value)
(atna01) 032   : 13 vif (Volume l)
(atna01) 033 C!: 4B020000 ("total_m3":0.587)
(atna01) 037   : 02 dif (16 Bit Integer/Binary Instantaneous value)
(atna01) 038   : 3B vif (Volume flow l/h)
(atna01) 039 C!: 0000 ("volume_flow_m3h":0)
(atna01) 041   : 44 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
(atna01) 042   : 13 vif (Volume l)
(atna01) 043 C!: F9010000 ("target_m3":0.505)
(atna01) 047   : 42 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
(atna01) 048   : 6C vif (Date type G)
(atna01) 049 C!: 0133 ("target_date":"2024-03-01")
(atna01) 051   : 02 dif (16 Bit Integer/Binary Instantaneous value)
(atna01) 052   : 27 vif (Operating time days)
(atna01) 053 C!: E70B ("operating_time_h":73128)
(atna01) 055   : 03 dif (24 Bit Integer/Binary Instantaneous value)
(atna01) 056   : FD vif (Second extension FD of VIF-codes)
(atna01) 057   : 17 vife (Error flags (binary))
(atna01) 058 C!: 1E0300 ("error_flag":"ERROR_FLAG_31E")
(atna01) 061   : 04 dif (32 Bit Integer/Binary Instantaneous value)
(atna01) 062   : FF vif (Manufacturer specific)
(atna01) 063   : 0A vife (?)
(atna01) 064 C!: 04010100 ("manuf_spec_0A":"MAN_SPEC_0A_10104")
(atna01) 068   : 02 dif (16 Bit Integer/Binary Instantaneous value)
(atna01) 069   : FF vif (Manufacturer specific)
(atna01) 070   : 0B vife (?)
(atna01) 071 C!: 975B ("manuf_spec_0B":"MAN_SPEC_0B_5B97")
(atna01) 073   : 03 dif (24 Bit Integer/Binary Instantaneous value)
(atna01) 074   : FF vif (Manufacturer specific)
(atna01) 075   : 0C vife (?)
(atna01) 076 C!: 0800C3 ("manuf_spec_0C":"MAN_SPEC_0C_C30008")
(atna01) 079 C?: 0F manufacturer specific data 00012E09FF0002020100
```

And please focus on recognized elements marked by C! this mean:

```
(atna01) 021 C!: 28170500 ("fabrication_no":"00051728")
(atna01) 027 C!: 0C2F1133 ("meter_datetime":"2024-03-17 15:12")
(atna01) 033 C!: 4B020000 ("total_m3":0.587)
(atna01) 039 C!: 0000 ("volume_flow_m3h":0)
(atna01) 043 C!: F9010000 ("target_m3":0.505)
(atna01) 049 C!: 0133 ("target_date":"2024-03-01")
(atna01) 053 C!: E70B ("operating_time_h":73128)
(atna01) 058 C!: 1E0300 ("error_flag":"ERROR_FLAG_31E")
(atna01) 064 C!: 04010100 ("manuf_spec_0A":"MAN_SPEC_0A_10104")
(atna01) 071 C!: 975B ("manuf_spec_0B":"MAN_SPEC_0B_5B97")
(atna01) 076 C!: 0800C3 ("manuf_spec_0C":"MAN_SPEC_0C_C30008")
```
After C! and colon there is HEX (or DEC??) data but order of this data is inverted.

Examples

`(atna01) 021 C!: 28170500 ("fabrication_no":"00051728")
`
we see "28170500" but should be "00 05 17 28"

In this cause not sure should be DEC or HEX, now is DEC, always is HEX, maybe next small thing to fix?? 

Next cause
`
(atna01) 033 C!: 4B020000 ("total_m3":0.587)`

0x4B020000 = DEC 1258422272 what is wrong,  inverted order is "00 00 02 4B"=  0x24B = DEC 587

Another example

`(atna01) 053 C!: E70B ("operating_time_h":73128)`

0xE70B =  DEC 59147.   Should be "0B E7" = 0x BE7 = DEC 3047.  Because is in hours then 3047*24 = 73128

Not only sure - this data should be always HEX or DEC like fabrication_no?

Best Regards, Krzysiek

weetmuts, 2024-03-18:
The HEX is the actual bytes sent/received and what you see being transmitted over the serial lines.
And (w)mbus always sends with the least significant byte first. 

The value presented to the right is the HEX bytes properly decoded. I.e. the dataformat specifier DIF is applied to the raw data bytes and the value is then scaled using the VIF.

So, this is as it should be. :-)

kzajac83, 2024-03-18:
Thx Fredrik, I will close the issue but before do this will continiu regarding the driver - there are some documentation how build xmq ? 

BTW - there are bug, if you use [TAB] in xmq file then get below error, if use one or many [SPACE] then pass, no error.

```
root@wyse:/etc/wmbusmeters.drivers.d# wmbusmeters --debug --pollinterval=60s --ppjson --format=json /dev/ttyUSB0:mbus:2400 counter atna01.xmq 00051728 NOKEY
(drivers) added builtin driver elster
(drivers) added builtin driver iperl
(driver) error loading wmbusmeters driver file atna01.xmq
atna01.xmq:7:1: error: unexpected tab character (remember tabs must be quoted)
        mvt = APA,03,07

``` 

Additional seems you need pass full path to the xmq driver or be in /etc/wmbusmeters/wmbusmeters.drivers.d/ directory if use just driver_name.xmq

kzajac83, 2024-03-18:
PS. And seems xmq is not working with real devices because below working and give some answer and correct values:

`wmbusmeters --debug "6856566808017228170500010603073CFA00000C7828170500046D012E123304134D020000023B00004413F9010000426C01330227E80B03FD171E030004FF0A0404010002FF0B975B03FF0C0800C30F0001121BFF00020201000B16" licznik atna01.xmq 00051728 NOKEY
`

but this now work:

`wmbusmeters --debug --pollinterval=60s /dev/ttyUSB0:mbus:2400 licznik atna01.xmq 00051728 NOKEY
`

because give answer (after now answer ~30s put ctrl+c)

```
(drivers) added builtin driver elster
(drivers) added builtin driver iperl
(driver) loading driver atna01 from file atna01.xmq
(driver) register detection atna01.xmq APA  7 03
(driver) register detection atna01.xmq APA  3 07
(drivers) newly loaded driver atna01.xmq overrides builtin driver
(wmbusmeters) version: 1.16.1-12-g1f2cd10
(config) using device: /dev/ttyUSB0:mbus:2400
(config) number of meters: 1
(config) template licznik 00051728 atna01
(lookup) with file/hex "/dev/ttyUSB0" mbus
(lookup) driver: mbus
(main) opening /dev/ttyUSB0:mbus:2400
Started config mbus on /dev/ttyUSB0
(mbus) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B2400) input() output() control(CLOCAL|CREAD|CSIZE|PARENB) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B2400) input() output() control(CLOCAL|CREAD|CSIZE|PARENB) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (mbus)
Sending NKE to mbus
(serial /dev/ttyUSB0) sent "1040004016"
(serial) EVENT thread interrupted
(serial) received binary "E5"
(mbus) checkMBUSFrame
 "E5"
(mbus) received E5 single byte frame.
(mbus) checkMBUSFrame
 ""
(mbus) less than 6 bytes, partial frame
(main) regular reset of mbus  on /dev/ttyUSB0 will happen every 82800 seconds
(wmbus) no alarm (expected activity) for mbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
^C(serial) stopping manager
(serial) MAIN thread interrupted
(serial) closing 1 devices
(serial) TIMER thread interrupted
(serial) EVENT thread interrupted
(serial) event loop stopped!
(wmbus) disconnected /dev/ttyUSB0 mbus
(serialtty) closed /dev/ttyUSB0 (mbus)
(wmbus) deleted mbus
(serial) closing 1 devices
```

When use build-in driver then the same command give me answer from water counter

```
wmbusmeters --debug  --pollinterval=60s --ppjson --format=json /dev/ttyUSB0:mbus:2400 licznik atna01 00051728 NOKEY
(drivers) added builtin driver elster
(drivers) added builtin driver iperl
(wmbusmeters) version: 1.16.1-12-g1f2cd10
(config) using device: /dev/ttyUSB0:mbus:2400
(config) number of meters: 1
(config) template licznik 00051728 atna01
(meter) created licznik atna01 00051728 encrypted
(lookup) with file/hex "/dev/ttyUSB0" mbus
[... 81 lines cut ...]
 "685656680801722817050001"
(mbus) not enough bytes, partial frame 12 92
(serial) received binary "06"
(mbus) checkMBUSFrame
 "68565668080172281705000106"
(mbus) not enough bytes, partial frame 13 92
(serial) received binary "03"
(mbus) checkMBUSFrame
 "6856566808017228170500010603"
(mbus) not enough bytes, partial frame 14 92
(serial) received binary "07"
(mbus) checkMBUSFrame
 "685656680801722817050001060307"
(mbus) not enough bytes, partial frame 15 92
(serial) received binary "3E"
(mbus) checkMBUSFrame
 "6856566808017228170500010603073E"
(mbus) not enough bytes, partial frame 16 92
(serial) received binary "FA"
(mbus) checkMBUSFrame
......

```

BR, Krzysiek

weetmuts, 2024-03-19:
Can you post your new xmq driver please? Then I can test what you have written.
The xmq format is documented here: https://libxmq.org

weetmuts, 2024-03-19:
> tna01.xmq:7:1: error: unexpected tab character (remember tabs must be quoted)

I have improved the error message in future releases of xmq:
case XMQ_ERROR_UNEXPECTED_TAB: return "unexpected tab character (remember tabs must be quoted)";
case XMQ_ERROR_UNEXPECTED_TAB: return "invalid tab character found, remember that tab is not allowed as a field separator, to store tab as content it must be quoted";

kzajac83, 2024-03-19:
[atna0x.zip](https://github.com/wmbusmeters/wmbusmeters/files/14649935/atna0x.zip)

Thank you, now understood what is XMQ. 

Because earlier working on cc driver in C++ (is need only somehow encode manufacturer specification fields) then XMQ driver not yet ready, but give some data for testing. 

However there are a mentioned problem, if give as argument telegram then working fine but when use instead telegram -> device (in my cause /dev/ttyUSB0:2400) then no answer from device, probably LinkMode is hard coded into wmbus (T1, T2 etc) for XMQ driver or something like that.

kzajac83, 2024-03-19:
PS. Regarding my question about XMQ documentation I have in mind XMQ driver structure this mean what fields/structures are mandatory, what optional, what available/possible to use in the driver, how decode not standard (w)m-bus fields, general asking about documentation regarding drivers under wmbusmeters.

weetmuts, 2024-03-22:
The dynamic loadable driver code is being developed. So, there is no documentation yet.....

kzajac83, 2024-03-23:
Underwood, is clear. 

Only I just want to point out that XMQ / dynamic loadable driver at this moment not work with real device like my example /dev/ttyUSB0:2400 (no any read frame / telegram from slave devices, only received E5 single byte frame). For standard driver working.

closed 2024-06-22 by github-actions[bot]