wmbusmeters/wmbusmeters #1219: Driver for Eltako DSZ15DM three-phase electricity wired mbus meters

huhndomi, 2024-03-24:
### Type of request

New device support

### Meter description

I installed the Eltako DSZ15DM as a wired M-Bus three-phase electricity meter. This is not yet stored in the drivers. An initial analysis of Telegram showed the highest hit rate of 27/47 with the iem3000 driver for the Schneider Electric electricity meter.

A detailed description of the manufacturer for the Telegram can be downloaded from the following link: [https://www.eltako.com/fileadmin/downloads/de/_bedienung/M-Bus_Stromzaehler_Technische_Daten_Einlernhilfe.pdf](url)

Auto driver  : not found!
Best driver  : iem3000 26/47
Using driver : iem3000 26/47
000   : 68 start
001   : 92 length (146 bytes)
002   : 92 length again (146 bytes)
003   : 68 start
004   : 08 dll-c (RSP_UD2)
005   : 00 dll-a primary (0)
006   : 72 tpl-ci (long header)
007   : 36124523 tpl-id (23451236)
011   : 9415 tpl-mfct (ELT)
013   : 01 tpl-version
014   : 02 tpl-type (Electricity meter)
015   : cd tpl-acc-field
016   : 00 tpl-sts-field (OK)(OK)
017   : 0000 tpl-cfg 0000 ( )
019   : 8C dif (8 digit BCD Instantaneous value)
020   : 10 dife (subunit=0 tariff=1 storagenr=0)
021   : 04 vif (Energy 10¹ Wh)
022 C!: 68890200 ("total_energy_consumption_tariff_1_kwh":289.68)
026   : 8C dif (8 digit BCD Instantaneous value)
027   : 11 dife (subunit=0 tariff=1 storagenr=2)
028   : 04 vif (Energy 10¹ Wh)
029 C?: 68890200
033   : 8C dif (8 digit BCD Instantaneous value)
034   : 20 dife (subunit=0 tariff=2 storagenr=0)
035   : 04 vif (Energy 10¹ Wh)
036 C!: 00000000 ("total_energy_consumption_tariff_2_kwh":0)
040   : 8C dif (8 digit BCD Instantaneous value)
041   : 21 dife (subunit=0 tariff=2 storagenr=2)
042   : 04 vif (Energy 10¹ Wh)
043 C?: 00000000
047   : 02 dif (16 Bit Integer/Binary Instantaneous value)
048   : FD vif (Second extension FD of VIF-codes)
049   : C9 vife (10^0 Volts)
050   : FF combinable vif (MfctSpecific)
051   : 01 combinable extension vife
052 C!: EE00 ("voltage_l1_n_v":238)
054   : 02 dif (16 Bit Integer/Binary Instantaneous value)
055   : FD vif (Second extension FD of VIF-codes)
056   : DB vife (10^-1 Ampere)
057   : FF combinable vif (MfctSpecific)
058   : 01 combinable extension vife
059 C!: 1000 ("current_l1_a":1.6)
061   : 02 dif (16 Bit Integer/Binary Instantaneous value)
062   : AC vif (Power 10¹ W)
063   : FF combinable vif (MfctSpecific)
064   : 01 combinable extension vife
065 C!: 2000 ("active_consumption_l1_kw":0.32)
067   : 82 dif (16 Bit Integer/Binary Instantaneous value)
068   : 40 dife (subunit=1 tariff=0 storagenr=0)
069   : AC vif (Power 10¹ W)
070   : FF combinable vif (MfctSpecific)
071   : 01 combinable extension vife
072 C?: 0A00
074   : 02 dif (16 Bit Integer/Binary Instantaneous value)
075   : FD vif (Second extension FD of VIF-codes)
076   : C9 vife (10^0 Volts)
077   : FF combinable vif (MfctSpecific)
078   : 02 combinable extension vife
079 C!: EF00 ("voltage_l2_n_v":239)
081   : 02 dif (16 Bit Integer/Binary Instantaneous value)
082   : FD vif (Second extension FD of VIF-codes)
083   : DB vife (10^-1 Ampere)
084   : FF combinable vif (MfctSpecific)
085   : 02 combinable extension vife
086 C!: 1100 ("current_l2_a":1.7)
088   : 02 dif (16 Bit Integer/Binary Instantaneous value)
089   : AC vif (Power 10¹ W)
090   : FF combinable vif (MfctSpecific)
091   : 02 combinable extension vife
092 C!: 2400 ("active_consumption_l2_kw":0.36)
094   : 82 dif (16 Bit Integer/Binary Instantaneous value)
095   : 40 dife (subunit=1 tariff=0 storagenr=0)
096   : AC vif (Power 10¹ W)
097   : FF combinable vif (MfctSpecific)
098   : 02 combinable extension vife
099 C?: 0700
101   : 02 dif (16 Bit Integer/Binary Instantaneous value)
102   : FD vif (Second extension FD of VIF-codes)
103   : C9 vife (10^0 Volts)
104   : FF combinable vif (MfctSpecific)
105   : 03 combinable extension vife
106 C!: EE00 ("voltage_l3_n_v":238)
108   : 02 dif (16 Bit Integer/Binary Instantaneous value)
109   : FD vif (Second extension FD of VIF-codes)
110   : DB vife (10^-1 Ampere)
111   : FF combinable vif (MfctSpecific)
112   : 03 combinable extension vife
113 C!: 0E00 ("current_l3_a":1.4)
115   : 02 dif (16 Bit Integer/Binary Instantaneous value)
116   : AC vif (Power 10¹ W)
117   : FF combinable vif (MfctSpecific)
118   : 03 combinable extension vife
119 C!: 1C00 ("active_consumption_l3_kw":0.28)
121   : 82 dif (16 Bit Integer/Binary Instantaneous value)
122   : 40 dife (subunit=1 tariff=0 storagenr=0)
123   : AC vif (Power 10¹ W)
124   : FF combinable vif (MfctSpecific)
125   : 03 combinable extension vife
126 C?: 0600
128   : 02 dif (16 Bit Integer/Binary Instantaneous value)
129   : FF vif (Manufacturer specific)
130   : 68 vife (?)
131 C?: 0100
133   : 02 dif (16 Bit Integer/Binary Instantaneous value)
134   : AC vif (Power 10¹ W)
135   : FF combinable vif (MfctSpecific)
136   : 00 combinable extension vife
137 C?: 6200
139   : 82 dif (16 Bit Integer/Binary Instantaneous value)
140   : 40 dife (subunit=1 tariff=0 storagenr=0)
141   : AC vif (Power 10¹ W)
142   : FF combinable vif (MfctSpecific)
143   : 00 combinable extension vife
144 C?: 1900
146   : 01 dif (8 Bit Integer/Binary Instantaneous value)
147   : FF vif (Manufacturer specific)
148   : 13 vife (?)
149 C?: 00
150   : F4 crc
151   : 16 end

{
    "media":"electricity",
    "meter":"iem3000",
    "name":"",
    "id":"23451236",
    "active_consumption_l1_kw":0.32,
    "active_consumption_l2_kw":0.36,
    "active_consumption_l3_kw":0.28,
    "current_l1_a":1.6,
    "current_l2_a":1.7,
    "current_l3_a":1.4,
    "total_energy_consumption_tariff_1_kwh":289.68,
    "total_energy_consumption_tariff_2_kwh":0,
    "voltage_l1_n_v":238,
    "voltage_l2_n_v":239,
    "voltage_l3_n_v":238,
    "status":"OK",
    "timestamp":"2024-03-24T22:16:34Z"
}

Using: wmbusmeters: 1.15.0-1-g097f91f
097f91fac0585c6e47beec7dcb086cce3fcb0e5e

### Logged telegrams from meter in decrypted format

```shell
689292680800723612452394150102cd0000008c1004688902008c1104688902008c2004000000008c21040000000002fdc9ff01ee0002fddBff01100002acff0120008240acff010a0002fdc9ff02ef0002fddBff02110002acff0224008240acff02070002fdc9ff03ee0002fddBff030e0002acff031c008240acff03060002ff68010002acff0062008240acff00190001ff1300f416
```

probutus, 2024-05-25:
I have this meter and created a driver for it
[driver_eltako.tar.gz](https://github.com/wmbusmeters/wmbusmeters/files/15445449/driver_eltako.tar.gz)
I will create a pull request later on. unfortunately the test suite does not compile due to some unrelated error

probutus, 2024-05-26:
Created Pull Request here:
https://github.com/wmbusmeters/wmbusmeters/pull/1291

closed 2024-08-25 by github-actions[bot]

Saicopate, 2025-04-22:
May I ask, where did you get the driver for Schneider IEM3000 series that you used the first time?

probutus, 2025-04-22:
> May I ask, where did you get the driver for Schneider IEM3000 series that you used the first time?

Hi @Saicopate 
I am not aware of a Schneider electricity meter. But I can have a look if you like. I would need a raw telegram from the electricty meter that I can have a look.

BR,

Alex

Saicopate, 2025-06-07:
Ok I managed.

probutus, 2025-06-11:
HI @Saicopate ,
sorry for late reply, I was on vacation. From what I see the telegram looks ok. Do you know what was the issue?