wmbusmeters/wmbusmeters #1220: Low bandwidth remote source

PovilasID, 2024-03-26:
### Type of request

Question

### OS version

OpenWRT

### wmbusmeters version

1.15

### Your message goes here

Hey,

I have rtl_433 running capturing both weather station and wMbus water meter. To parse the wMbus meter I need to forward it wmbusmeter docker container running remotely (on same machine I forward MQTT messages for weather stations)
I could probably do it using `-F rtl_tcp` on rtl_433 but I am running rtl_433 it on low power OpenWRT router and do not have the processing power or bandwidth to maintain such connection.
In the docs of rtl_433 I saw  `-F syslog:127.0.0.1:1514` but I did not see what protocol does it use to capture aka what do I provide in device section for wmbusmeter `rtl433:CMD(nc 192.168.x.x:1514)` 
What is low bandwidth option to transport captured signals using minimal resources?

lmmodesto, 2024-04-23:
Hi,

At the moment I don't have an Openwrt machine to test and do some benchmarks

Maybe I'll try something like this on a TP-Link WDRT3600 in the future

```bash
# Machine 1 - TP-Link WDRT3600
# Filter protocol 104 = Wireless M-Bus, SAP = SAPPEL manufacturer
$ rtl_433  -f 434.47M -s 1600k -R 104 -F csv | awk 'BEGIN {FS=","} $22=="SAP" {print $25}' |  nc -u 192.168.1.193 4444
```
And this in another machine (eg.: PC running a docker image)

```bash
# Machine 2  
# Fx hex telegram length and remove CRC16 
$ nc -q -1 -lku 4444 | gawk '{t= substr($0,3,length($0)-6); printf("%02x%s\n", length(t)/2 , t);fflush(); }'  | ./wmbusmeters  --format=json stdin:hex MyCounter izar '*' NOKEY
```

Using something like this, I guess that you'll have some significant CPU usage (telegrams decoding) on Openwrt machine, but less bandwidth usage.

I guess you will have always to chosse between CPU usage and bandwidth usage.

Maybe you could also try running the decoding and filtering part for short periods of time on the Openwrt machine and then forwarding the telegrams to the second machine...

 If you could describe your Openwrt machine and share some benchmarking that would be great!

Marcelo Modesto

PovilasID, 2024-04-24:
@lmmodesto Thanks for an option I am running Belkin AX 3200 so it has plenty of CPU to process the telegrams with rtl_433 with my usage and rtl_433 running to send weather station data to mqtt server my daily CPU load average is a little shy of 20%

[image]

1. OpenWRT  comes with stripped down `nc` package (listen only) you have to install   `opkg install netcat` have a server mode
2. I edited your awk example to suite my settings better : `'BEGIN {FS=","}  $15=="0000000" {print $50}'` I have all my neighbors sensors are transmitting too, so I am filtering by the ID of my sensor. I would prefer to still have data presented in the log but lets solve one problem at the time.

Problem:
I am running wmbusmeter in docker container and now I am a little confused how to structure my `wmbusmeters.conf`. I tried this... did not work:
```
loglevel=debug
device=stdin:hex:CMD(nc -q -1 -lku 4444 | gawk '{t= substr($0,3,length($0)-6); printf("%02x%s\n", length(t)/2 , t);fflush(); }')
donotprobe=/dev/ttyAMA0
logtelegrams=false
format=json
meterfiles=/wmbusmeters_data/logs/meter_readings
meterfilesaction=overwrite
logfile=/wmbusmeters_data/logs/wmbusmeters.log
```
Also I saw that stdin:rtlwmbus should 'Works for rtl433 as well', so maybe `awk` is not necessary? Maybe I pass kv output directly?

I attached to docker container to test if I run ` nc -q -1 -lku 4444 | gawk '{t= substr($0,3,length($0)-6); printf("%02x%s\n", length(t)/2 , t);fflush(); }' `
I get output :
```
00
00
00
00
00
00
00
00
00
00
00
00
00
00
00
00
00
2444b4096153240317068c008b7a460000000c1309450100046d0d2d18340f030000000000
```
Apart from bunch of extra 00 it looks like I am getting the data transmitted just not sure how to consume it inside of the container. 
A possible solution may be overriding docker containers entry command or adding an extra command however then  I think I would lose logging in container.

lmmodesto, 2024-04-24:
> 1. OpenWRT  comes with stripped down `nc` package (listen only) you have to install   `opkg install netcat` have a server mode

```nc``` was just an example of how to send the data to the second machine. 

Maybe you could chosse another way - eg.: mosquitto_pub on the machine 1 and mosquitto_sub on the second one.

The idea was mainly focused on lower bandwidth - ```csv``` format and ```awk``` filter

> 2. I edited your awk example to suite my settings better : `'BEGIN {FS=","}  $15=="0000000" {print $50}'` I have all my neighbors sensors are transmitting too, so I am filtering by the ID of my sensor. I would prefer to still have data presented in the log but lets solve one problem at the time.

Nice! 

It looks like you have a different decoder (you are filtering based on the field 50). 

I chose ```rtl_433 R -104``` and fields 22 and 25 because I want Wmbus data 

> Problem: I am running wmbusmeter in docker container and now I am a little confused how to structure my `wmbusmeters.conf`. I tried this... did not work:
>
> ```
> loglevel=debug
> device=stdin:hex:CMD(nc [...]

I don't know ```wmbusmeters``` in deep. I'm starting to test it. The same for the ```rtl_433```. 

There's a note about using parentheses in ```rtlwmbus:CMD(<command line>)```. 

I don't know if the same applies to ```stdin:hex```... 

In my tests, I've noted that the options bellow produce the same output 

```bash 

$ rtl_433 -R 0 -f 434.47M -s 1600k -w - | rtl_wmbus -p s -a 

$ rtl_sdr -d 0 -f 434.47M -s 1.6e6 - 2>/dev/null | rtl_wmbus -p s -a
```

I'm not sure, but I think it won't work if you try to send ```rtl_433``` ```kv``` format directly to ```wmbusmeters```.

Trying to find a way to reduce bandwidth,  I chose the ```rtl_433``` ```csv``` format to easily apply a basic filter.

Trying to reduce the number of processes, I choose ```stdin:hex``` ```wmbusmeters``` device and removed ```rtl_wmbus```.

> I attached to docker container to test if I run `nc -q -1 -lku 4444 | gawk '{t= substr($0,3,length($0)-6); printf("%02x%s\n", length(t)/2 , t);fflush(); }'` I get output :
>
> ```
> 00
> 00
> 00
> 00
> 00
> 00
> 00 [...]

It would be better to collect some data on machine 1 and then try to debug the ```gawk``` part.

It works for my case:  wmbus protocol (```rtl_433 -R 104```) and Izar PRIOS protocol on ```wmbusmeters ``` (my meter is a IZAR 3.5) . 

Basically, I'm correcting the size of hex data expected for Izar decoder. 

I don't know if this is necessary for your protocol and decoder...

PovilasID, 2024-04-25:
1. mosquitto_sub  on remote machine is grate suggestion because it will eliminate some duplication of the data transmission.
2. I have some meta data options enabled for `rtl_433` that is probably why datagram positions do not match but that is an esy fix for anybody.
3. I think `gwak` would actually work fine. I tried replacing command and entrypoint commands on wmbusmeter docker container and they would print you without the `00` lines and if copy the datagram sample you will see that decodable just fine https://wmbusmeters.org/ . Maybe there is an issue I just could not see due to lack of logs...
4. OpenWRT has no `rtl_wmbus` available, so I can not use on to format outgoing data.
5. I did a test with `rtlwmbus:CMD(<command line>)` but found out that docker container does not have gawk package 
```
(rtlwmbus) no eol found, partial frame
(serial) received ascii "gawk: not found<0A>"
(rtlwmbus) checkRTLWMBusFrame "/bin/sh: gawk: not found<0A>"
```
just nc 
For me the main problem persists... How do I pass the data into wmbusmeter itself if it running not in pipe but in container. I tested overriding `command` or `entrypoint` commands to use piped input but it did not work even when setting a log file and without  no logs so have no clue how to fix it.
`nc` alone worked but it did not pass the data... so I think I will try to look in how `rtlwmbus` formatting looks  and use mosquitto_sub inside of CMD to pull the datagram in process it to match expected format...

closed 2024-04-25 by PovilasID

lmmodesto, 2024-04-25:
> 2. I have some meta data options enabled for `rtl_433` that is probably why datagram positions do not match but that is an esy fix for anybody.

Ok!

> 3. I think `gwak` would actually work fine. I tried replacing command and entrypoint commands on wmbusmeter docker container and they would print you without the `00` lines and if copy the datagram [...]

I think that you can use ```awk``` on the Openwrt machine to do the filtering and payload length correction parts.

I suggested ```awk``` on machine 1 and ```gawk``` on machine 2 just to reduce the CPU load on Openwrt, but maybe the overhead is too small and negligible. And you have a powerfull Openwrt machine...

> 4. OpenWRT has no `rtl_wmbus` available, so I can not use on to format outgoing data.

Yes, it seens that Openwrt doesn't have ```rtl_wmbus``` and ```wmbusmeters``` packages. 

Maybe they might need a machine as powerful as yours, although the decoding part done by ```rtl_433``` is still CPU intensive

> 5. I did a test with `rtlwmbus:CMD(<command line>)` but found out that docker container does not have gawk package
> 
> ```
> (rtlwmbus) no eol found, partial frame
> (serial) received ascii "gawk: not found<0A>"
> (rtlwmbus) checkRTLWMBusFrame "/bin/sh: gawk: not found<0A>"
> ```
> 

You can install it on the docker image and then use ```docker commit``` to cerate a new image

> just nc For me the main problem persists... How do I pass the data into wmbusmeter itself if it running not in pipe but in container. I tested overriding `command` or `entrypoint` commands to use piped input but it did not work even when setting a log file and without no logs so have no clue how to fix it. `nc` alone worked but it did not pass the data... so I think I will try to look in how `rtlwmbus` formatting looks and use mosquitto_sub inside of CMD to pull the datagram in process it to match expected format...
> 
> EDIT:
> 
> VICTORY! I found a way to use MQTT as a source for wmbusmeters. Here is the config:
> 
> ```
> loglevel=debug
> device=hex:CMD(/usr/bin/mosquitto_sub -h 192.168.x.x -t rtl_433/device/devices/6/Wireless-MBus/+/data | tr -d '\n'  )
> donotprobe=/dev/ttyAMA0
> logtelegrams=true
> format=json
> meterfiles=/wmbusmeters_data/logs/meter_readings
> meterfilesaction=overwrite
> logfile=/wmbusmeters_data/logs/wmbusmeters.log
> shell=/usr/bin/mosquitto_pub -h 192.168.x.x  -t water/"$METER_ID" -m "$METER_JSON"
> ```
> 
> 1. In the test folder I found `hex:CMD` that I could use as device.
> 2. Mqtt sub prints out a datagram you just need to trim end of line.
> 
> * is a wildcard symbol you should replace that with your meters id. To get full topic path use mqtt explorer
> 
> 3. I also added an option to publish processed data back into mqtt

Congrats! 

Thank you for sharing your configuration file. Maybe I'll use it as a reference later.

At this moment, I'm tryng to inspect ```wmbusmeters``` Izar driver code (see #1257). 

Maybe I'll try to port it to ```rtl_433``` in my spare time.

sippe2, 2024-04-28:
It looks like the same method can be used with ESP32+CC1101. I modified and "corrected" the https://github.com/chester4444/esp-multical21 code in a very ugly way and the end result is the same functionality.

PovilasID, 2024-04-28:
@sippe2 and [TTGO Lora32](https://www.lilygo.cc/products/lora3) these things are like 15-16 USD and if my neighbor asks how to pull in their water sensor data I would just tell them to buy that and put as close to bathroom as they can and give them config file with '*' for their device and just let filter it to their sensor.
Tinny TTGo devices can run rtl_433 but can not run wmbusmeter, so this is definitely the cheapest thing if you are looking for something specific.
No frequency tuning, cheap, low power, tiny footprint, almost self-contained, minimal config.
However, I have not tested their reliability...
Using MQTT as source permits pulling many more exotics sources because MQTT is soo ubiquitous.

sippe2, 2024-04-28:
@PovilasID 

This is very dirty proto with ESP32+CC1101 but it works. (ESP32 was something like 7€ and CC1101 was 2€)

[image]

Now confirmed that ESP8266+CC1101 works also with mqtt broker method. This is the cheapest solution. Total costs are under 6€ on this combo. In both chips, there is built-in 2.4GHz WiFi, so these are actually perfect for data collection when connecting a CC1101 chip to the ESP board. Only jumper wires are needed for the connection. 

Confirmed combinations:
ESP32 devkitv1 + cc1101
ESP8266 new nodemcu v3 + cc1101
ESP8266 nodemcu v2 + cc1101 (issues with debug serial port but works)
D1_MINI + cc1101

weetmuts, 2024-04-29:
@PovilasID This was cool:
>> device=hex:CMD(/usr/bin/mosquitto_sub -h 192.168.x.x -t rtl_433/device/devices/6/Wireless-MBus/+/data | tr -d '\n'  )

Would you like make a pull-request to the README updating the different ways to pipe data into wmbusmeters? "Using wmbusmeters in a pipe"

PovilasID, 2024-04-29:
@weetmuts something like this? https://github.com/wmbusmeters/wmbusmeters/pull/1262#issue-2269506375
I was not using it in pipe have not tested it but it should work

weetmuts, 2024-04-29:
Yep! Thanks!

Patbonamy, 2025-01-01:
> @PovilasID
> 
> This is very dirty proto with ESP32+CC1101 but it works. (ESP32 was something like 7€ and CC1101 was 2€)
> 
> ![IMG_6246](https://private-user-images.githubusercontent.com/31264116/326289829-f51667ff-5e4f-41eb-ba3c-93a1580d1d4a.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU3MzE2MzEsIm5iZiI6MTczNTczMTMzMSwicGF0aCI6Ii8zMTI2NDExNi8zMjYyODk4MjktZjUxNjY3ZmYtNWU0Zi00MWViLWJhM2MtOTNhMTU4MGQxZDRhLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTAxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDEwMVQxMTM1MzFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04Mzg2ZTA0MTc5YjI5NDg0ZGM0ZDNhNjhhZmJiNjA1MGJjNTAyOWM4NDY1Y2ZkYjliYWMyMWRjZGMxYTg3NTRjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.Q1KAHJPMWimQ4DFTgHxCFFCOW2ggLi_mzA9q5NJR7mo) At the moment, this small antenna has managed to listen to the data from 48 meters. I've had this prototype ready for several weeks now, but I haven't figured out a way to send the collected data to wmbusmeters. Now all the necessary pieces are in place.
> 
> Now confirmed that ESP8266+CC1101 works also with mqtt broker method. This is the cheapest solution. Total costs are under 6€ on this combo. In both chips, there is built-in 2.4GHz WiFi, so these are actually perfect for data collection when connecting a CC1101 chip to the ESP board. Only jumper wires are needed for the connection.
> 
> Confirmed combinations: ESP32 devkitv1 + cc1101 ESP8266 new nodemcu v3 + cc1101 ESP8266 nodemcu v2 + cc1101 (issues with debug serial port but works) D1_MINI + cc1101

Hi Povilas,
I have the same configuration I receive telegrams on my esphome but I don't know how to configure Wmbusmeters Addon to read the value how you do this?
Thanks and Happy NEW YEAR!!!

PovilasID, 2025-01-01:
@Patbonamy Happy NY to you too! 
In the `wmbusmeters.conf`
under `device` you need to listen to MQTT that comes from `rtl_433` something like:
```
device=hex:CMD(/usr/bin/mosquitto_sub -h 192.168.1.10 -t rtl_433/espdevice/devices/6/Wireless-MBus/+/data | tr -d '\n'  )
```

`192.168.1.10` is  IP of your MQTT server how you access that one you have figure it out.
`rtl_433/espdevice/devices/6/Wireless-MBus/+/data` is a topic  that the your ESP device outputs the device that you need listen too. You can check it using MQTT explorer. Note: + is wild card so if you are running in docker container or have multiple devices listening you can merge those input at this step.

On the shell input you have to set to publish processed wmbus singnals so it will look like something:
```
shell=/usr/bin/mosquitto_pub -h 192.168.1.10  -t wmbusmeters/"$METER_ID" -m "$METER_JSON"
```
It is a bit work around since data is routed from and back to MQTT but it works well and permits controlling rtl_433 devices independently which is cool. 
Have fun! Hope this helps.

jptuomi, 2025-09-15:
@PovilasID Have you had any problem with this solution recently? I've had the problem of the two first bytes missing and have been reading through the discussions here and in rtl_433, seems like we're back in a broken state at least for this solution...

https://github.com/merbanan/rtl_433/issues/3089
https://github.com/wmbusmeters/wmbusmeters/issues/1443