wmbusmeters/wmbusmeters #1223: seg fault in meters.cc line 682

FreeBear-nc, 2024-03-27:
Testing a L&R T230 heat meter.

```
~$ mbus-serial-scan-secondary -b 2400 /dev/ttyUSB0
Found a device on secondary address 68803557A7320704 [using address mask 6FFFFFFFFFFFFFFF]
```

Using 68803557A7320704 as the address, with wmbusmeters on the command line, I get:

```
~$ wmbusmeters --pollinterval=60s MAIN=/dev/ttyUSB0:mbus:2400 MyTempMeter auto:MAIN:mbus 68803557A7320704 NOKEY

Started config MAIN=mbus on /dev/ttyUSB0
AddressSanitizer:DEADLYSIGNAL
=================================================================
==20316==ERROR: AddressSanitizer: SEGV on unknown address (pc 0x55cde11bef79 bp 0x7fc5b41fd960 sp 0x7fc5b41fd5b0 T2)
==20316==The signal is caused by a READ memory access.
==20316==Hint: this fault was caused by a dereference of a high value address (see register values below).  Dissassemble the provided pc to learn which register was used.
    #0 0x55cde11bef79 in MeterCommonImplementation::poll(std::shared_ptr<BusManager>) src/meters.cc:682
    #1 0x55cde11991d9 in MeterManagerImplementation::pollMeters(std::shared_ptr<BusManager>) src/metermanager.cc:315
    #2 0x55cde1bba471 in regular_checkup(Configuration*) src/main.cc:447
    #3 0x55cde1bbc9e0 in operator() src/main.cc:643
    #4 0x55cde1bc5934 in __invoke_impl<void, start(Configuration*)::<lambda()>&> /usr/include/c++/10/bits/invoke.h:60
    #5 0x55cde1bc3d9d in __invoke_r<void, start(Configuration*)::<lambda()>&> /usr/include/c++/10/bits/invoke.h:153
    #6 0x55cde1bc26e1 in _M_invoke /usr/include/c++/10/bits/std_function.h:291
    #7 0x55cde129ba13 in std::function<void ()>::operator()() const /usr/include/c++/10/bits/std_function.h:622
    #8 0x55cde128734f in SerialCommunicationManagerImp::executeTimerCallbacks() src/serial.cc:884
    #9 0x55cde1287c5c in SerialCommunicationManagerImp::timerLoop() src/serial.cc:929
    #10 0x55cde1283283 in operator() src/serial.cc:680
    #11 0x55cde12990a0 in __invoke_impl<void, SerialCommunicationManagerImp::SerialCommunicationManagerImp(time_t, bool)::<lambda()>&> /usr/include/c++/10/bits/invoke.h:60
    #12 0x55cde1298bf9 in __invoke_r<void, SerialCommunicationManagerImp::SerialCommunicationManagerImp(time_t, bool)::<lambda()>&> /usr/include/c++/10/bits/invoke.h:153
    #13 0x55cde12985c1 in _M_invoke /usr/include/c++/10/bits/std_function.h:291
    #14 0x55cde129ba13 in std::function<void ()>::operator()() const /usr/include/c++/10/bits/std_function.h:622
    #15 0x55cde12b7a77 in dispatch(void*) src/threads.cc:56
    #16 0x7fc5ba0b7ea6 in start_thread (/lib/x86_64-linux-gnu/libpthread.so.0+0x7ea6)
    #17 0x7fc5b9caaa6e in __clone (/lib/x86_64-linux-gnu/libc.so.6+0xfba6e)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV src/meters.cc:682 in MeterCommonImplementation::poll(std::shared_ptr<BusManager>)
Thread T2 created by T0 here:
    #0 0x7fc5ba2e52a2 in __interceptor_pthread_create ../../../../src/libsanitizer/asan/asan_interceptors.cpp:214
    #1 0x55cde12b7ba3 in startTimerLoopThread(std::function<void ()>) src/threads.cc:74
    #2 0x55cde1283722 in SerialCommunicationManagerImp::SerialCommunicationManagerImp(long, bool) src/serial.cc:680
    #3 0x55cde128b2d1 in createSerialCommunicationManager(long, bool) src/serial.cc:1060
    #4 0x55cde1bbdd41 in start(Configuration*) src/main.cc:542
    #5 0x55cde1b91ab5 in main src/main.cc:177
    #6 0x7fc5b9bd2d09 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x23d09)

==20316==ABORTING
```

Although the error is reported as meters.cc:682, I believe the preceding line is at fault:
```
        AddressExpression &ae = addressExpressions().back();
        if (ae.has_wildcard)
```
I'm of the opinion that ae needs a check to confirm it is valid else exit with an error.

After working out what I was doing wrong, I got this result:
```
~$ wmbusmeters --pollinterval=60s MAIN=/dev/ttyUSB0:mbus:2400 MyTempMeter qheat_55_us:MAIN:mbus p12 NOKEY
Started config MAIN=mbus on /dev/ttyUSB0
MyTempMeter	68803557	0 kWh	0 kWh	2024-03-27 21:28:34
```
Side note: the ultraheat driver is the obvious one to use with the L&G T230, and appears to report flow & return temperatures correctly.

weetmuts, 2024-03-28:
Thanks for the bug report! There was no support for parsing the libmbus secondary address. But I have now added support for this! Also the segfault has been translated into a warning if somehow a bad address would get to that part of wmbusmeters.

Please test again!

weetmuts, 2024-03-28:
A fully specified libmbus secondary address is 16 hex digits, for example:
100002842941011B
this includes the id 10000284 and the manufacturer 2941 (translates to PII) and the version 01 and the type 1b.
wmbusmeters had support for the fully specified address:
10000284.M=PII.V=01.T=1b
where you can leave out M,V,T as you like.

But since libmbus is frequently used, I added support for their version of a fully specified address as well. :-)

weetmuts, 2024-03-28:
As for the driver, I am working on a new text based driver system. As long as you get good data, the warning can be ignored. It is probably just another version of the meter but with a lot of similar values, perhaps all.
Please post a telegram here so that we can take a look.

FreeBear-nc, 2024-03-28:
Landis+Gyr rebadge some of their meters for other suppliers. e.g. The ista ultego 3 is exactly the same as the T230 even down to the ordering information.

Currently exploring the driver.xmq to save hacking around with the ultraheat.cc source - Any documentation available aside from the dvparser.h ?

FreeBear-nc, 2024-03-28:
> Thanks for the bug report! There was no support for parsing the libmbus secondary address. But I have now added support for this! Also the segfault has been translated into a warning if somehow a bad address would get to that part of wmbusmeters.
> 
> Please test again!

Appears to be working, and exits gracefully with an invalid address.

And in response to your request of a telegram:
```
~$ wmbusmeters --pollinterval=60s --logtelegrams MAIN=/dev/ttyUSB0:mbus:2400 t230 auto:MAIN:mbus 68803557A7320704 NOKEY
Started config MAIN=mbus on /dev/ttyUSB0
telegram=|68E2E268080C7257358068A732070403100000|0974040970080C06000000000C14000000000B2D0000000B3B0000000B5A7701000B5E9001000B621300F00C7857358068891071073C22467206000C22467206000C26000000008C901006000000009B102D0000009B103B0000009B105A3104009B105E1105009410AD6F000000009410BB6F000000009410DA6F2F07B4279410DE6F08091A274C06000000004C14000000007C22525106004C2600000000CC90100600000000DB102D000000DB103B000000DB105A310400DB105E110500848F0F6D0000E1F1046D34061C330F1407006601|2516|+5
t230	68803557	2024-03-28 12:46:36
```

closed 2024-03-29 by FreeBear-nc