wmbusmeters/wmbusmeters #1225: mfct: (IMT) INTEGRA METERING, Switzerland (0x25b4) type: Water meter (0x07) ver: 0x01

tbandixen, 2024-03-29:
### Type of request

New device support

### Meter description

Started auto rtlwmbus[00000001] listening on t1
(meter) water: meter detection could not find driver for id: 05743967 mfct: (IMT) INTEGRA METERING, Switzerland (0x25b4) type: Water meter (0x07) ver: 0x01
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) newly created meter (water 10066838,R05743967 auto) did not handle telegram!

### Logged telegrams from meter in decrypted format

```shell
did not handle telegram...
```

gh0stface33, 2024-04-03:
@tbandixen 
Hi,
I have same watermeter.
Have you a decryption key ?
Can you share your key or try to decrypt my telegram with your key ?
Thanks

https://wmbusmeters.org/analyze/4E44B4252178101010077A1A00402571B721D881E204544353B85C1E05267927264C75D8ABAB21F6D7EC8081CD4494C59D1F24FD590FB5B253F768FA3DF8FA3DFC8404AB6910ABC01C17DCCCC38C24E06449

tbandixen, 2024-04-03:
@gh0stface33 
It looks like you're already one step further.
Which driver did you choose? I got an app (ParamApp) from the manufacturer where I can set an AES key myself. But I haven't got that far yet. Because I don't receive Telegram. My meter is in the initial state and I assume that no key is stored. But I could set it myself. But I don't have the right driver.

gh0stface33, 2024-04-04:
@tbandixen 
Thanks
To decrypt telegram you don't need correct driver.
To have correct driver you need to open an issue to request a new driver, you have to send decrypted telegram.

tbandixen, 2024-04-04:
@gh0stface33 
Now I got the telegrams, but still no driver.

https://wmbusmeters.org/analyze/4644B42538680610050e7267397405B42501070e0030252f2f04164105000084101600000000046d2a32043444163e050000426c1f3302fd17000002fd74c8162f2f2f2f2f2f2f:auto:44b05d756d184b4a44b05d756d184b4a

tbandixen, 2024-04-04:
I get these msgs all the time:

```shell
Started auto rtlwmbus listening on t1
telegram=|4644B42538680610050E7267397405B4250107A90030252F2F_04164105000084101600000000046D1733043444163E050000426C1F3302FD17000002FD74C8162F2F2F2F2F2F2F|+712
Lost  closing rtlwmbus
Started auto rtlwmbus listening on t1
Lost  closing rtlwmbus
Started auto rtlwmbus listening on t1
Lost  closing rtlwmbus
Started auto rtlwmbus listening on t1
Lost  closing rtlwmbus
telegram=|4644B42538680610050E7267397405B4250107AA0030252F2F_04164105000084101600000000046D1733043444163E050000426C1F3302FD17000002FD74C8162F2F2F2F2F2F2F|+728
Started auto rtlwmbus listening on t1
Lost  closing rtlwmbus
Started auto rtlwmbus listening on t1
Lost  closing rtlwmbus

```

tbandixen, 2024-04-05:
If I remove the AES Key i'll get the data.... sorry, can be closed

closed 2024-04-05 by tbandixen