wmbusmeters/wmbusmeters #1237: Water 5.5 Qundis

cool338, 2024-04-06:
Explain the problem..... 

https://wmbusmeters.org/analyze/49449344806281141a063246780dff5f350082c20035cdB226a05d9da9de22ecef0B3ec22a8381cd96234e7ca10a56c23e189dd3de4c7B1292161162681acB9559e456f314699da396d39f29046d320B0634675d 

[12:50:51][V][rxLoop:167]: Have 86 bytes from CC1101 Rx, RSSI: -51 dBm LQI: 128
[12:50:51][D][mbus:013]: Processing C1 A frame
[12:50:51][V][mbus:017]: Frame: 49449344806281141A063246780DFF5F350082C20035CDB226A05D9DA9DE22ECEF0B3EC22A8381CD96234E7CA10A56C23E189DD3DE4C7B1292161162681ACB9559E456F314699DA396D39F29046D320B0634675D (84) [with CRC]
[12:50:51][V][mbus:095]: Validating CRC for Block1
[12:50:51][V][crc:031]:     calculated: 0x3246, read: 0x3246
[12:50:51][V][mbus:115]: Validating CRC for Block2
[12:50:51][V][crc:031]:     calculated: 0xA9DE, read: 0xA9DE
[12:50:51][V][mbus:115]: Validating CRC for Block3
[12:50:51][V][crc:031]:     calculated: 0x56C2, read: 0x56C2
[12:50:51][V][mbus:115]: Validating CRC for Block4
[12:50:51][V][crc:031]:     calculated: 0x59E4, read: 0x59E4
[12:50:51][V][mbus:115]: Validating CRC for Block5
[12:50:51][V][crc:031]:     calculated: 0x675D, read: 0x675D
[12:50:51][V][mbus:062]: Frame: 49449344806281141A06780DFF5F350082C20035CDB226A05D9D22ECEF0B3EC22A8381CD96234E7CA10A3E189DD3DE4C7B1292161162681ACB9556F314699DA396D39F29046D320B0634 (74) [without CRC]

[image]

https://wmbusmeters.org/analyze/49449344713180141A0763FB780DFF5F350082C30035AE4F225B383E1A19394980BB13B4CBE18D33E03772CD137E1AD00C95A74D9445050DD7D0F2263E25E46FF7B89F584E890CBEA985ED94046D320B0634798D

[12:50:47][V][rxLoop:167]: Have 86 bytes from CC1101 Rx, RSSI: -58 dBm LQI: 128
[12:50:47][D][mbus:013]: Processing C1 A frame
[12:50:47][V][mbus:017]: Frame: 49449344713180141A0763FB780DFF5F350082C30035AE4F225B383E1A19394980BB13B4CBE18D33E03772CD137E1AD00C95A74D9445050DD7D0F2263E25E46FF7B89F584E890CBEA985ED94046D320B0634798D (84) [with CRC]
[12:50:47][V][mbus:095]: Validating CRC for Block1
[12:50:47][V][crc:031]:     calculated: 0x63FB, read: 0x63FB
[12:50:47][V][mbus:115]: Validating CRC for Block2
[12:50:47][V][crc:031]:     calculated: 0x1A19, read: 0x1A19
[12:50:47][V][mbus:115]: Validating CRC for Block3
[12:50:47][V][crc:031]:     calculated: 0x1AD0, read: 0x1AD0
[12:50:47][V][mbus:115]: Validating CRC for Block4
[12:50:47][V][crc:031]:     calculated: 0xF7B8, read: 0xF7B8
[12:50:47][V][mbus:115]: Validating CRC for Block5
[12:50:47][V][crc:031]:     calculated: 0x798D, read: 0x798D
[12:50:47][V][mbus:062]: Frame: 49449344713180141A07780DFF5F350082C30035AE4F225B383E394980BB13B4CBE18D33E03772CD137E0C95A74D9445050DD7D0F2263E25E46F9F584E890CBEA985ED94046D320B0634 (74) [without CRC]

[image]

Kann mir jemand erklären was ich falsch mache?

Und ist es möglich Rauchmelder mit zu intrigieren? 

Can someone explain to me what I'm doing wrong? 

And is it possible to integrate smoke detectors?
https://wmbusmeters.org/analyze/58442515667383020C1AEFE47AA10000000BFD0F000102046D370A06710B3402FD17000082206CE828426C1033846B7B40FF2C000F11008250FD61000082506C232601018260FD6100008360FD31000000823706606C01018270FD61000082706C01013F9C

gotsila, 2024-04-27:
Hallo,
benutze als Treiber: qwater nicht auto auch nicht den wme5.
Interessant wäre noch welchen Empfänger-Stick du benutzt. Ich benutze den iM871A-USB - Wireless M-Bus USB-adapter 868 MHz und damit habe ich in einem 3 Familienhaus einige qwater und qheat am laufen.

cool338, 2024-04-28:
> Hallo, benutze als Treiber: qwater nicht auto auch nicht den wme5. Interessant wäre noch welchen Empfänger-Stick du benutzt. Ich benutze den iM871A-USB - Wireless M-Bus USB-adapter 868 MHz und damit habe ich in einem 3 Familienhaus einige qwater und qheat am laufen.

moin,

Also laut Wasser Uhr 2,138m³ im daten Paket steht was anderes.
{
    "media":"water",
    "meter":"qwater",
    "name":"",
    "id":"14803171",
    "due_date":"2024-12-25",
    "due_17_date":"2048-11-18",
    "due_17_date_m3":68661.528,
    "due_date_m3":66231.797,
    "total_m3":199690.017,
    "meter_datetime":"2024-04-28 08:12",
    "status":"OK",
    "timestamp":"2024-04-28T07:15:58Z"
}
49449344713180141a07780dff5f3500822400357a0862603f6835da9868Bc193c97e7f164126B1B156668aB71195B91d7120642e850e0ac4ef057fe7df0811e8f2981ad046d0c081c34

hin und wieder bekomm ich auch so welche Pakete

`[09:16:45][V][mbus:017]: Frame: 48449344713180141A0758687A250030253E2A8A3C6EE9DAF3D8715ACEDB29E343B9FC9847121AD193FA5083810D052224697CA3F001D5484E2977A4F9073923608C4D1B69CB3C326CFFFF046D10081C34F2A1 (83) [with CRC]
[09:16:45][V][mbus:095]: Validating CRC for Block1
[09:16:45][V][crc:031]:     calculated: 0x5868, read: 0x5868
[09:16:45][V][mbus:115]: Validating CRC for Block2
[09:16:45][V][crc:031]:     calculated: 0xCEDB, read: 0xCEDB
[09:16:45][V][mbus:115]: Validating CRC for Block3
[09:16:45][V][crc:031]:     calculated: 0x0522, read: 0x0522
[09:16:45][V][mbus:115]: Validating CRC for Block4
[09:16:45][V][crc:031]:     calculated: 0x608C, read: 0x608C
[09:16:45][V][mbus:115]: Validating CRC for Block5
[09:16:45][V][crc:031]:     calculated: 0xF2A1, read: 0xF2A1
[09:16:45][V][mbus:062]: Frame: 48449344713180141A077A250030253E2A8A3C6EE9DAF3D8715A29E343B9FC9847121AD193FA5083810D24697CA3F001D5484E2977A4F90739234D1B69CB3C326CFFFF046D10081C34 (73) [without CRC]`

ich nutze CC1101 Antennen und esp32. Alles in HA

TECNOIOT 2pcs CC1101 868MHz Wireless Module Long Distance Transmission with Antenna
[https://www.amazon.de/dp/B07RH67FWV?psc=1&ref=ppx_yo2ov_dt_b_product_details](url)

MfG

gotsila, 2024-04-28:
Morgen,
fragst du nur c1 Protokoll ab? bei mir Device= /dev/ttyUSB1:im871a:c1
Weil das Datenpaket sieht grundlegend falsch aus. Due_17_date sollte letzte Tag vom letzen Monat sein.
Ansonsten würde ich empfehlen mal nen anderen Stick (idealerweise den  iM871A-USB) probieren, evtl. bei nem Kollegen leihen. ich setzte das auch alles unter HA ein.
Andere Frage hast du noch mehr Zähler oder ist das der einzige.

Bei mir sieht das so aus. :
`(shell) env "METER_JSON={"media":"warm water","meter":"qwater","name":"51946926_qwater_warm_dg","id":"51946926","due_date":"2023-12-31","due_17_date":"2024-03-31","due_17_date_m3":6.856,"due_date_m3":0.034,"error_date":"2128-03-31","total_m3":9.757,"volume_flow_m3h":0,"meter_datetime":"2024-04-28 08:39","status":"OK","timestamp":"2024-04-28T07:39:33Z","device":"im871a[00103921]","rssi_dbm":-74}"`

in HA:
[image]

cool338, 2024-04-28:
Moin,
Also ich nutze den code hier der ist von https://github.com/zibous/ha-watermeter
womöglich da drin der fehler?
Es läuft alles über esphome

`
## ------------------------------------------------------------------
##           WMBUS CC1101 --> ESP32 az-delivery-devkit-v4
## ------------------------------------------------------------------
##
##
##                                                               o 1 (3.3V)
##                                                               |
##   ╭――x――x――x――x――x――x――x――x――x――x――x――x――x――x――x――x――x――x――x――o―╮
##   |                                                             |
##   |                          EP32                               |
## - | 5v               az-delivery-devkit-v4                      | -- ANT
##   |                                                             |
##   |                          16 17 5  18 19               23    |
##   ╰――x――x――x――x――x――x――x――x――o――x――o――o――o――o――o――o――o――o――o――o―╯
##                              |  |  |  |  |                 |   |
##                              o  |  |  o  |                 |   ╰-o - 2 (GND)
##                              7  o  |  4  o                 o
##                            GDO0 6  | CLK 5                 3
##                               GD02 o    MISO              M0SI
##                                    8
##                                   CSN
##
## ------------------------------------------------------------------
wmbus:
  mosi_pin: GPIO23  ## SI:   braun   3: MOSI Attached to Hardware SPI controller MOSI SPI Interface
  miso_pin: GPIO19  ## SO:   grün    5: MISO Attached to Hardware SPI controller MISO SPI Interface
  clk_pin: GPIO18   ## SCLK: violett 4: SCK  Attached to Hardware SPI controller CLK
  cs_pin: GPIO05    ## CSN:  orange  8: CSN  Attached to Hardware SPI controller
  gdo0_pin: GPIO16  ## GD00: gelb    7: RX Clock output. High Impedance !
  gdo2_pin: GPIO17  ## GD02: weiss   6: TX FIFO status signals. High Impedance !

  # log_unknown (Optional): Show telegrams from not configured meters in log. Defaults to False
  log_unknown: False

  ## -------------------------------------------------
  ## Using wmbusmeters i a pipe (optional)
  ## uncomment for using
  ## -------------------------------------------------
  # clients:
  #  - name: "wmbusmeters"
  #    ip_address: !secret wmbusmeter_host
  #    port: 7227
  #    format: rtlwMBus
  #    transport: UDP

  ## -------------------------------------------------
  ## setting: wmbusmeters.conf on wbusmeters server
  ## -------------------------------------------------
  ## loglevel: normal
  ## device: rtlwmbus:CMD(nc -lku 7227)
  ## donotprobe: /dev/ttyAMA0
  ## logtelegrams: false
  ## format: json
  ## logfile: /dev/stdout
  ## shell: /wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"

  ## -------------------------------------------------
  ## led blink mode (optional) xx = PIN
  ## -------------------------------------------------
  # led_pin: GPIOxx
  # led_blink_time: "100ms"

## ---------------------------------------------------
## BINARY SENSOR
[... 204 lines cut ...]
        then:
          - lambda: |-
              int error_code = int(x);
              std::string message = "";
              if(error_code==0){
                  message = id(alarm_error_text)[error_code];
                  id(watermeter_alarm_message).publish_state(message);
                  ESP_LOGD("wmbus", "Alarm message: %s, error code: %d", message, error_code);
              }else{
                for (int i = 1; i < 10; ++i) {
                  if (error_code & (1 << i)) {
                    if (!message.empty()) {
                      message += ", ";
                    }
                    message += id(alarm_error_text)[i];
                  }
                }
                ESP_LOGW("wmbus", "WARNING Alarm message: %s, error code: %d", message.c_str(), error_code);
                id(watermeter_alarm_timestamp).publish_state(id(time_sntp).now().strftime("%Y-%m-%dT%H:%M:%S %Z").c_str());
              }
              id(watermeter_alarm_perv_message).publish_state(message);
              if(id(watermeter_alarm_timestamp).has_state() == false) {
                id(watermeter_alarm_timestamp).publish_state("--");
              }
`

closed 2024-04-28 by cool338

gotsila, 2024-04-28:
Uff, also dazu kann ich nix sagen. Aber hier sind wir ja bei wmbusmeters und ich nutze das HA-AddOn,  
https://github.com/wmbusmeters/wmbusmeters-ha-addon

Denke die Frage wäre in dem anderen Fork besser aufgehoben, deshalb gibts hier vermutlich keine Hilfe.

cool338, 2024-04-28:
> Uff, also dazu kann ich nix sagen. Aber hier sind wir ja bei wmbusmeters und ich nutze das HA-AddOn, https://github.com/wmbusmeters/wmbusmeters-ha-addon
> 
> Denke die Frage wäre in dem anderen Fork besser aufgehoben, deshalb gibts hier vermutlich keine Hilfe.

Ja das ist auch das Problem wohl. 

Das addon kann nicht auf den esp32 zugreifen und deswegen klappt es wohl net so ganz. 

Dennoch danke sehr für die Hilfe :)

MfG

sippeeey, 2024-06-07:
@cool338 ich habe genau das gleiche Problem, dass die Werte, die ausgelesen werden, nicht der Realität entsprechen.

Beispiel: https://wmbusmeters.org/analyze/49449344708828521a072198780dff5f350082a70035865B2883cf79596625201701faccdcedd4506930ee48B78676cf439BeB7e6054de780a19938a15235695B5c3e7af7140e85Bfaefe1dd046d011307365690

Tatsächlicher Verbrauch: 1885m³

Ich setze ebenfalls das ESP32 Projekt und habe diese Antenne: https://www.amazon.de/dp/B075PFQ57G?psc=1&ref=ppx_yo2ov_dt_b_product_details (Laqiya CC1101 868MHZ Funkübertragung Antenne Transceiver Modul)

Hast du es bereits lösen können?