wmbusmeters/wmbusmeters #1241: Handle compressed telegrams

vohum, 2024-04-11:
### Type of request

Existing device improvement

### Meter description

I'm trying to read Kamstrup Heat meter wmbus data is transfered by signal repeater, but there are a lot of meters in range and the repeater misses some frames because it cannot send everything(10-15 frames per second). Is it possible to parse a compressed frame? I see energy data in telegram.

compressed:
`25442D2C53160567300C8D2092C155F824389079C252C471_**90DE00**E9DD0034EE171F33000000`
full frame:
`2E442D2C53160567300C8D2093D055F824AD3A78_030E**90DE00**430EE9DD00031334EE17426C1F33022D000001FF2100`

**90DE00** - energy value

Regards

### Logged telegrams from meter in decrypted format

```shell
25442D2C53160567300C8D2092C155F824389079C252C471_90DE00E9DD0034EE171F33000000

2E442D2C53160567300C8D2093D055F824AD3A78_030E90DE00430EE9DD00031334EE17426C1F33022D000001FF2100
```

weetmuts, 2024-04-11:
wmbusmeters already supports compressed telegrams, but it has to receieve the full length first before it can decode the compressed one.
Put this content into the file simulation_test.txt
telegram=2E442D2C53160567300C8D2093D055F824AD3A78_030E90DE00430EE9DD00031334EE17426C1F33022D000001FF2100
telegram=25442D2C53160567300C8D2092C155F824389079C252C471_90DE00E9DD0034EE171F33000000

Then run
wmbusmeters --debug simulation_foo.txt HEAT kamheat 67051653 NOKEY

If you read the debug output, then you ca see that it properly decoded the second compressed telegram.

weetmuts, 2024-04-11:
If you restart wmbusmeters between telegrams, then it will lose the memory of the full telegram format.

vohum, 2024-04-11:
thanks for the information, I guess the error is in a different place than I thought, sometimes the telegrams I receive do not match the key, so I will try to increase the baudrate of receiver

closed 2024-04-12 by vohum