wmbusmeters/wmbusmeters #1242: Some kamstrup compressed telegrams cannot be parsed.

vohum, 2024-04-12:
### Type of request

Enhancement

### OS version

Ubuntu

### wmbusmeters version

1.15

### Your message goes here

wmbusmetees interprets it as an incorrect decryption key, but it seems to me that the error is in the TPL block, I am sending 3 incorrect telegrams that I received and last correct one, I am a bit worried because some meters send 99% of incorrect telegrams 

key:
`3AF0304A8EB6C4EBF38DDD9D91F3BF8B`

```
_25442D2C89140567300C8D305891A4F824C87C007CA8C6AD5DCD3C527BED3C2A5582B0F9E124
_25442D2C89140567300C8D208860A5F82409D669C67DD3039EADAF1C794AF4BBB582CFFE2AFA
_25442D2C89140567300C8D208652A5F8244079B94F7C04730E3EDF986F31E2C7EFE9012BC1A5
_25442D2C89140567300C8D208451A5F824B9F392372B872FAD41BFD9917ACB4A8386ED6EBAE4
```

weetmuts, 2024-04-12:
When the wrong key is used for AES CTR, then the payload crc will fail, since the decrypted content will not match the crc. You can see this  when using wmbusmeters.org or --analyze 

The Kamstrup meters are very well tested by many users, so, I think you just have the wrong key. :-) The same question has been asked several times before and in every single case the last couple of years have been because the key was wrong. 
 
You say 99%? If you can find a meter that both fails decryption 99% and succeeds in decryption 1%
and you manage to collect these two types of telegrams success/fail. Then I can debug this. :-)

vohum, 2024-04-12:
Correct telegram debug:

`wmbusmeters --debug _25442D2C89140567300C8D208860A5F82409D669C67DD3039EADAF1C794AF4BBB582CFFE2AFA test kamheat 67051489 3AF0304A8EB6C4EBF38DDD9D91F3BF8B
(config) using device: <_25442D2C89140567300C8D208860A5F82409D669C67DD3039EADAF1C794AF4BBB582CFFE2AFA>:any 
(config) number of meters: 1
(lookup) with file/hex "_25442D2C89140567300C8D208860A5F82409D669C67DD3039EADAF1C794AF4BBB582CFFE2AFA" unknown
(lookup) driver: simulation hex
(main) opening <_25442D2C89140567300C8D208860A5F82409D669C67DD3039EADAF1C794AF4BBB582CFFE2AFA>:any
Started config simulation listening on any
(simulation) in 
(main) regular reset of simulation  will happen every 82800 seconds
(main) added  to files
(wmbus) no alarm (expected activity) for simulation
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(simulation) from file "_25442D2C89140567300C8D208860A5F82409D669C67DD3039EADAF1C794AF4BBB582CFFE2AFA"
(mbus) checkMBUSFrame
 "25442D2C89140567300C8D208860A5F82409D669C67DD3039EADAF1C794AF4BBB582CFFE2AFA"
(wmbus) checkWMBUSFrame "25442D2C89140567300C8D208860A5F82409D669C67DD3039EADAF1C794AF4BBB582CFFE2AFA"
(simulator) is_mbus=false is_wmbus=true
(simulator) is wmbus telegram.
(meter) no meter handled  checking 1 templates.
(wmbus) parseDLL @0 38
(wmbus) parseELL @10 28
(meter) test: for me? 67051489 in 
(meter) test: yes for me
(meter) created test kamheat 67051489 encrypted
(meter) used meter template test  kamheat to match 67051489
(meter) started meter 1 (test 67051489 kamheat)
(wmbus) parseDLL @0 38
(wmbus) parseELL @10 28
(meter) test: for me? 67051489 in 67051489
(meter) test: yes for me
(meter) test(1) kamheat  handling telegram from 67051489
(meter) test 67051489 "25442D2C89140567300C8D208860A5F82409D669C67DD3039EADAF1C794AF4BBB582CFFE2AFA"
(wmbus) parseDLL @0 38
(telegram) DLL L=25 C=44 (from meter SND_NR) M=2c2d (KAM) A=67051489 VER=30 TYPE=0c (Heat volume at inlet meter) (driver kamheat) DEV= RSSI=0
(wmbus) parseELL @10 28
(ELL) decrypting "09D669C67DD3039EADAF1C794AF4BBB582CFFE2AFA"
(ELL) IV 2D2C89140567300C2060A5F824000000
(ELL) block 0 block_size 16 offset 0
(ELL) decrypted "7A1E79C2525B894D88004D8800B81009"
(ELL) block 1 block_size 5 offset 16
(ELL) decrypted "1F33000000"
(ELL) decrypted "7A1E79C2525B894D88004D8800B810091F33000000"
(telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=88 SN=60a5f824 (AES_CTR session=0 time=5212758) CRC=7a1e
(wmbus) parseNWL @19 19
(wmbus) parseAFL @19 19
(wmbus) parseTPL @19 19
(wmbus) ignoring compressed telegram since format signature hash 0x52c2 is yet unknown.
     this is not a problem, since you only need wait for at most 8 telegrams
     (8*16 seconds) until an full length telegram arrives and then we know
     the format giving this hash and start decoding the telegrams properly.
(meter) newly created meter (test 67051489 kamheat) did not handle telegram!
(wmbus) telegram from 67051489 ignored by all configured meters!
(serial) stopping manager
(serial) waiting for stop
(serial) TIMER thread interrupted
(serial) EVENT thread interrupted
(serial) event loop stopped!
(wmbus) deleted simulation
`

Incorrect telegram debug:

```
wmbusmeters --debug _25442D2C89140567300C8D208860A5F82409D669C67DD3039EADAF1C794AF4BBB582CFFE2AFA test kamheat 67051489 3AF0304A8EB6C4EBF38DDD9D91F3BF8B
(config) using device: <_25442D2C89140567300C8D305891A4F824C87C007CA8C6AD5DCD3C527BED3C2A5582B0F9E124>:any 
(config) number of meters: 1
(lookup) with file/hex "_25442D2C89140567300C8D305891A4F824C87C007CA8C6AD5DCD3C527BED3C2A5582B0F9E124" unknown
(lookup) driver: simulation hex
[... 23 lines cut ...]
(meter) test: for me? 67051489 in 67051489
(meter) test: yes for me
(meter) test(1) kamheat  handling telegram from 67051489
(meter) test 67051489 "25442D2C89140567300C8D305891A4F824C87C007CA8C6AD5DCD3C527BED3C2A5582B0F9E124"
(wmbus) parseDLL @0 38
(telegram) DLL L=25 C=44 (from meter SND_NR) M=2c2d (KAM) A=67051489 VER=30 TYPE=0c (Heat volume at inlet meter) (driver kamheat) DEV= RSSI=0
(wmbus) parseELL @10 28
(ELL) decrypting "C87C007CA8C6AD5DCD3C527BED3C2A5582B0F9E124"
(ELL) IV 2D2C89140567300C3091A4F824000000
(ELL) block 0 block_size 16 offset 0
(ELL) decrypted "DADC41F4511D3F873F101DB2EA7FF739"
(ELL) block 1 block_size 5 offset 16
(ELL) decrypted "FC64AAED69"
(ELL) decrypted "DADC41F4511D3F873F101DB2EA7FF739FC64AAED69"
(wmbus) WARNING! decrypted payload crc failed check, did you use the correct decryption key? dadc payload crc (calculated c519) Permanently ignoring telegrams from id: 67051489 mfct: (KAM) Kamstrup Energi (0x2c2d) type: Heat volume at inlet meter (0x0c) ver: 0x30
(telegram) ELL CI=8d CC=30 (slow_resp sync relayed) ACC=58 SN=91a4f824 (AES_CTR session=1 time=5212745) CRC=dadc
(meter) newly created meter (test 67051489 kamheat) did not handle telegram!
(wmbus) telegram from 67051489 ignored by all configured meters!
(serial) stopping manager
(serial) waiting for stop
(serial) TIMER thread interrupted
(serial) EVENT thread interrupted
(serial) event loop stopped!
(wmbus) deleted simulation
```

weetmuts, 2024-04-12:
Ok, so I put these telegrams in a simulation file:
telegram=|2E442D2C53160567300C8D2093D055F824AD3A78_030E90DE00430EE9DD00031334EE17426C1F33022D000001FF2100|
telegram=|25442D2C89140567300C8D208860A5F82409D669C67DD3039EADAF1C794AF4BBB582CFFE2AFA|
telegram=|25442D2C89140567300C8D305891A4F824C87C007CA8C6AD5DCD3C527BED3C2A5582B0F9E124|

(The first is from another meter that you provided, its ignored here, except that it provides the full frame so that we can decompress the short versions.)

The from the command:
./build/wmbusmeters --debug simulation_good_bad.txt test kamheat 67051489 3AF0304A8EB6C4EBF38DDD9D91F3BF8B > banan 2>&1

And then skip the first telegram output, put the next telegram into alfa, and the third telegram in beta.
Then diff alfa and beta.

The second telegram does indeed decrypted bad. However there is a difference in the 
ell-cc field between the two telegrams. The working one has no relayed bit, but the failing one
has the relayed bit set.

(telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=88 SN=60a5f824 (AES_CTR session=0 time=5212758) CRC=7a1e
(telegram) ELL CI=8d CC=30 (slow_resp sync relayed) ACC=58 SN=91a4f824 (AES_CTR session=1 time=5212745) CRC=dadc

Do you have a repeater somewhere?

Could it be that my IV calculation for the crypto should depend on this bit, but doesn't or vice versa.

//Fredrik

weetmuts, 2024-04-12:
The session nr is also different, perhaps that should affect the IV too, or not. Hmmm.

weetmuts, 2024-04-12:
Ok, there is a problem. When the telegram is repeated, then the IV should NOT include the hop-bit 0x10 nor
the R-field bit 0x20. 

In this case, this can be fixed by modifying the line
 iv[i++] = t->ell_cc;
TO
 iv[i++] = t->ell_cc & ~(1 << 4) & ~(1 << 1);
in
wmbus_utils.cc

unfortunately it still does not decrypt properly.
I have to understand if the session=1 is relevant...

Anyway, this explains your problem. 99% of the telegrams are repeated and cannot be decrypted.
1% of the telegrams reach wmbusmetes without the repeater, and can be decrypted!

weetmuts, 2024-04-13:
Can you run wmbusmeters from your location where you can receive both the repeated message and the original message almost every time. Then collect a log of ~50 telegrams, both repeated and original messages, for a single meter.
Then email me this log.

I need more data to debug this.

weetmuts, 2024-04-13:
What kind of repeater is it?

weetmuts, 2024-04-13:
I have pushed fix where the HOP bit is zeroed out. Please test, perhaps there is an improvement somewhere.

vohum, 2024-04-13:
Works perfect thanks! :) Repeater is RC1180.

closed 2024-04-13 by vohum

weetmuts, 2024-04-13:
Great news! But can you post a relayed telegram from the same meter you posted the key for, 
that now successfully decrypts. I need to add this to the test-suite to prevent regressions.

vohum, 2024-04-16:
some decrypted telegrams
```
Apr 16 07:19:11  telegram=|25442D2C89140567300C8D2046F10EFA247A1E79C2525B89_4D88004D8800B810091F33000000|+22668
Apr 16 07:19:24  telegram=|25442D2C89140567300C8D2047000FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+22681
Apr 16 07:19:40  telegram=|25442D2C89140567300C8D2048000FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+22697
Apr 16 07:19:56  telegram=|2E442D2C89140567300C8D2049010FFA2432C278_030E4D8800430E4D88000313B81009426C1F33022D000001FF2100|+22713
Apr 16 07:20:12  telegram=|25442D2C89140567300C8D204A100FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+22729
Apr 16 07:20:28  telegram=|25442D2C89140567300C8D204B110FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+22745
Apr 16 07:20:44  telegram=|25442D2C89140567300C8D204C110FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+22761
Apr 16 07:21:00  telegram=|25442D2C89140567300C8D204D120FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+22777
Apr 16 07:21:16  telegram=|25442D2C89140567300C8D204E120FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+22793
Apr 16 07:21:32  telegram=|25442D2C89140567300C8D204F200FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+22809
Apr 16 07:21:47  telegram=|25442D2C89140567300C8D2050200FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+22824
Apr 16 07:22:03  telegram=|2E442D2C89140567300C8D2051210FFA2432C278_030E4D8800430E4D88000313B81009426C1F33022D000001FF2100|+22840
Apr 16 07:22:19  telegram=|25442D2C89140567300C8D2052300FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+22856
Apr 16 07:22:35  telegram=|25442D2C89140567300C8D2053310FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+22872
Apr 16 07:23:07  telegram=|25442D2C89140567300C8D2055320FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+22904
Apr 16 07:23:38  telegram=|25442D2C89140567300C8D2057400FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+22935
Apr 16 07:23:50  telegram=|25442D2C89140567300C8D3057400FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+22947
Apr 16 07:23:54  telegram=|25442D2C89140567300C8D2058400FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+22951
Apr 16 07:24:10  telegram=|2E442D2C89140567300C8D2059410FFA2432C278_030E4D8800430E4D88000313B81009426C1F33022D000001FF2100|+22967
Apr 16 07:24:26  telegram=|25442D2C89140567300C8D205A500FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+22983
Apr 16 07:24:42  telegram=|25442D2C89140567300C8D205B510FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+22999
Apr 16 07:25:45  telegram=|25442D2C89140567300C8D205F610FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+23062
Apr 16 07:26:16  telegram=|2E442D2C89140567300C8D2061700FFA2432C278_030E4D8800430E4D88000313B81009426C1F33022D000001FF2100|+23093
Apr 16 07:26:32  telegram=|25442D2C89140567300C8D2062710FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+23109
Apr 16 07:26:48  telegram=|25442D2C89140567300C8D2063720FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+23125
Apr 16 07:27:04  telegram=|25442D2C89140567300C8D2064720FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+23140
Apr 16 07:27:19  telegram=|25442D2C89140567300C8D2065800FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+23156
Apr 16 07:27:35  telegram=|25442D2C89140567300C8D2066800FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+23172
Apr 16 07:27:50  telegram=|25442D2C89140567300C8D2067810FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+23187
Apr 16 07:28:06  telegram=|25442D2C89140567300C8D2068810FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+23203
Apr 16 07:28:22  telegram=|2E442D2C89140567300C8D2069900FFA2432C278_030E4D8800430E4D88000313B81009426C1F33022D000001FF2100|+23219
Apr 16 07:28:38  telegram=|25442D2C89140567300C8D206A910FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+23235
Apr 16 07:28:53  telegram=|25442D2C89140567300C8D206B920FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+23250
Apr 16 07:29:09  telegram=|25442D2C89140567300C8D206C920FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+23266
Apr 16 07:29:25  telegram=|25442D2C89140567300C8D206DA00FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+23282
Apr 16 07:29:40  telegram=|25442D2C89140567300C8D206EA00FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+23297
Apr 16 07:29:56  telegram=|25442D2C89140567300C8D206FA10FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+23313
Apr 16 07:30:14  telegram=|25442D2C89140567300C8D2070A10FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+23331
Apr 16 07:30:27  telegram=|2E442D2C89140567300C8D2071B00FFA2432C278_030E4D8800430E4D88000313B81009426C1F33022D000001FF2100|+23344
Apr 16 07:30:59  telegram=|25442D2C89140567300C8D2073B20FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+23376
Apr 16 07:31:30  telegram=|25442D2C89140567300C8D2075C00FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+8
Apr 16 07:31:45  telegram=|25442D2C89140567300C8D2076C00FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+23
Apr 16 07:32:01  telegram=|25442D2C89140567300C8D2077C10FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+39
Apr 16 07:32:17  telegram=|25442D2C89140567300C8D2078C10FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+55
Apr 16 07:32:32  telegram=|2E442D2C89140567300C8D2079D00FFA2432C278_030E4D8800430E4D88000313B81009426C1F33022D000001FF2100|+70
Apr 16 07:32:48  telegram=|25442D2C89140567300C8D207AD10FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+86
Apr 16 07:33:03  telegram=|25442D2C89140567300C8D207BD20FFA247A1E79C2525B89_4D88004D8800B810091F33000000|+101
[... cut ...]
Apr 16 07:56:14  telegram=|25442D2C89140567300C8D20D35011FA247A1E79C2525B89_4D88004D8800B810091F33000000|+100
Apr 16 07:56:30  telegram=|25442D2C89140567300C8D20D45011FA247A1E79C2525B89_4D88004D8800B810091F33000000|+116
Apr 16 07:56:46  telegram=|25442D2C89140567300C8D20D55111FA247A1E79C2525B89_4D88004D8800B810091F33000000|+132
Apr 16 07:57:36  telegram=|25442D2C89140567300C8D20D86011FA247A1E79C2525B89_4D88004D8800B810091F33000000|+182
Apr 16 07:58:01  telegram=|2E442D2C89140567300C8D20D96111FA2432C278_030E4D8800430E4D88000313B81009426C1F33022D000001FF2100|+207
```