wmbusmeters/wmbusmeters #1243: Metersit G4 support (169 MHz Gas Meter with AMB3665)

BlackShift25, 2024-04-12:
### Type of request

New device support

### Meter description

Metersit Domusnext G4
Mbus 169 Mhz version
https://www.sitgroup.it/en/product/domusnext-g4-g6-2/

### Logged telegrams from meter in decrypted format

```shell
Most likely from different meters but they should be the same model: 

Received telegram from: 04678995
          manufacturer: (MTS) MeteRSit, Italy (0x3693)
                  type: Gas meter (0x03)
                   ver: 0x78
                device: amb8465[00000000]
                  rssi: -46 dBm
                driver: unknown!
telegram=|_464493369589670478037DF50000800101DB0033300000081EC8E1B8D346DF83C605BC2F4619A8EE77D0330320868CCA95765C2B3496B4C95F85C34621A1FBD8214DA5EDD9AA39|+6106
Received telegram from: 06830948
          manufacturer: (MTS) MeteRSit, Italy (0x3693)
                  type: Gas meter (0x03)
                   ver: 0x78
                device: amb8465[00000000]
                  rssi: -92 dBm
                driver: unknown!
telegram=|_464493364809830678037D3C0000800101DB0033300000029DDEA9011E735B356571F2B673F175E5ABCFCE1A52EBFF407E13AB455420B3AD552C99F461BB6DEE8206956C5963DC|+13536
Received telegram from: 04638262
          manufacturer: (MTS) MeteRSit, Italy (0x3693)
                  type: Gas meter (0x03)
                   ver: 0x78
                device: amb8465[00000000]
                  rssi: -65 dBm
                driver: unknown!
```

xjols, 2024-04-18:
Hello! Which hardware are you using and software to read 169MHz telegrams and decode this ? I'm also trying to decode water meter in this frequency without any output telegram for now...

Thank you

BlackShift25, 2024-04-21:
Hi, I'm using the AMB3665 with a docker container running wmbusmeters.

BlackShift25, 2024-04-30:
Anyone willing to help with this ?

closed 2024-10-23 by github-actions[bot]