wmbusmeters/wmbusmeters #1249: Device Support for "(IMT) INTEGRA METERING, Switzerland (0x25b4) type: Bus/System component (0x0e) ver: 0x05"

tbandixen, 2024-04-19:
### Type of request

New device support

### Meter description

(IMT) INTEGRA METERING, Switzerland (0x25b4) type: Bus/System component (0x0e) ver: 0x05
Integra aquastream® Radio (AQS-W8)
Protocoll description can be found in the attachment
[aquastream 9-730-PT-EN-06.pdf](https://github.com/wmbusmeters/wmbusmeters/files/15037922/aquastream.9-730-PT-EN-06.pdf)

### Logged telegrams from meter in decrypted format

```shell
https://wmbusmeters.org/analyze/4644B42538680610050E7267397405B4250107BE0030252F2F041309CB140084101300000000046D382913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F

telegram=|4644B42538680610050E7267397405B4250107980030252F2F_041308CB140084101300000000046D2E2913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107990030252F2F_041308CB140084101300000000046D2E2913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B42501079A0030252F2F_041308CB140084101300000000046D2E2913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B42501079B0030252F2F_041309CB140084101300000000046D2F2913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B42501079C0030252F2F_041309CB140084101300000000046D2F2913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B42501079D0030252F2F_041309CB140084101300000000046D2F2913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B42501079E0030252F2F_041309CB140084101300000000046D2F2913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B42501079F0030252F2F_041309CB140084101300000000046D302913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107A00030252F2F_041309CB140084101300000000046D302913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107A10030252F2F_041309CB140084101300000000046D302913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107A20030252F2F_041309CB140084101300000000046D312913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107A30030252F2F_041309CB140084101300000000046D312913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107A40030252F2F_041309CB140084101300000000046D312913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107A50030252F2F_041309CB140084101300000000046D312913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107A60030252F2F_041309CB140084101300000000046D322913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107A70030252F2F_041309CB140084101300000000046D322913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107A80030252F2F_041309CB140084101300000000046D322913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107A90030252F2F_041309CB140084101300000000046D322913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107AA0030252F2F_041309CB140084101300000000046D332913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107AB0030252F2F_041309CB140084101300000000046D332913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107AC0030252F2F_041309CB140084101300000000046D332913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107AD0030252F2F_041309CB140084101300000000046D332913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107AE0030252F2F_041309CB140084101300000000046D342913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107AF0030252F2F_041309CB140084101300000000046D342913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107B00030252F2F_041309CB140084101300000000046D342913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107B10030252F2F_041309CB140084101300000000046D342913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107B20030252F2F_041309CB140084101300000000046D352913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107B30030252F2F_041309CB140084101300000000046D352913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107B40030252F2F_041309CB140084101300000000046D352913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107B50030252F2F_041309CB140084101300000000046D362913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107B60030252F2F_041309CB140084101300000000046D362913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107B70030252F2F_041309CB140084101300000000046D362913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107B80030252F2F_041309CB140084101300000000046D362913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107B90030252F2F_041309CB140084101300000000046D372913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107BA0030252F2F_041309CB140084101300000000046D372913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107BB0030252F2F_041309CB140084101300000000046D372913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107BC0030252F2F_041309CB140084101300000000046D372913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107BD0030252F2F_041309CB140084101300000000046D382913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
telegram=|4644B42538680610050E7267397405B4250107BE0030252F2F_041309CB140084101300000000046D382913344413397B1400426C1F3302FD17000002FD74B9162F2F2F2F2F2F2F|+0
```

tbandixen, 2024-04-19:
It seams to work with the "hydrus" driver, but I don't know if its the right one.
In the logs there is a message saying:
`(meter) water: meter detection did not match the selected driver hydrus! correct driver is: unknown!
(meter) Not printing this warning again for id: 10066838 mfct: (IMT) INTEGRA METERING, Switzerland (0x25b4) type: Bus/System component (0x0e) ver: 0x05
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination`

tbandixen, 2024-04-19:
Maybe this PR could handle this?
https://github.com/wmbusmeters/wmbusmeters/pull/1106

tbandixen, 2024-04-19:
Thanks for your support

tbandixen, 2024-06-19:
not stale?

closed 2024-06-20 by tbandixen

nie3mals, 2024-08-05:
@tbandixen ,
thanks for the issue for adding device support. I have the same MBus Watermeter from INTEGRA. All works fine, however I need to decrypt the telegrams. Can you let me know how to do this step? Any help would be appriciated!

tbandixen, 2024-08-06:
@nie3mals there are two scenarios:

1. You own the meter
Then you can set the key for encryption by yourself (with the "Param App") so you know it (my scenario).
3. You don't own the meter
You have to ask the "Wasserwerke", if they tell you the key for your device. Maybe you can reverse engineer it? I don't know...

michaelfrossard, 2026-08-31:
Just got this exact meter (IMT, Bus/System component 0x0e, ver 0x05 — Integra aquastream® AQS-W8) working end-to-end with wmbusmeters 3.0.0-RC1, so posting the fix in case it saves someone else the debugging time.

**The `aquastream` driver already supports this meter — the trick is which ID to use.**

The AQS-W8 sends a telegram where the *outer* DLL address (CI=0x72, long header) identifies the radio module itself (type 0x0e, "Bus/System component"), but it also carries a *second*, nested TPL address right after — and that inner address is the one with type 0x07 ("Water meter") that the `aquastream` driver's `detect { mvt = IMT,01,07 }` is looking for.

Example telegram:
```
4644B42530990710050E7270260006B4250107C50030252F2F_04132481020084101300000000046D2E295F384413DD460200426C5F3702FD17000002FD74D3142F2F2F2F2F2F2F
```

Breaking down the header:
- Outer DLL address: `10079930` — mfct IMT, version `0x05`, type `0x0e` (this is the serial printed on the module itself)
- CI = `0x72` (long TPL header) → nested secondary address follows
- Inner TPL address: id `06002670` (BCD), mfct IMT, version `0x01`, type `0x07` ← **this is the ID that needs to go in the meter config**, not the module serial.

Once configured with:
```
driver=aquastream
id=06002670
key=<AES key from the device>
```
...it detects and decodes cleanly — total_m3, target_m3/target_date, status, battery_y all come through correctly.

I suspect this is exactly the pattern in the driver's own built-in test telegram (`drivers/src/aquastream.xmq`), which also has an outer 0x0e / inner 0x07 structure — so this isn't really a "missing device support" case, just a documentation/discoverability gap. Might be worth a note in the driver's `info` field or the wiki pointing out that the *module* serial ≠ the *meter* id for this device.

Separately: the HA add-on (`wmbusmeters-ha-addon`) has no `mqtt_discovery/aquastream.json`, so even once decoding works, no HA entities get auto-created via MQTT discovery. I put together a working one (fields: total_m3, target_m3, target_date, total_tariff1_m3, battery_y, meter_datetime, timestamp, status, rssi_dbm) — happy to open a PR against `wmbusmeters/wmbusmeters-ha-addon` with it if that's useful.

michaelfrossard, 2026-08-31:
see https://github.com/wmbusmeters/wmbusmeters-ha-addon/pull/85 for aquastream.json