wmbusmeters/wmbusmeters #1251: not detecting right driver

dragonat0r, 2024-04-19:
### Type of request

New device support

### Meter description

I receive some records in log for not finding right driver:

(meter) Баня Водомер Топла Вода: meter detection did not match the selected driver qwater! correct driver is: unknown!
(meter) Not printing this warning again for id: 51961316 mfct: (QDS) Qundis, Germany (0x4493) type: Warm Water (30°C-90°C) meter (0x06) ver: 0x1a
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) Тоалетна Водомер Топла Вода: meter detection did not match the selected driver qwater! correct driver is: unknown!
(meter) Not printing this warning again for id: 51961314 mfct: (QDS) Qundis, Germany (0x4493) type: Warm Water (30°C-90°C) meter (0x06) ver: 0x1a
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination 

### Logged telegrams from meter in decrypted format

```shell
telegram=|49449344161396511A0678_0DFF5F350082E400005E0107C113FFFF77850100FF2C340000001F33615301000080008000800080008000800300000000002000C1021C032F046D02111334|+90

telegram=|49449344141396511A0678_0DFF5F350082D200000F0007C113FFFF63460000FF2C340000001F33484000000080008000800080008000800300000000000700C900C1002F046D02111334|+102
```

BIBOLV, 2024-04-19:
Check that returned meter reading value is correct and just use qwater to get supported readings [decoded](https://wmbusmeters.org/analyze/49449344161396511A06780DFF5F350082E400005E0107C113FFFF77850100FF2C340000001F33615301000080008000800080008000800300000000002000C1021C032F046D02111334).

dragonat0r, 2024-04-20:
Thanks. I'm already using qwater as a driver and meter reading looks correct. Just opened this topic for adding my devices as supported.
Can provide more info if it is needed.

closed 2024-07-20 by github-actions[bot]