wmbusmeters/wmbusmeters #1252: Support of RTL-SDR V4 Dongle and its drivers

Milvus11, 2024-04-19:
### Type of request

Enhancement

### OS version

Raspberry Pi OS 64 bit legacy lite

### wmbusmeters version

wmbusmeters: 1.16.1-27-gd90dbb1 d90dbb196e28122e42f0f3394a697570dc54b42d

### Your message goes here

Hey, 
first of all, sorry if the quesiton is silly, I am a newbie. I bought an RTL-SDR V4, installed the wmbusmeters and tried for weeks to receive any message from my Techems. Today I think I found the issue: The rtl-sdr driver does not work for the RTL-SDR V4 as I found out on the support page:

https://www.rtl-sdr.com/v4/

However, today I tried to follow the steps mentioned by that page to install the correct driver. Problem is, when trying to run wmbusmeters, that when inserting

"tail -f /var/log/wmbusmeters/wmbusmeters.log" 

i get: 

[2024-04-19_19:58:32] (rtlwmbus) error: when starting as daemon, wmbusmeters looked for /rtl_sdr and /usr/bin/rtl_sdr, but found neither!

The new driver files are installed in another folder, how can I manage to tell wmbusmeters?

To clarify, what I did, here are the commands I used to install everything:

 1  sudo apt update
    2  sudo apt upgrade -y
    3  sudo apt-get install libusb-1.0-0-dev git cmake pkg-config
    4  git clone https://github.com/rtlsdrblog/rtl-sdr-blog
    5  cd rtl-sdr-blog
    6  mkdir build
    7  cd build
    8  cmake ../ -DINSTALL_UDEV_RULES=ON
    9  make
   10  sudo make install
   11  sudo cp ../rtl-sdr.rules /etc/udev/rules.d/
   12  sudo ldconfig
   13  cd
   14  git clone https://github.com/weetmuts/wmbusmeters.git
   15  cd wmbusmeters
   16  make
   17  cd
   18  ls
   19  sudo rm -r wmbusmeters
   20  git clone https://github.com/wmbusmeters/wmbusmeters.git
   21  cd wmbusmeters
   22  make
   23  cd
   24  sudo apt install -y libxml2-dev
   25  git clone https://github.com/wmbusmeters/wmbusmeters.git
   26  cd wmbusmeters
   27  make
   28  cd
   29  ls
   30  sudo rm -r wmbusmeters
   31  git clone https://github.com/wmbusmeters/wmbusmeters.git
   32  cd wmbusmeters
   33  make
   34  sudo make install
   35  cd
   36  sudo nano /etc/wmbusmeters.conf
   37  sudo systemctl daemon-reload
   38  sudo systemctl enable wmbusmeters.service
   39  sudo systemctl start wmbusmeters.service
   40  sudo systemctl status wmbusmeters.service
   41  history
   42  sudo systemctl status wmbusmeters.service
   43  tail -f /var/log/wmbusmeters/wmbusmeters.log
   44  git clone https://github.com/xaelsouth/rtl-wmbus.git
   45  cd rtl-wmbus
   46  make
   47  sudo make install
   48  sudo reboot
   49  tail -f /var/log/wmbusmeters/wmbusmeters.log

if you could help me out, I'd really appreciate it

BIBOLV, 2024-04-19:
Hi, 
Firstly test if your installation detects dongle `/usr/local/bin/rtl_test -t`, if it works then you can use `device=rtlwmbus:CMD(/usr/local/bin/rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a -f)` in configuration and everything should work.
There are other approaches also to fix your issue - wmbusmeters is looking for rtl_sdr either in $PATH or in /usr/bin/, so you can add /usr/local/bin/ to $PATH or add symlink to /usr/bin/ or copy or move to /usr/bin/ or recompile and set proper installation directory - `cmake ../ -DINSTALL_UDEV_RULES=ON -DCMAKE_INSTALL_PREFIX=/usr`.

Milvus11, 2024-04-19:
Thank you for your response! It worked! I tried kind of everything you suggested, but now I get files coming in!! To make the path clear also to maybe other people struggling, I tell, what I did:

Following your stepts I got the following output:

```
/usr/local/bin/rtl_test -t
Found 1 device(s):
  0:  RTLSDRBlog, Blog V4, SN: 00000001

Using device 0: Generic RTL2832U OEM

Kernel driver is active, or device is claimed by second instance of librtlsdr.
In the first case, please either detach or blacklist the kernel module
(dvb_usb_rtl28xxu), or enable automatic detaching at compile time.

usb_claim_interface error -6
Failed to open rtlsdr device #0.
```
so I did:

1. blacklisting: 
`sudo nano /etc/modprobe.d/no-rtl.conf`

entering: `blacklist dvb_usb_rtl28xxu`

then unload the driver

`lsmod | grep dvb_usb_rtl28xxu`

```
 lsmod | grep rtl
lsmod | grep dvb
rtl2832_sdr            24576  0
rtl2832                20480  1
i2c_mux                16384  1 rtl2832
regmap_i2c             16384  1 rtl2832
dvb_usb_rtl28xxu       32768  1
dvb_usb_v2             28672  1 dvb_usb_rtl28xxu
dvb_core              126976  2 dvb_usb_v2,rtl2832
videobuf2_vmalloc      16384  2 rtl2832_sdr,bcm2835_v4l2
videobuf2_v4l2         32768  5 rtl2832_sdr,bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem,bcm2835_isp
videobuf2_common       69632  9 rtl2832_sdr,bcm2835_codec,videobuf2_vmalloc,videobuf2_dma_contig,videobuf2_v4l2,bcm2835_v4l2,v4l2_mem2mem,videobuf2_memops,bcm2835_isp
videodev              274432  7 rtl2832_sdr,bcm2835_codec,videobuf2_v4l2,bcm2835_v4l2,videobuf2_common,v4l2_mem2mem,bcm2835_isp
dvb_usb_rtl28xxu       32768  1
dvb_usb_v2             28672  1 dvb_usb_rtl28xxu
dvb_core              126976  2 dvb_usb_v2,rtl2832
mc                     61440  8 videodev,bcm2835_codec,dvb_usb_v2,videobuf2_v4l2,dvb_core,videobuf2_common,v4l2_mem2mem,bcm2835_isp

```

```
sudo rmmod dvb_usb_rtl28xxu
sudo rmmod dvb_usb_v2
sudo rmmod rtl2832
sudo rmmod rtl2832_sdr
sudo rmmod rtl2832

```

`/usr/local/bin/rtl_test -t`

And then tried do configure what you mentioned:

`sudo nano wmbusmeters.conf`

`device=rtlwmbus:CMD(/usr/local/bin/rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a -f)`

`export PATH=$PATH:/usr/local/bin`

`echo 'export PATH=$PATH:/usr/local/bin' >> ~/.bashrc`

```
 sudo ln -s /usr/local/bin/rtl_sdr /usr/bin/rtl_sdr
  sudo systemctl restart wmbusmeters.service
 sudo systemctl status wmbusmeters.service
```

And it worked! I am pretty sure I did a lot wrong now, but I receive telegrams every minute and quite a lot, so somehow it works now! 

I am so grateful and reliefed, thank you so much. I guess I got to make an image of my raspi sd now, cause as I don't have a full reliable manual, I would never get it running again..

thanks again!!

closed 2024-04-19 by Milvus11