wmbusmeters/wmbusmeters #1260: Double CUL-868-433 USB stick not recognized

sectune, 2024-04-25:
### Type of request

New device support

### Meter description

Hello,
I am very sorry for the beginner question but I could get the information from the wiki and I also tried reading the open/closed issues with no avail.
I have bought this device:
https://smart-home-komponente.de/en/products/doppel-cul-868-433-usb-stick-fur-fhem-1x-868mhz-1x-433mhz
and trying to run it with both 868 as well as the second cul with 433 in the wmbusmeters add-on in Home Assistant.
In the terminal I see that the device is recognised:
```
rwxrwxrwx    1 root     root            13 Apr 25 00:28 usb-STM32_MapleCUL_3a5d9184-if00 -> ../../ttyACM1
lrwxrwxrwx    1 root     root            13 Apr 25 00:28 usb-STM32_MapleCUL_3a5d9184-if02 -> ../../ttyACM2
lrwxrwxrwx    1 root     root            13 Apr 25 00:28 usb-STM32_MapleCUL_3a5d9184-if04 -> ../../ttyACM3
```
However, I do get the following output when trying to run with `device=auto:t1` (nothing changed, reset config multiple times) :
```
No wmbus device detected, waiting for a device to be plugged in.
No meters configured. Printing id:s of all telegrams heard!
```

Can you maybe be so kind and give me a hint on how to proceed and what I am doing wrong? I am very grateful for any advice!

### Logged telegrams from meter in decrypted format

```shell
no telegrams logged
```

sectune, 2024-06-24:
Still kindly looking for support

weetmuts, 2024-06-24:
You have to do:`wmbusmeters --debug auto:t1`
and post the log here.

closed 2024-09-23 by github-actions[bot]