wmbusmeters/wmbusmeters #1264: Qundis module 5.5 water - decode frame

tantosik, 2024-05-05:
### Type of request

Enhancement

### OS version

ESP32 

### wmbusmeters version

1.16.1-23-gefcc41d

### Your message goes here

Hello, I've got the frame:
`3C449344886799363537728867993693443507C3000020ABD3B3AA2BEDBE1E80EC8068D339A54525EC8A9BE2281163E1FB93D62968F7E7046D1C0F1E34046D1C0F1E34`

There is the frame from Qundis Radio Adapter (Q module 5.5 water) and it do not seem to be a valid mbus frame (because 0x3C length). Total length of this frame should be 60 bytes but my frame has 67 bytes of length. But the last 6 bytes are repeated twice: `046D1C0F1E34`. Without this repetition the whole frame is valid in order the length. 
So overall it should look like this (valid frame): `3C449344886799363537728867993693443507C3000020ABD3B3AA2BEDBE1E80EC8068D339A54525EC8A9BE2281163E1FB93D62968F7E7046D1C0F1E34`

Any idea how to decode this frame?
Wmbus analyze: [https://wmbusmeters.org/analyze/3C449344886799363537728867993693443507C3000020ABD3B3AA2BEDBE1E80EC8068D339A54525EC8A9BE2281163E1FB93D62968F7E7046D1C0F1E34:qwater](url)

weetmuts, 2024-05-05:
There is more wrong with this frame than just the last bytes! :-) Check the other content, it does not make sense. It looks like the underlying data is mostly random. This could be the case if it is encrypted. What is the hardware setup? Has it been passed through another system?

tantosik, 2024-05-06:
I was wondering what is wrong with this frame and the major issue is: I was trying to encrypted the non crypted frame :). And the final was as above. Thanks for help :).