wmbusmeters/wmbusmeters #1267: Heitland Qalcosonic W1 YEAR:2024

extern-joerg, 2024-05-08:
### Type of request

Bug

### OS version

raspi p4 docker

### wmbusmeters version

(wmbusmeters) version: 1.16.1-29-g1a4169e

### Your message goes here

First of all:
Please help me

Problem:
Heitland Qalcosonic W1 YEAR:2024
Does not provide decoded data.
The telegram is also not displayed in the list of telegrams collected every x seconds.
In debug mode without key I find the counter with this information and with a telegram
(meter) cold water meter-07022303-q400 07022303.M=AXI.V=10.T=07 "434409070323020710078C20A4900F002C252C1C0000237887F628D82AFC7AA41020071004A6E981CDD9BA11B4AF6F1441D3EE2E41C49FA491024E55D0EBF3681BB1019F"

Is:
So far I have used a water meter from Kampstrup multitical21. The Kamstrup and the decoding and mosquitto work perfectly.
I use a Raspi with Docker with this version, see the excerpt from the log file in debug mode below.

Behave:
I use 3 files in wmbusmeter.d. In debug mode without file 2#3 (see below), wmbusmeter.log gives me the information that it finds the homeland. It says, among other things, M=AXI and driver=q400. As a result, I created file 2#3 with key=0101… then the meter appears with this (config) template cold water meter-07022303-q400 07022303 q400
  NO longer in the wmbusmeter.log file.
If I create file 2#3 without a key and start debug mode, the following wmbusmeter.log file appears.

Header of the wmbusmeter.log in debug mode
(wmbusmeters) version: 1.16.1-29-g1a4169e
(config) number of meters: 3
(config) template cold water meter-25575311-multical21 25575311 multical21
(config) template cold water meter-07022303-q400 07022303 q400
(config) template unknown-iot-device * auto
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers

This is my wmbusmeters.conf (prod)
=================================
loglevel=oneshot
device=auto
donotprobe=/dev/ttyAMA0
listento=c1,t1,s1
logtelegrams=true
format=json
meterfilesaction=overwrite
shell=/usr/bin/mosquitto_pub -h mosquitto -p 1883 -u pi -P 123456 -t wmbusmeters/$METER_ID -m "$METER_JSON"

3 files in wmbusmeter.d see the following

1#3 wmbusmeter.d/kamstrup (works perfectly)
=============================================
name=cold water meter-25575311-multical21
id=25575311
key=010101010101010101010101010101010
driver=auto
type=multical21

2#3 wmbusmeter.d/heitland (does not work)
======================================
name=cold water meter-07022303-q400
id=07022303
key=DD6087D76FEC9624DC2DCDFA678EOE8F
driver=q400
type=q400

3#3wmbusmeter.d/unknown (works fine)
============================================
name=unknown-iot-device
id=*
driver=auto

[wmbusmeters.log](https://github.com/wmbusmeters/wmbusmeters/files/15248481/wmbusmeters.log)

weetmuts, 2024-05-08:
Heh, replace the O (oooh) with a 0 (zero) in your aes key. :-)

weetmuts, 2024-05-08:
https://wmbusmeters.org/analyze/434409070323020710078c20a4900f002c252c1c0000237887f628d82afc7aa41020071004a6e981cdd9Ba11B4af6f1441d3ee2e41c49fa491024e55d0eBf3681BB1019f:auto:DD6087D76FEC9624DC2DCDFA678E0E8F

closed 2024-08-07 by github-actions[bot]