wmbusmeters/wmbusmeters #1270: Generic Driver for all Devices

bkggkt3d, 2024-05-13:
### Type of request

Enhancement

### OS version

Debian Trixie

### wmbusmeters version

master

### Your message goes here

I'm currently implementing wmbusmeters in multiple buildings where a lot of (10....30 each) older M-Bus devices are already installed. So far testet, the most devices do not have a driver available in wmbusmeters.
When testing with libmbus I see the important data from inside the telegram.
So, it would be great to have a kind of "generic" driver in wmbus which simply parse the telegram and show all data inside as JSON/Fields/HR without any device specific filtering. This could be a good fallback driver to be used if no other driver works.

weetmuts, 2024-05-13:
You can also write your own driver as a text file and just use that. Check this one out:
https://github.com/wmbusmeters/wmbusmeters-drivers/blob/main/wmbusmeters.drivers.d/kampress.xmq

Change the name to your new driver name and change the file name to the same.
Then just put the xmq file into /etc/wmbusmeters.drivers.d and you use the driver directly.

A generic driver is pretty useless except for the most simple meters. What you want to add is the knowledge
of what the bits mean and what volume in storage 1 means, last day, last month last year etc.

weetmuts, 2024-05-13:
What kind of meters are the ones you are reading? Water, electricity, other?

bkggkt3d, 2024-05-15:
Sorry for asking so much stupid questions, I'm really new within M-Bus technology and miss a lot of basics. Meanwhile I've multiple projects with a lot of older wired M-Bus devices of all kind (Electrical, Heat, Gas, WaterMeters) already installed. The aim is to read them out. So far I've testet 2 devices and both don't have an driver available (GossenMetrawatt U1289 and EMU 1/40 both ElectricalMeter) 
The wmbusmeters-drivers project with the .xmq text-files is probably covering what I need, I will check this soon - thanks.

closed 2024-08-15 by github-actions[bot]