wmbusmeters/wmbusmeters #1277: Maddalena Microclima - add device to

nightwissh, 2024-05-18:
### Type of request

New device support

### Meter description

Hi,
I am getting a "Device unsupported" message for the Maddalena Microclima Evo device.
Despite the message the telegram is handled well and I am receiving all the information so the device is kind of supported already but maybe not for certain mfct,media,version combination present in my case.
Here is the information from the log:

```
(meter) MadelenaMicroClima: meter detection did not match the selected driver microclima! correct driver is: unknown!
(meter) Not printing this warning again for id: 33758504 mfct: (MAD) Maddalena, Italy (0x3424) type: Heat volume at inlet meter (0x0c) ver: 0x00
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(wmbusmeters) started meter 3 (MadelenaMicroClima 33758504 microclima) identity mode: id R33758504
```

This error is not visible in [wmbusmeters](https://wmbusmeters.org/analyze/a444243404857533000c7a410000202f2f046d0f2f1934040e7d0f0000041366e6010001fd1700426cff2c440e31030000c4010e5a0e0000c4020ea60c0000c4030e070a0000c4040e31030000c4050e00000000c4060e00000000c4070e00000000c4080e00000000c4090e00000000c40a0e00000000c40B0e00000000c40c0e00000000c40d0e00000000c40e0e00000000c40f0e0000000003fd0c05010002fd0B0111:microclima)
but I see a version of wmbusmeters used on that page is higher (1.16) than the current version in GIT (1.15)
Apologize for the month-old telegrams but it took me a bit to find time to submit this issue. 

Please let me know if you need any further information.

### Logged telegrams from meter in decrypted format

```shell
telegram=|A444243404857533000C7A94000020_2F2F046D3A311934040E7E0F0000041374E6010001FD1700426CFF2C440E31030000C4010E5A0E0000C4020EA60C0000C4030E070A0000C4040E31030000C4050E00000000C4060E00000000C4070E00000000C4080E00000000C4090E00000000C40A0E00000000C40B0E00000000C40C0E00000000C40D0E00000000C40E0E00000000C40F0E0000000003FD0C05010002FD0B0111|+10741
telegram=|A444243404857533000C7A97000020_2F2F046D39261A34040EA20F0000041331EB010001FD1700426CFF2C440E31030000C4010E5A0E0000C4020EA60C0000C4030E070A0000C4040E31030000C4050E00000000C4060E00000000C4070E00000000C4080E00000000C4090E00000000C40A0E00000000C40B0E00000000C40C0E00000000C40D0E00000000C40E0E00000000C40F0E0000000003FD0C05010002FD0B0111|+57481
telegram=|A444243404857533000C7A98000020_2F2F046D3B261A34040EA20F0000041337EB010001FD1700426CFF2C440E31030000C4010E5A0E0000C4020EA60C0000C4030E070A0000C4040E31030000C4050E00000000C4060E00000000C4070E00000000C4080E00000000C4090E00000000C40A0E00000000C40B0E00000000C40C0E00000000C40D0E00000000C40E0E00000000C40F0E0000000003FD0C05010002FD0B0111|+57599
telegram=|A444243404857533000C7A99000020_2F2F046D01271A34040EA30F000004133EEB010001FD1700426CFF2C440E31030000C4010E5A0E0000C4020EA60C0000C4030E070A0000C4040E31030000C4050E00000000C4060E00000000C4070E00000000C4080E00000000C4090E00000000C40A0E00000000C40B0E00000000C40C0E00000000C40D0E00000000C40E0E00000000C40F0E0000000003FD0C05010002FD0B0111|+57716
telegram=|A444243404857533000C7A9A000020_2F2F046D03271A34040EA30F0000041345EB010001FD1700426CFF2C440E31030000C4010E5A0E0000C4020EA60C0000C4030E070A0000C4040E31030000C4050E00000000C4060E00000000C4070E00000000C4080E00000000C4090E00000000C40A0E00000000C40B0E00000000C40C0E00000000C40D0E00000000C40E0E00000000C40F0E0000000003FD0C05010002FD0B0111|+57834
telegram=|A444243404857533000C7A9B000020_2F2F046D05271A34040EA30F000004134CEB010001FD1700426CFF2C440E31030000C4010E5A0E0000C4020EA60C0000C4030E070A0000C4040E31030000C4050E00000000C4060E00000000C4070E00000000C4080E00000000C4090E00000000C40A0E00000000C40B0E00000000C40C0E00000000C40D0E00000000C40E0E00000000C40F0E0000000003FD0C05010002FD0B0111|+57952
telegram=|A444243404857533000C7A9C000020_2F2F046D07271A34040EA30F0000041353EB010001FD1700426CFF2C440E31030000C4010E5A0E0000C4020EA60C0000C4030E070A0000C4040E31030000C4050E00000000C4060E00000000C4070E00000000C4080E00000000C4090E00000000C40A0E00000000C40B0E00000000C40C0E00000000C40D0E00000000C40E0E00000000C40F0E0000000003FD0C05010002FD0B0111|+58070
telegram=|A444243404857533000C7A9D000020_2F2F046D09271A34040EA30F000004135AEB010001FD1700426CFF2C440E31030000C4010E5A0E0000C4020EA60C0000C4030E070A0000C4040E31030000C4050E00000000C4060E00000000C4070E00000000C4080E00000000C4090E00000000C40A0E00000000C40B0E00000000C40C0E00000000C40D0E00000000C40E0E00000000C40F0E0000000003FD0C05010002FD0B0111|+58187
telegram=|A444243404857533000C7AA0000020_2F2F046D0F271A34040EA40F000004136FEB010001FD1700426CFF2C440E31030000C4010E5A0E0000C4020EA60C0000C4030E070A0000C4040E31030000C4050E00000000C4060E00000000C4070E00000000C4080E00000000C4090E00000000C40A0E00000000C40B0E00000000C40C0E00000000C40D0E00000000C40E0E00000000C40F0E0000000003FD0C05010002FD0B0111|+58541
telegram=|A444243404857533000C7AA1000020_2F2F046D11271A34040EA40F0000041376EB010001FD1700426CFF2C440E31030000C4010E5A0E0000C4020EA60C0000C4030E070A0000C4040E31030000C4050E00000000C4060E00000000C4070E00000000C4080E00000000C4090E00000000C40A0E00000000C40B0E00000000C40C0E00000000C40D0E00000000C40E0E00000000C40F0E0000000003FD0C05010002FD0B0111|+58660
telegram=|A444243404857533000C7AA2000020_2F2F046D13271A34040EA40F000004137DEB010001FD1700426CFF2C440E31030000C4010E5A0E0000C4020EA60C0000C4030E070A0000C4040E31030000C4050E00000000C4060E00000000C4070E00000000C4080E00000000C4090E00000000C40A0E00000000C40B0E00000000C40C0E00000000C40D0E00000000C40E0E00000000C40F0E0000000003FD0C05010002FD0B0111|+58778
telegram=|A444243404857533000C7AA4000020_2F2F046D17271A34040EA40F000004138AEB010001FD1700426CFF2C440E31030000C4010E5A0E0000C4020EA60C0000C4030E070A0000C4040E31030000C4050E00000000C4060E00000000C4070E00000000C4080E00000000C4090E00000000C40A0E00000000C40B0E00000000C40C0E00000000C40D0E00000000C40E0E00000000C40F0E0000000003FD0C05010002FD0B0111|+59014
telegram=|A444243404857533000C7AB1000020_2F2F046D30271A34040EA60F00000413E5EB010001FD1700426CFF2C440E31030000C4010E5A0E0000C4020EA60C0000C4030E070A0000C4040E31030000C4050E00000000C4060E00000000C4070E00000000C4080E00000000C4090E00000000C40A0E00000000C40B0E00000000C40C0E00000000C40D0E00000000C40E0E00000000C40F0E0000000003FD0C05010002FD0B0111|+60558
telegram=|A444243404857533000C7AD4000020_2F2F046D3B281A34040EAB0F00000413D6EC010001FD1700426CFF2C440E31030000C4010E5A0E0000C4020EA60C0000C4030E070A0000C4040E31030000C4050E00000000C4060E00000000C4070E00000000C4080E00000000C4090E00000000C40A0E00000000C40B0E00000000C40C0E00000000C40D0E00000000C40E0E00000000C40F0E0000000003FD0C05010002FD0B0111|+64778
telegram=|A444243404857533000C7ADC000020_2F2F046D0F291A34040EAC0F00000413EAEC010001FD1700426CFF2C440E31030000C4010E5A0E0000C4020EA60C0000C4030E070A0000C4040E31030000C4050E00000000C4060E00000000C4070E00000000C4080E00000000C4090E00000000C40A0E00000000C40B0E00000000C40C0E00000000C40D0E00000000C40E0E00000000C40F0E0000000003FD0C05010002FD0B0111|+65733
telegram=|A444243404857533000C7A75000020_2F2F046D172E1A34040EAC0F00000413EAEC010001FD1700426CFF2C440E31030000C4010E5A0E0000C4020EA60C0000C4030E070A0000C4040E31030000C4050E00000000C4060E00000000C4070E00000000C4080E00000000C4090E00000000C40A0E00000000C40B0E00000000C40C0E00000000C40D0E00000000C40E0E00000000C40F0E0000000003FD0C05010002FD0B0111|+84230
telegram=|A444243404857533000C7AB9000020_2F2F046D24301A34040EAC0F00000413EAEC010001FD1700426CFF2C440E31030000C4010E5A0E0000C4020EA60C0000C4030E070A0000C4040E31030000C4050E00000000C4060E00000000C4070E00000000C4080E00000000C4090E00000000C40A0E00000000C40B0E00000000C40C0E00000000C40D0E00000000C40E0E00000000C40F0E0000000003FD0C05010002FD0B0111|+92232
telegram=|A444243404857533000C7ABA000020_2F2F046D26301A34040EAC0F00000413EAEC010001FD1700426CFF2C440E31030000C4010E5A0E0000C4020EA60C0000C4030E070A0000C4040E31030000C4050E00000000C4060E00000000C4070E00000000C4080E00000000C4090E00000000C40A0E00000000C40B0E00000000C40C0E00000000C40D0E00000000C40E0E00000000C40F0E0000000003FD0C05010002FD0B0111|+92352
```

Hama4tux, 2024-08-05:
Any update about microclima driver ?

closed 2024-11-04 by github-actions[bot]

nikito7, 2024-11-10:
telegram=|A444243404857533000C7ABA000020_2F2F046D26301A34040EAC0F00000413EAEC010001FD1700426CFF2C440E31030000C4010E5A0E0000C4020

A4 - size
44 - T1
2434 - MAD
0485 - version?
753300 - ID ? (003375)
0C - ?
7A - block 2
BA - counter
000020_2F2F046D26301A34040EAC0F00000413EAEC010001FD1700426CFF2C440E31030000C4010E5A0E0000C4020