wmbusmeters/wmbusmeters #1278: Failed to start wmbusmerts.service

Marcus54542, 2024-05-19:
### Type of request

Question

### OS version

Debian GNU/Linux 12 (bookworm)

### wmbusmeters version

the last

### Your message goes here

Hello
The system is installed on a Raspberry Pi 3

I can read the value of my meters.
On the other hand, when launching the service I have this error
`sudo systemctl start wmbusmerts

Failed to start wmbusmerts.service: Unit wmbusmerts.service not found.

root@compteur:/# sudo systemctl status wmbusmeters

○ wmbusmeters.service - "wmbusmeters service"
     Loaded: loaded (/lib/systemd/system/wmbusmeters.service; disabled; preset: enabled)
     Active: inactive (dead)
       Docs: https://github.com/wmbusmeters/wmbusmeters
             man:wmbusmeters(1)
`

Did I forget something

Thank you

closed 2024-05-19 by Marcus54542

volkerdeHaas, 2024-06-07:
same problem: Service is loaded on reboot but not activated. After manually starting the service things are running smoothly up to next reboot.

Marcus54542, 2024-06-07:
try 
sudo systemctl enable wmbusmeters

volkerdeHaas, 2024-06-07:
> try sudo systemctl enable wmbusmeters

You are my hero!!! So easy... I need to learn much more about Debian basics...
Thnx!

pswiatki, 2024-06-07:
This belongs to  `systemd` fundamentals  :)