wmbusmeters/wmbusmeters #1280: use rtl_433 as the MBUS demodulator

pswiatki, 2024-05-19:
### Type of request

Question

### OS version

n/a

### wmbusmeters version

n/a

### Your message goes here

I know it is possible to replace `rtl_wmbus` with `rtl_433` to demodulate and form `mbus` telegrams. However, I also know `rtl_433` can be used to decode other protocols simultaneously. If one wanted to use that for things unrelated to `mbus` how would one go about making it happen? What delivery mechanism could one use to selectively send `mbus` datagrams to `wmbusmeters` while other decoded frames to a different destination for further processing?
I am almost sure this has been done, but cannot find it anywhere.

weetmuts, 2024-05-19:
Yes, you can do it like this:

rtl_433 -F csv -f 868.95M | tee >(wmbusmeters stdin:rtl433) 

Now the rtl_433 output is sent to wmbusmeters and continue on the pipe to some other tools where you might
want to detect other types of telegrams.

Which other telegrams on 868.95 do you want to listen to?

weetmuts, 2024-05-19:
rtl_433 -F csv -f 868.95M | tee >(wmbusmeters stdin:rtl433) | grep othertelegrams

weetmuts, 2024-05-19:
You can also publish mqtt messages using rtl433 and the use mosquitto to read these.
Check the README.

Telegrams can also be pulled in by listening on MQTT topics if they were captured by other tools like [rtl_433]

weetmuts, 2024-05-19:
```shell
wmbusmeters 'hex:CMD(/usr/bin/mosquitto_sub -h 192.168.x.x -t rtl_433/device/devices/6/Wireless-MBus/+/data | tr -d "\n" )'
```

pswiatki, 2024-05-19:
> Which other telegrams on 868.95 do you want to listen to?

I can get nice packets from my PIR motion sensors. it would be nice to have those.

hessel-a, 2024-05-30:
I think i can also use this. I run rtl_433 on 868.95 .. And collect info from weather stations "Fineoffset-WH" and put the information in homeassistant. I also see info from "Marlec-Solar" panels. Not mine  For now i will try to collect information from my electricity meter.

pswiatki, 2024-07-30:
> Which other telegrams on 868.95 do you want to listen to?

I am interested in transmissions from remotes for automatic gate openers, as well as TH sensors (I use multiple in different places around my house - trying to identify how and where the building heats up in the summer / cools down in winter - very old house, badly insulated). Sure, I also used thermal imagery to spot the major leaks, but this is mainly possible when it is really cold outside, so in winter.

closed 2024-10-30 by github-actions[bot]