wmbusmeters/wmbusmeters #1281: (shell) /bin/sh exited with non-zero return code: 127

Marcus54542, 2024-05-19:
### Type of request

Question

### OS version

Debian GNU/Linux 12 (bookworm)

### wmbusmeters version

the last

### Your message goes here

Hello

I try to follow tutorials and I have a little problem
On a raspberry 3 I have wmbusmeters that it works : 192.168.0.128

It is supposed to send to another raspberry 3 where Domoticz 192.168.0.7 is installed

When I launch the command :
sudo systemctl status wmbusmeters 
status I have this error:
**(shell) /bin/sh exited with non-zero return code: 127**

In wmbusmeters.conf I have the lines:

```
loglevel=normal
device=rtlwmbus
logtelegrams=false
format=json
meterfiles=/var/lib/wmbusmeters/meter_readings
meterfilesaction=overwrite
shell=/usr/bin/mosquitto_pub -h 192.168.0.7 -p 1883  -t domoticz/in -m "{\"idx\":1136,\"svalue\":\"$METER_TOTAL_M3\"}"
```

The error would seem to mean that the Shell line is formatted incorrectly

Can you tell me what's wrong

weetmuts, 2024-05-19:
Can you run with loglevel=debug and post here.

Marcus54542, 2024-05-19:
I'm a beginner in Linux, I don't really know where to put the command you propose.

For my problem, I found on the internet a part of the Mosquitto solution was not installed.
apt install mosquitto-clients

but I still have a problem I have this error message now
(shell) /bin/sh exited with non-zero return code: 1

weetmuts, 2024-05-19:
In your wmbusmeters.conf change
loglevel=normal
to
loglevel=debug

Marcus54542, 2024-05-19:
Ok 
`sudo systemctl status wmbusmeters
● wmbusmeters.service - "wmbusmeters service"
     Loaded: loaded (/lib/systemd/system/wmbusmeters.service; disabled; preset: enabled)
     Active: active (running) since Sun 2024-05-19 19:13:13 CEST; 4min 21s ago
       Docs: https://github.com/wmbusmeters/wmbusmeters
             man:wmbusmeters(1)
    Process: 16405 ExecStartPre=/bin/mkdir -p /var/lib/wmbusmeters/meter_readings (code=exited, status=0/SUCCESS)
    Process: 16406 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/lib/wmbusmeters/meter_readings (code=exited, status=0/SUCCESS)
    Process: 16407 ExecStartPre=/bin/mkdir -p /var/log/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 16408 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/log/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 16409 ExecStartPre=/bin/mkdir -p /run/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 16410 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /run/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 16411 ExecStart=/usr/sbin/wmbusmetersd /run/wmbusmeters/wmbusmeters.pid (code=exited, status=0/SUCCESS)
   Main PID: 16412 (wmbusmetersd)
      Tasks: 8 (limit: 756)
        CPU: 4min 25.683s
     CGroup: /system.slice/wmbusmeters.service
             ├─16412 /usr/sbin/wmbusmetersd /run/wmbusmeters/wmbusmeters.pid
             ├─16426 /bin/sh -c "ERRFILE=\$(mktemp -t wmbusmeters_rtlsdr.XXXXXXXXXXXX) ; echo ERRFILE=\$ERRFILE ;  date -Iseconds > \$ERRFILE ; tail -f \$ERRFILE & /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>>\$ERRFILE | /usr/bin/rtl_wmbus -s -f"
             ├─16429 tail -f /tmp/wmbusmeters_rtlsdr.iBfCMZe2Vjwq
             ├─16430 /usr/bin/rtl_sdr -d 0 -f 868.625M -s 1.6e6 -
             └─16431 /usr/bin/rtl_wmbus -s -f

May 19 19:17:28 compteur wmbusmetersd[16412]: (rtlwmbus) checkRTLWMBusFrame "T1;1;1;2024-05-19 17:17:28.000;42;53;91400645;0x1e44a511450640917b077ab0001005aabd0cc795e90ce397ea07954c3fcc09<0A>"
May 19 19:17:28 compteur wmbusmetersd[16412]: (rtlwmbus) received full frame
May 19 19:17:28 compteur wmbusmetersd[16412]: (wmbus) skipping already handled telegram leng=31.
May 19 19:17:29 compteur wmbusmetersd[16412]: (serial) received ascii "T1;1;1;2024-05-19 17:17:29.000;33;22;25006296;0x1e44a511966200257b077ad5001005a3e723794cf99f7ab0744ba4b2e48749<0A>T1;1;1;2024-05-19 17:17:29.000;26;32;25006296;0x1e44a511966200257b077ad5001005a3e723794cf99f7ab0744ba4b2e48>
May 19 19:17:29 compteur wmbusmetersd[16412]: (rtlwmbus) checkRTLWMBusFrame "T1;1;1;2024-05-19 17:17:29.000;33;22;25006296;0x1e44a511966200257b077ad5001005a3e723794cf99f7ab0744ba4b2e48749<0A>T1;1;1;2024-05-19 17:17:29.000;26;32;25006296;0x1e44a511966200257b077ad5001005a3e723794cf99f7ab0744ba>
May 19 19:17:29 compteur wmbusmetersd[16412]: (rtlwmbus) received full frame
May 19 19:17:29 compteur wmbusmetersd[16412]: (wmbus) skipping already handled telegram leng=31.
May 19 19:17:29 compteur wmbusmetersd[16412]: (rtlwmbus) checkRTLWMBusFrame "T1;1;1;2024-05-19 17:17:29.000;26;32;25006296;0x1e44a511966200257b077ad5001005a3e723794cf99f7ab0744ba4b2e48749<0A>"
May 19 19:17:29 compteur wmbusmetersd[16412]: (rtlwmbus) received full frame
May 19 19:17:29 compteur wmbusmetersd[16412]: (wmbus) skipping already handled telegram leng=31.
`

weetmuts, 2024-05-19:
You need to wait for a telegram. :-)

Marcus54542, 2024-05-19:
I receive many messages, I can read the value of my counter

I think the mosquitto broker was not installed on my raspberry pi.

I just did the following manipulations by following this tutorial:
https://www.lesalexiens.fr/domotique/tutoriel-installer-mosquitto-mqtt-sur-raspberry-pi/

On my PC with MQTT Explorer, I have "Disconnected from server" ☹

Marcus54542, 2024-05-19:
For the log earlier it was because of the  MyTapWater
		name=MyTapWater
		id=24509300
		driver=dme_07
I had written type instead of driver

I think I have more than the problem of the MQTT transmission

`sudo systemctl status wmbusmeters
● wmbusmeters.service - "wmbusmeters service"
     Loaded: loaded (/lib/systemd/system/wmbusmeters.service; disabled; preset: enabled)
     Active: active (running) since Sun 2024-05-19 20:05:32 CEST; 5min ago
       Docs: https://github.com/wmbusmeters/wmbusmeters
             man:wmbusmeters(1)
    Process: 3087 ExecStartPre=/bin/mkdir -p /var/lib/wmbusmeters/meter_readings (code=exited, status=0/SUCCESS)
    Process: 3093 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/lib/wmbusmeters/meter_readings (code=exited, status=0/SUCCESS)
    Process: 3098 ExecStartPre=/bin/mkdir -p /var/log/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 3100 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/log/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 3101 ExecStartPre=/bin/mkdir -p /run/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 3104 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /run/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 3105 ExecStart=/usr/sbin/wmbusmetersd /run/wmbusmeters/wmbusmeters.pid (code=exited, status=0/SUCCESS)
   Main PID: 3106 (wmbusmetersd)
      Tasks: 8 (limit: 756)
        CPU: 5min 24.513s
     CGroup: /system.slice/wmbusmeters.service
             ├─3106 /usr/sbin/wmbusmetersd /run/wmbusmeters/wmbusmeters.pid
             ├─3121 /bin/sh -c "ERRFILE=\$(mktemp -t wmbusmeters_rtlsdr.XXXXXXXXXXXX) ; echo ERRFILE=\$ERRFILE ;  date -Iseconds > \$ERRFILE ; tail -f \$ERRFILE & /usr/bin/rtl_sdr  -d 0 -f 868.625M >
             ├─3124 tail -f /tmp/wmbusmeters_rtlsdr.DLWkfM8eyv1v
             ├─3125 /usr/bin/rtl_sdr -d 0 -f 868.625M -s 1.6e6 -
             └─3126 /usr/bin/rtl_wmbus -s -f

May 19 20:10:44 compteur wmbusmetersd[3106]: (rtlwmbus) checkRTLWMBusFrame "T1;1;1;2024-05-19 18:10:44.000;25;22;25201261;0x1e44a511611220257b077a050010053456ac63a30078dd76717950a3e21b85<0A>"
May 19 20:10:44 compteur wmbusmetersd[3106]: (rtlwmbus) received full frame
May 19 20:10:44 compteur wmbusmetersd[3106]: (wmbus) skipping already handled telegram leng=31.
May 19 20:10:46 compteur wmbusmetersd[3106]: (serial) received ascii "T1;1;1;2024-05-19 18:10:46.000;61;65;24509300;0x1e44a511009350247b077a560010058151b3d439e6fc4dccafb412299f09d9<0A>T1;1;1;2024-05>
May 19 20:10:46 compteur wmbusmetersd[3106]: (rtlwmbus) checkRTLWMBusFrame "T1;1;1;2024-05-19 18:10:46.000;61;65;24509300;0x1e44a511009350247b077a560010058151b3d439e6fc4dccafb412299f09d9<0A>T1;1;1;2>
May 19 20:10:46 compteur wmbusmetersd[3106]: (rtlwmbus) received full frame
May 19 20:10:46 compteur wmbusmetersd[3106]: (wmbus) skipping already handled telegram leng=31.
May 19 20:10:46 compteur wmbusmetersd[3106]: (rtlwmbus) checkRTLWMBusFrame "T1;1;1;2024-05-19 18:10:46.000;58;54;24509300;0x1e44a511009350247b077a560010058151b3d439e6fc4dccafb412299f09d9<0A>"
May 19 20:10:46 compteur wmbusmetersd[3106]: (rtlwmbus) received full frame
May 19 20:10:46 compteur wmbusmetersd[3106]: (wmbus) skipping already handled telegram leng=31.`

Marcus54542, 2024-05-20:
Hello @weetmuts 
[wmbusmeters.log](https://github.com/wmbusmeters/wmbusmeters/files/15373810/wmbusmeters.log)

While browsing it I noticed 2 errors

(shell) output: >>>/usr/bin/rtl_wmbus: invalid option -- 'h'

and especially the ones that bother me the most is to send the data to MQTT

(shell) /bin/sh: return code 1
[2024-05-20_08:46:20] (shell) /bin/sh exited with non-zero return code:

closed 2024-05-20 by Marcus54542