wmbusmeters/wmbusmeters #1282: Device cul:c1 is not working

mwildbolz, 2024-05-20:
### Type of request

Bug

### OS version

Debian 11

### wmbusmeters version

1.16.1-18-gc1509f6

### Your message goes here

Hello!

After some time of working without problems, I suddenly get a timeout after a reset command leading to a not working state of wmbusmeters.
A restart of the container is not sufficient to restore operation. The CUL USB stick has to be removed and re-attached to the USB port. Afterwards, a container restart is needed.

My CUL stick is: nanoCUL USB Stick FTDI CC1101 868MHz FW 1.67 Knick-Antenne FHEM CUL 868
Meter is a Kamstrup FlowIQ 2200

Do you have any idea, what the problem could be? A docker-topic, or maybe wmbusmeters-topic, or a basic linux problem?
Time between failures is approx. some days to 20 days.

Logfile details as following:
```
(wmbus) reset completed /dev/ttyUSB0:cul[]
(wmbus) resetting /dev/ttyUSB0:cul[]
(wmbus) reset completed /dev/ttyUSB0:cul[]
(wmbus) resetting /dev/ttyUSB0:cul[]
(wmbus device) timeout request id 1
(cul) setting link mode(s) c1 is not supported for this cul device!
[ALARM SpecifiedDeviceNotFound] the device cul:c1 is not working
No wmbus device detected, waiting for a device to be plugged in.
[ALARM SpecifiedDeviceNotFound] the device cul:c1 is not working
[ALARM SpecifiedDeviceNotFound] the device cul:c1 is not working
[ALARM SpecifiedDeviceNotFound] the device cul:c1 is not working
[ALARM SpecifiedDeviceNotFound] the device cul:c1 is not working
[ALARM SpecifiedDeviceNotFound] the device cul:c1 is not working
[ALARM SpecifiedDeviceNotFound] the device cul:c1 is not working
No wmbus device detected, waiting for a device to be plugged in.
```

Docker-Compose File is:
```
version: "2"
services:
  wmbusmeters:
    image: wmbusmeters/wmbusmeters:latest
    container_name: wmbusmeters
    restart: always
    network_mode: host
    volumes:
      - ./config:/wmbusmeters_data
      - /etc/localtime:/etc/localtime:ro

    devices:
      - /dev/ttyUSBcul:/dev/ttyUSB0
```

mwildbolz, 2024-05-29:
Today, same thing again, after working for 8 days without any problems.
Any thoughts, where I could look to determine the root cause?

```
[2024-05-28_21:28:44] (wmbus) resetting /dev/ttyUSB0:cul[]
[2024-05-28_21:28:47] (wmbus) reset completed /dev/ttyUSB0:cul[]
[2024-05-28_23:28:46] (wmbus) resetting /dev/ttyUSB0:cul[]
[2024-05-28_23:28:54] (wmbus device) timeout request id 1
[2024-05-28_23:28:54] (cul) setting link mode(s) c1 is not supported for this cul device!
[2024-05-28_23:28:56] [ALARM SpecifiedDeviceNotFound] the device cul:c1 is not working
[2024-05-28_23:28:56] No wmbus device detected, waiting for a device to be plugged in.
[2024-05-28_23:29:58] [ALARM SpecifiedDeviceNotFound] the device cul:c1 is not working
[2024-05-28_23:31:00] [ALARM SpecifiedDeviceNotFound] the device cul:c1 is not working
[2024-05-28_23:32:02] [ALARM SpecifiedDeviceNotFound] the device cul:c1 is not working
[2024-05-28_23:33:04] [ALARM SpecifiedDeviceNotFound] the device cul:c1 is not working
[2024-05-28_23:34:06] [ALARM SpecifiedDeviceNotFound] the device cul:c1 is not working
[2024-05-28_23:35:08] [ALARM SpecifiedDeviceNotFound] the device cul:c1 is not working
[2024-05-28_23:36:10] [ALARM SpecifiedDeviceNotFound] the device cul:c1 is not working
[2024-05-28_23:37:12] [ALARM SpecifiedDeviceNotFound] the device cul:c1 is not working
[2024-05-28_23:38:14] [ALARM SpecifiedDeviceNotFound] the device cul:c1 is not working
```

Solving means re-plugging the USB stick AND re-starting wmbusmeters docker container. Restart of wmbusmeters alone does not solve the problem.

weetmuts, 2024-05-29:
Can you try the new rtl_reset command that is installed in the docker container?

weetmuts, 2024-05-29:
Ah, I know that rtl_reset by name only should reset rtl devices and not the cul, however what it does is that it
invokes libusb_reset_device which is a generic usb reset. Perhaps it can be used for cul.

https://github.com/ED6E0F17/rtl_reset/blob/master/rtl_reset.c

mwildbolz, 2024-05-31:
> Can you try the new rtl_reset command that is installed in the docker container?

So, you mean in case of the next hang, i should issue this command instead of unplugging and re-plugging the usb stick?
How can I issue the command inside the container? I just had a look and didn't find anything...

Thanks

closed 2024-08-30 by github-actions[bot]

mwildbolz, 2024-09-20:
Just had the same issue again.
I tried with the linked tool - built it on normal OS (not inside container) and executed the program, but it terminates with
`No supported devices found.`

Only solution is to re-plug the CUL USB stick

mwildbolz, 2024-09-20:
For me it looks like a problem with serial communication?
Maybe someone can find a hint in the following debug logs!

### Log 1 BEFORE re-plugging the cul stick:
```
[2024-09-20_20:07:18] (wmbusmeters) version: 1.17.1-12-gaaf45d8
[2024-09-20_20:07:18] (config) store meter files in: "/wmbusmeters_data/logs/meter_readings"
[2024-09-20_20:07:18] (config) using device: cul:c1
[2024-09-20_20:07:18] (config) number of meters: 1
[2024-09-20_20:07:18] (config) template Ortswasser 53052773 flowiq2200
[2024-09-20_20:07:18] (main) device /dev/ttyACM0 not currently used, detect contents...
[2024-09-20_20:07:18] (serial) check if /dev/ttyACM0 can be accessed
[2024-09-20_20:07:18] (serial) cannot open/find tty /dev/ttyACM0.
[2024-09-20_20:07:18] (main) ignoring /dev/ttyACM0, it does not respond as any of the supported wmbus devices.
[2024-09-20_20:07:18] (main) device /dev/ttyS1 not currently used, detect contents...
[2024-09-20_20:07:18] (serial) check if /dev/ttyS1 can be accessed
[2024-09-20_20:07:18] (serial) cannot open/find tty /dev/ttyS1.
[2024-09-20_20:07:18] (main) ignoring /dev/ttyS1, it does not respond as any of the supported wmbus devices.
[2024-09-20_20:07:18] (main) device /dev/ttyUSB0 not currently used, detect contents...
[2024-09-20_20:07:18] (serial) check if /dev/ttyUSB0 can be accessed
[2024-09-20_20:07:18] (serial) tty /dev/ttyUSB0 can be accessed
[2024-09-20_20:07:18] (serial) EVENT thread interrupted
[2024-09-20_20:07:18] (serial) before config: /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(214,185,237,102,228,61,135,32) modem(DTR|RTS)
[2024-09-20_20:07:18] (serial) after config:  /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(214,185,237,102,16) modem(DTR|RTS)
[2024-09-20_20:07:18] (serialtty) opened /dev/ttyUSB0 fd 3 (detect cul)
[2024-09-20_20:07:18] (serial /dev/ttyUSB0) sent "560D0A"
[2024-09-20_20:07:18] (serial) EVENT thread interrupted
[2024-09-20_20:07:18] (serial) received binary "5620312E36"
[2024-09-20_20:07:18] (cul) probe response "V 1.6"
[2024-09-20_20:07:19] (serial /dev/ttyUSB0) sent "560D0A"
[2024-09-20_20:07:19] (serial) EVENT thread interrupted
[2024-09-20_20:07:19] (serial) received binary "5620"
[2024-09-20_20:07:19] (cul) probe response "V "
[2024-09-20_20:07:19] (serial /dev/ttyUSB0) sent "560D0A"
[2024-09-20_20:07:19] (serial) EVENT thread interrupted
[2024-09-20_20:07:20] (serial) received binary "5620312E3637206E61"
[2024-09-20_20:07:20] (cul) probe response "V 1.67 na"
[2024-09-20_20:07:20] (serial) EVENT thread interrupted
[2024-09-20_20:07:20] (serialtty) closed /dev/ttyUSB0 (detect cul)
[2024-09-20_20:07:20] (cul) are you there? no
[2024-09-20_20:07:20] (main) ignoring /dev/ttyUSB0, it does not respond as any of the supported wmbus devices.
[2024-09-20_20:07:20] (main) device /dev/ttyUSB1 not currently used, detect contents...
[2024-09-20_20:07:20] (serial) check if /dev/ttyUSB1 can be accessed
[2024-09-20_20:07:20] (serial) cannot open/find tty /dev/ttyUSB1.
[2024-09-20_20:07:20] (main) ignoring /dev/ttyUSB1, it does not respond as any of the supported wmbus devices.
[2024-09-20_20:07:20] (main) device /dev/ttyUSB2 not currently used, detect contents...
[2024-09-20_20:07:20] (serial) check if /dev/ttyUSB2 can be accessed
[2024-09-20_20:07:20] (serial) cannot open/find tty /dev/ttyUSB2.
[2024-09-20_20:07:20] (main) ignoring /dev/ttyUSB2, it does not respond as any of the supported wmbus devices.
[2024-09-20_20:07:20] [ALARM SpecifiedDeviceNotFound] the device cul:c1 is not working
[2024-09-20_20:07:20] No wmbus device detected, waiting for a device to be plugged in.
[2024-09-20_20:07:20] (shell) exec (capture output) "/bin/pidof"
[2024-09-20_20:07:20] (shell) arg "wmbusmetersd"
[2024-09-20_20:07:20] (serial) EVENT thread interrupted
[2024-09-20_20:07:20] (shell) output: >>><<<
[2024-09-20_20:07:20] (shell) return code 1
[2024-09-20_20:07:20] (shell) 10 exited
[2024-09-20_20:07:20] (shell) exec (capture output) "/bin/pidof"
[2024-09-20_20:07:20] (shell) arg "wmbusmeters"
[2024-09-20_20:07:20] (shell) output: >>>7
<<<
[2024-09-20_20:07:20] (serial) EVENT thread interrupted
[2024-09-20_20:07:20] (shell) return code 0
[2024-09-20_20:07:20] (shell) 11 exited
[2024-09-20_20:07:20] (serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
[2024-09-20_20:07:20] (serial) waiting for stop
[2024-09-20_20:07:20] (serial) EVENT thread interrupted
[2024-09-20_20:07:20] (serial) TIMER thread interrupted
[2024-09-20_20:08:22] [ALARM SpecifiedDeviceNotFound] the device cul:c1 is not working
```
[... 45 lines cut ...]
[2024-09-20_20:13:09] (serialtty) opened /dev/ttyUSB0 fd 3 (cul)
[2024-09-20_20:13:09] (main) regular reset of cul  on /dev/ttyUSB0 will happen every 21600 seconds
[2024-09-20_20:13:09] (cul) set link mode c
[2024-09-20_20:13:09] (serial /dev/ttyUSB0) sent "6272630D0A"
[2024-09-20_20:13:09] (serial) EVENT thread interrupted
[2024-09-20_20:13:09] (serial) received binary "434D4F44450D0A"
[2024-09-20_20:13:09] (cul) checkCULFrame "CMODE<0D><0A>"
[2024-09-20_20:13:09] (cul) no leading 'b' so it is text and no frame
[2024-09-20_20:13:09] (cul) received "CMODE"[2024-09-20_20:13:09] (serial /dev/ttyUSB0) sent "5832310D0A"
[2024-09-20_20:13:09] (wmbus) no alarm (expected activity) for cul
[2024-09-20_20:13:09] (serial) EVENT thread interrupted
[2024-09-20_20:13:09] (main) device /dev/ttyUSB1 not currently used, detect contents...
[... cut ...]
[2024-09-20_20:13:09] (serial) cannot open/find tty /dev/ttyUSB2.
[2024-09-20_20:13:09] (main) ignoring /dev/ttyUSB2, it does not respond as any of the supported wmbus devices.
[2024-09-20_20:13:09] (serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
[2024-09-20_20:13:09] (serial) waiting for stop
[2024-09-20_20:14:06] (serial) received binary "625933313434333732433733"
[2024-09-20_20:14:06] (cul) checkCULFrame "bY3144372C73"
[2024-09-20_20:14:06] (cul) no eol found yet, partial frame
```

mwildbolz, 2024-10-27:
Same happened again.
Just tried only replugging the CUL-stick but both is needed: Re-plugging and re-starting the docker container of wmbusmeters

mwildbolz, 2024-12-21:
Same again from time to time.
This time I tried to just restart wmbusmeters. But same error message again after 1 minute.
```
... restarted here ...
[2024-12-21_13:49:45] No wmbus device detected, waiting for a device to be plugged in.
[2024-12-21_13:50:47] [ALARM SpecifiedDeviceNotFound] the device cul:c1 is not working
```

Re-plugging and re-starting the docker container solves the problem.

mwildbolz, 2024-12-21:
@weetmuts: Any additional ideas?
This issue was closed because I had a too long time without activity, but I added additional information over the past months. Or should I open a new issue?