wmbusmeters/wmbusmeters #1284: CYBLE GT3-ITR

Anas92230, 2024-05-22:
### Type of request

Question

### OS version

Ubuntu

### wmbusmeters version

1.16.1-29-g1a4169e

### Your message goes here

Hello,
Veolia change my watermater with :
-CYBLE G3-ITR
868MHZ
Birdz-Veolia G3 Technology
I hope that this device is supported :-)
I use this command : 
  - wmbusmeters --logtelegrams --verbose auto:t1 => nothing
If you have an idea :-)
Thanks

Anas92230, 2024-06-09:
Hi @weetmuts,
did you ever seen this watermater :-)?
thanks for help

toff63, 2024-08-17:
Based on this ["Product Specification"](https://www.birdz.com/wp-content/uploads/2022/06/G3-FP.pdf), it looks like this is the same as issue #479 . TLDR; you need a lorawan dongle like iU880B, use `:lora` instead of `t1` and pray for Veolia to not use encryption. I am curious about your findings as I have the same at home.

closed 2024-11-16 by github-actions[bot]

xamyp, 2025-01-07:
hello @Anas92230 @famillenounoux did you solve the problem because I am in the same situation ?