wmbusmeters/wmbusmeters #1287: water meter in liters

Marcus54542, 2024-05-23:
### Type of request

Question

### OS version

Debian GNU/Linux 12 (bookworm)

### wmbusmeters version

the last

### Your message goes here

Hello
I would like to try to display the value of my water meter in liters
I found this command @weetmuts   --calculate_corrected_total_m3='total_m3*1000 factor'
where should I put it please ? in  wmbusmeters.conf ?

My counter :  `{"media":"water","meter":"dme_07","name":"MyTapWater","id":"24509399","total_m3":29.547,"status":"OK","timestamp":"2024-05-23T07:03:05Z","device":"rtlwmbus[00000001]","rssi_dbm":122}`

Thank you

Marcus54542, 2024-05-23:
in MyTapWater :)

calculate_corrected_total_m3=total_m3*1000 factor

closed 2024-05-23 by Marcus54542

weetmuts, 2024-05-23:
calculate_corrected_total_l=total_m3

is much much much better! wmbusmeters tracks the units. you can auto convert between units of the same quantity. ie volume.

also you do not lie in the json. ie do not use m3 for l (litres).

:-)

weetmuts, 2024-05-23:
calculate_total_l=total_m3

should also work.