wmbusmeters/wmbusmeters #1290: Using multiple inputs in wmbusmeters (e.g. RTLSDR for wmbus devices and a serial dongle for mbus meters)

probutus, 2024-05-25:
### Type of request

Question

### OS version

Debian Bookworm

### wmbusmeters version

current master

### Your message goes here

I am happily using wmbusmeters for my heat and water meters. Now I got a wired mbus electricity meter which is connected via UART and level shifter. I can read out mbus registers manually on the device. Is it possible to give wmbusmeters two devices like rtlsdr and also the uart device?

My config looks like this:
(the commented out serial device was my old nanocul which could not digest the needed c1 telegrams at that thime)

`loglevel=normal
#device=/dev/serial/by-id/usb-SHK_NANO_CUL_868-if00-port0:t1
device=rtlwmbus:t1,c1
donotprobe=/dev/ttyAMA0
logtelegrams=false
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=overwrite
logfile=/var/log/wmbusmeters/wmbusmeters.log
shell=/usr/bin/mosquitto_pub -h nas -t wmbusmeters/$METER_ID -m "$METER_JSON"
ignoreduplicates=true
`

weetmuts, 2024-05-25:
Yep, check the README! :-)

probutus, 2024-05-26:
shame on me :). I found the correct way of handling this. Now I see that the driver I needed did not exist so I created one and attached it to the post requesting it:

https://github.com/wmbusmeters/wmbusmeters/issues/1219

This post can be closed

closed 2024-05-26 by probutus